marcone/teslausb #707: Enhancement: Store many lightshows, use WiFi web UI to select active one

JHamCell, 2021-12-27:
Tesla only allows a single lightshow t be stored.
For me that means running back and forth between car and desktop to try new ones out, or carrying multiple thumb drives.
Please consider a feature that
- stores several light show configs in a sync'able filesystem.
- an option on the WebUI to select the active one.

Said feature could copy the required files to either 
- a /LightShow directory on the music drive
- or create its own lightshow.bin file that is then presented as a new USB device to the car.

Thank you for your continued work & support on TeslaUSB.

marcone, 2021-12-27:
it's a neat idea, but to be honest I think Lightshow is a gimmick that people won't be playing with anymore in a few weeks, so I'm not going to spend any time on it. You're welcome to submit a PR.

(note that you can somewhat automate this currently by using music sync and the "Trigger archive/sync" button in the web UI: drag whichever lightshow you want to your Lightshow folder on the server, then trigger a sync. Only works when you're in wifi range, of course)

marcone, 2021-12-31:
It was mentioned in another issue that adding a lightshow folder to the music drive causes the car to no longer load music from that drive. Given that, I'm closing this issue since I don't think adding support for a separate drive just for the lightshow feature is worth it.

closed 2021-12-31 by marcone