marcone/teslausb #709: RPi0w hangs on fresh install

hoopsta1423, 2021-12-30:
I'm having trouble installing on a 64gb card where things seem to stall out at:

'verify-and-configure-archive: Installing/updating required packages if needed'

RPi connects to network and I'm able to ssh into it. When plugged into my laptop, the CAM drive pops up for about a minute and then disappears. Pulling the logs show the following:

`Sat  6 Nov 17:45:32 GMT 2021 : Detecting whether to update wpa_supplicant.conf
Sat  6 Nov 17:45:32 GMT 2021 : Wifi variables specified, and no /boot/WIFI_ENABLED. Building wpa_supplicant.conf.
Sat  6 Nov 17:45:33 GMT 2021 : Rebooting...
Sat  6 Nov 17:46:12 GMT 2021 : Detecting whether to update wpa_supplicant.conf
Sat  6 Nov 17:46:12 GMT 2021 : Grabbing main setup file.
Sat  6 Nov 17:46:14 GMT 2021 : Starting setup.
Sat  6 Nov 17:46:15 GMT 2021 : curl -L https://github.com/marcone/teslausb/archive/main-dev.tar.gz
Sat  6 Nov 17:46:17 GMT 2021 : Downloaded /tmp/455/setup-teslausb ...
Sat  6 Nov 17:46:17 GMT 2021 : /root/bin/setup-teslausb is up to date
Sat  6 Nov 17:46:17 GMT 2021 : Downloaded /tmp/455/rc.local ...
Sat  6 Nov 17:46:17 GMT 2021 : rc.local is up to date
Sat  6 Nov 17:46:17 GMT 2021 : Updating package index files...
Thu 30 Dec 12:46:07 EST 2021 : Verifying that the requested configuration is valid...
Thu 30 Dec 12:46:07 EST 2021 : Downloaded /tmp/verify-configuration.sh ...
Thu 30 Dec 12:46:08 EST 2021 : USB_DRIVE is not set. SD card will be used.
Thu 30 Dec 12:46:08 EST 2021 : Verifying that there is sufficient space available on the MicroSD card...
Thu 30 Dec 12:46:08 EST 2021 : There is sufficient space available.
Thu 30 Dec 12:46:12 EST 2021 : Downloading additional setup scripts.
Thu 30 Dec 12:46:12 EST 2021 : Downloaded /root/bin/remountfs_rw ...
Thu 30 Dec 12:46:12 EST 2021 : Downloaded /root/bin/make_snapshot.sh ...
Thu 30 Dec 12:46:12 EST 2021 : Downloaded /root/bin/mount_snapshot.sh ...
Thu 30 Dec 12:46:12 EST 2021 : Downloaded /root/bin/release_snapshot.sh ...
Thu 30 Dec 12:46:12 EST 2021 : Downloaded /root/bin/force_sync.sh ...
Thu 30 Dec 12:46:13 EST 2021 : Downloaded /root/bin/mountoptsforimage ...
Thu 30 Dec 12:46:13 EST 2021 : Downloaded /root/bin/mountimage ...
Thu 30 Dec 12:46:13 EST 2021 : Downloaded /root/bin/enable_gadget.sh ...
Thu 30 Dec 12:46:13 EST 2021 : Downloaded /root/bin/disable_gadget.sh ...
Thu 30 Dec 12:46:13 EST 2021 : Downloaded /tmp/create-backingfiles-partition.sh ...
Thu 30 Dec 12:46:13 EST 2021 : Downloaded /tmp/create-backingfiles.sh ...
Thu 30 Dec 12:46:13 EST 2021 : Downloaded /tmp/make-root-fs-readonly.sh ...
Thu 30 Dec 12:46:13 EST 2021 : Downloaded /tmp/configure.sh ...
Thu 30 Dec 12:46:13 EST 2021 : Fixing the modules-load parameter in /boot/cmdline.txt...
Thu 30 Dec 12:46:13 EST 2021 : Fixed cmdline.txt.
Thu 30 Dec 12:46:18 EST 2021 : create-backingfiles-partition: Checking existing partitions...
Thu 30 Dec 12:46:19 EST 2021 : create-backingfiles-partition: Modifying partition table for backing files partition...
Thu 30 Dec 12:46:19 EST 2021 : create-backingfiles-partition: Modifying partition table for mutable (writable) partition for script usage...
Thu 30 Dec 12:46:20 EST 2021 : create-backingfiles-partition: Writing updated partitions to fstab and /boot/cmdline.txt
Thu 30 Dec 12:46:21 EST 2021 : create-backingfiles-partition: Formatting new partitions...
Thu 30 Dec 12:46:28 EST 2021 : Mounting the partition for the backing files...
Thu 30 Dec 12:46:29 EST 2021 : Mounted the partition for the backing files.
Thu 30 Dec 12:46:29 EST 2021 : Mounting the mutable partition...
Thu 30 Dec 12:46:29 EST 2021 : Mounted the mutable partition.
Thu 30 Dec 12:46:29 EST 2021 : Creating backing disk files.
Thu 30 Dec 12:46:29 EST 2021 : create-backingfiles: starting
Thu 30 Dec 12:46:29 EST 2021 : create-backingfiles: cam: 30G, music: 0, mountpoint: /backingfiles, exfat: false
Thu 30 Dec 12:46:30 EST 2021 : create-backingfiles: Allocating 31457280K for /backingfiles/cam_disk.bin...
Thu 30 Dec 12:46:31 EST 2021 : create-backingfiles: Creating filesystem with label 'CAM'
Thu 30 Dec 12:46:34 EST 2021 : create-backingfiles: created camera backing file
Thu 30 Dec 12:46:34 EST 2021 : create-backingfiles: done
Thu 30 Dec 12:46:34 EST 2021 : updated /etc/fstab for /mnt/cam
Thu 30 Dec 12:46:36 EST 2021 : calling configure.sh
Thu 30 Dec 12:46:37 EST 2021 : configure: Pushover enabled
Thu 30 Dec 12:46:37 EST 2021 : configure: Gotify not enabled.
Thu 30 Dec 12:46:37 EST 2021 : configure: IFTTT not enabled.
Thu 30 Dec 12:46:37 EST 2021 : configure: Discord not enabled.
Thu 30 Dec 12:46:37 EST 2021 : configure: Webhook not enabled.
Thu 30 Dec 12:46:37 EST 2021 : configure: Slack not enabled.
Thu 30 Dec 12:46:37 EST 2021 : configure: Matrix not configured.
Thu 30 Dec 12:46:37 EST 2021 : configure: Telegram not enabled.
Thu 30 Dec 12:46:37 EST 2021 : configure: SNS not configured.
Thu 30 Dec 12:46:37 EST 2021 : Downloaded /root/bin/send-push-message ...
Thu 30 Dec 12:46:37 EST 2021 : Downloaded /root/bin/send_sns.py ...
Thu 30 Dec 12:46:37 EST 2021 : Downloaded /root/bin/send_matrix.py ...
Thu 30 Dec 12:46:37 EST 2021 : configure: Checking archive configs: 
[... 199 lines cut ...]
Thu 30 Dec 13:03:16 EST 2021 : configure: Slack not enabled.
Thu 30 Dec 13:03:16 EST 2021 : configure: Matrix not configured.
Thu 30 Dec 13:03:16 EST 2021 : configure: Telegram not enabled.
Thu 30 Dec 13:03:16 EST 2021 : configure: SNS not configured.
Thu 30 Dec 13:03:16 EST 2021 : Downloaded /root/bin/send-push-message ...
Thu 30 Dec 13:03:16 EST 2021 : Downloaded /root/bin/send_sns.py ...
Thu 30 Dec 13:03:16 EST 2021 : Downloaded /root/bin/send_matrix.py ...
Thu 30 Dec 13:03:16 EST 2021 : configure: Checking archive configs: 
Thu 30 Dec 13:03:17 EST 2021 : configure: rsync seems to work OK
Thu 30 Dec 13:03:17 EST 2021 : configure: done
Thu 30 Dec 13:03:17 EST 2021 : configure: Using archive module: run/cifs_archive
Thu 30 Dec 13:03:17 EST 2021 : configure: Installing base archive scripts into /root/bin
Thu 30 Dec 13:03:17 EST 2021 : Downloaded /root/bin/archiveloop ...
Thu 30 Dec 13:03:17 EST 2021 : Downloaded /root/bin/waitforidle ...
Thu 30 Dec 13:03:17 EST 2021 : Downloaded /root/bin/remountfs_rw ...
Thu 30 Dec 13:03:17 EST 2021 : Downloaded /root/bin/awake_start ...
Thu 30 Dec 13:03:17 EST 2021 : Downloaded /root/bin/awake_stop ...
Thu 30 Dec 13:03:17 EST 2021 : configure: Skipping tesla_api.py install because no credentials were provided
Thu 30 Dec 13:03:17 EST 2021 : configure: Installing archive module scripts
Thu 30 Dec 13:03:17 EST 2021 : Downloaded /tmp/verify-and-configure-archive.sh ...
Thu 30 Dec 13:03:17 EST 2021 : Downloaded /root/bin/archive-clips.sh ...
Thu 30 Dec 13:03:17 EST 2021 : Downloaded /root/bin/connect-archive.sh ...
Thu 30 Dec 13:03:17 EST 2021 : Downloaded /root/bin/disconnect-archive.sh ...
Thu 30 Dec 13:03:17 EST 2021 : Downloaded /root/bin/archive-is-reachable.sh ...
Thu 30 Dec 13:03:17 EST 2021 : verify-and-configure-archive: Installing/updating required packages if needed`

marcone, 2021-12-30:
My guess is that the apt package index is corrupt, possibly due to a previously interrupted setup.
Try ssh-ing into the Pi and running `sudo apt-get -y --force-yes install hping3` manually.

hoopsta1423, 2021-12-30:
Thanks! dpkg install was interrupted (see below) After fixing that and rebooting, it seems to have solved the issue!

pi@teslausb-ModelX:~ $ sudo apt-get -y --force-yes install hping3 
W: --force-yes is deprecated, use one of the options starting with --allow instead.
E: dpkg was interrupted, you must manually run 'sudo dpkg --configure -a' to correct the problem. 
pi@teslausb-ModelX:~ $ sudo dpkg --configure -a
Setting up busybox-syslogd (1:1.30.1-4) ...
Processing triggers for man-db (2.8.5-2) ...
Processing triggers for systemd (241-7~deb10u8+rpi1) ...
pi@teslausb-ModelX:~ $ sudo apt-get -y --force-yes install hping3 
Reading package lists... Done
Building dependency tree       
Reading state information... Done
hping3 is already the newest version (3.a2.ds2-7).
0 upgraded, 0 newly installed, 0 to remove and 9 not upgraded.
W: --force-yes is deprecated, use one of the options starting with --allow instead.

closed 2021-12-30 by hoopsta1423