marcone/teslausb #711: CAM and MUSIC are not available when there is no Wifi during startup

adjego, 2022-01-01:
I’m using latest master which now has configfs and i see some strange behaviour. For me it always works when leaving from home where TeslaUSB is able to connect to the internet before leaving. If i leave from a different location, it is never working. I also checked this connecting my rapi zero to my windows machine and there the same happens. When the wifi is not available for TeslaUSB, it will not show the CAM and MUSIC disk on windows. When i then enable wifi accesspoint, it will show up.

When this occurs the led is showing the slowblink. The tricky part is that when it is not connected to Wifi, i can't check any debug. I will try to setup uart via usb to see if i can get in and do some extra debugging.

[diagnostics.txt](https://github.com/marcone/teslausb/files/7797568/diagnostics.txt)
[archiveloop.log](https://github.com/marcone/teslausb/files/7797571/archiveloop.log)
[teslausb-headless-setup.log](https://github.com/marcone/teslausb/files/7797572/teslausb-headless-setup.log)

hoopsta1423, 2022-01-01:
I just noticed this on mine this morning too. Using latest image, dashcam icon was present when i left my house this AM. After parking at a store for an hour, got in to go home and dashcam icon never came back

marcone, 2022-01-01:
I can't reproduce this. I see the CAM/MUSIC drives appear regardless of whether the Pi is booted within or outside wifi range. The only difference is that it takes about 10 seconds longer for the drives to appear when outside wifi range.

hoopsta1423, 2022-01-02:
I just went to trader Joe's. Left the house around 9am with dashcam working
correctly. Parked at the store and shopped for 30m with sentry
enabled/active. Came back out to car with 1 sentry event but unable to view
with error 'dashcam currently unavailable'. Drove home and dashcam icon
never came back on. Waiting to see what will happen if/when rpi connects to
home wifi

hoopsta1423, 2022-01-02:
> I just went to trader Joe's. Left the house around 9am with dashcam working correctly. Parked at the store and shopped for 30m with sentry enabled/active. Came back out to car with 1 sentry event [...]

The rpi did reconnect to home wifi once I got back. No sentry event was recorded even though the car stated there was 1 event  =(

Here are the logs, hopefully they are helpful:
[diagnostics.txt](https://github.com/marcone/teslausb/files/7799237/diagnostics.txt)
[archiveloop.log](https://github.com/marcone/teslausb/files/7799238/archiveloop.log)
[teslausb-headless-setup.log](https://github.com/marcone/teslausb/files/7799239/teslausb-headless-setup.log)

adjego, 2022-01-02:
I reverted back to commit cd2bfa9126bae177398ae8c64adfb84712d02602 to see if it is caused by the new configfs but that does not seems to be the problem. I now reverted back to commit 13705094cc566b2e9eee2a208f6725fc6a51b7d2 because that version did not show this issue in the past, so let’s see if this stops showing the problem.

I’m running 2021.44.25.2 on a Tesla Model 3 SR+ from 2019.

marcone, 2022-01-02:
> No sentry event was recorded even though the car stated there was 1 event

The car can be in Sentry mode even without a drive present, and will report how many times it triggered. It just won't record anything.

Your diagnostics show the Pi waking up around 6 am, restart some time between 8:17 and 8:27, and then restart again around 9:12, The archive is not reachable when it restarts at 9:12. If you left the house at 9, then that means the Pi restarted while you were driving or parked at the store. The Pi does reconnect the drive to the car at this point, but from your description it seems like the car ignored that. If you look at the RecentClips in the web UI, you might see recordings up to about 9:12.
At 9:37, the archive becomes reachable again. I assume this is when you came back from the store. It does not actually archive anything at this point, because there were no files to archive.
The Pi then restarts again around 9:40, and you capture the diagnostics at 9:52.

hoopsta1423, 2022-01-02:
> 

I left the house at 904, you can see the pi trying to reconnect to the archive server right at that time. The drive did mount in the car at that point because the recording icon was present on the MCU.

hoopsta1423, 2022-01-02:
Taking a look at the clips now, there are recent event clips from the time I leave (904am) until the time I get to the store (actually 910am). 

At this point, sentry was enabled and I don't know why the rpi would've restarted, nor why it wouldn't then remount in the car

hoopsta1423, 2022-01-03:
Yup I don't know what to do here unfortunately. Did a fresh install last night. Left the house to take my son to daycare this morning  at 733AM. You can see the Rpi trying to find the server after we left and transition to waiting for archive server. Dashcam Icon was active in the car, and recent clips show it was active until I took him out of the car at daycare (742AM)

From that point, the Rpi reboots at 749AM when I get back in the car - so no recordings while sentry mode is enabled.

Rpi also did not mount in the car after that reboot at 749 because I did not see the dashcam icon on the ride home

Got back home at 759 and the Rpi reconnected to home wifi just fine

[diagnostics (3).txt](https://github.com/marcone/teslausb/files/7802038/diagnostics.3.txt)

marcone, 2022-01-03:
What happens if you unplug and reconnect the Pi while you're away from wifi? (as opposed to just leaving it plugged in so it powers up at the same time as the car)

adjego, 2022-01-03:
For me it will still have the problem. Slow blink is visible from the led, but there is no disk is mounted in the car. But this scenario also happens when i connect the rapi zero to my windows laptop. I can easily disable the wifi which is used for the rapi, and then also on windows it does not show up. And when i then enable wifi AC it start to show up after some processing. Might it for example be something related to set_time? That some times out and ot does not recover?

marcone, 2022-01-03:
`set_time` is only called when the archive server is reachable, so wouldn't be called when the Pi is not on wifi. Your diagnostics also show that the Pi connected the drives to the car when it was booted away from wifi.
 
For debugging, it might be easier to configure AP_SSID and AP_PASS. You should then be able to connect to the Pi's AP even when the Pi itself isn't connected to wifi.

hoopsta1423, 2022-01-03:
Same for me, if i reconnect away from wifi, rpi boots into slow single blink but drive doesn't remount in the car. I'm wondering if v11 has made some changes that break this in some scenarios

adjego, 2022-01-03:
As stated, i can reproduce this when i connect my rapi to my windows laptop and don’t let it connect to the internet. Then it does not show the drives. When it is then connected they appear. I btw reverted now back to 13705094cc566b2e9eee2a208f6725fc6a51b7d2 and i don’t have the issue anymore.

marcone, 2022-01-04:
And unfortunately I cannot reproduce it that way. Can you please do the following:
- configure AP_SSID and AP_PASS so you can connect to the Pi's AP from your laptop
- let the Pi boot while it is not connected to internet
- connect to the Pi's AP and ssh in
- capture diagnostics
- let the Pi connect to internet
- capture diagnostics again
- attach both diagnostics files here
(note that you only need to get the diagnostics. archiveloop and setup log are already included in that)

jazzblu, 2022-01-04:
I have been having the same issue lately.  To reconnect the pi when there is no wifi I just have to soft reboot the car once. It's the only way I can make it work when I'm not near my garage. I too suspect the problem has something to do with the V11 update.

hoopsta1423, 2022-01-04:
> As stated, i can reproduce this when i connect my rapi to my windows laptop and don’t let it connect to the internet. Then it does not show the drives. When it is then connected they appear. I btw reverted now back to [1370509](https://github.com/marcone/teslausb/commit/13705094cc566b2e9eee2a208f6725fc6a51b7d2) and i don’t have the issue anymore.

Going to try this and see what happens

adjego, 2022-01-05:
> And unfortunately I cannot reproduce it that way. Can you please do the following:
> 
> * configure AP_SSID and AP_PASS so you can connect to the Pi's AP from your laptop
> * let the Pi boot while it is not connected to internet
> * connect to the Pi's AP and ssh in (or use the web interface)
> * capture diagnostics
> * let the Pi connect to internet
> * capture diagnostics again
> * attach both diagnostics files here
>   (note that you only need to get the diagnostics. archiveloop and setup log are already included in that)

Well, some strange things are going on i would say 😇 

Trying to reproduce the issue where also on Windows it fails i was unable to do so. So i then checked in the car where the issue was indeed still visible, so i enabled the AP function, and from the moment i enabled AP, it also started to work in the car. So when there is now no WIFI available when the raspberry is starting, it works with latest branch.

I have no clue how this can be the case, but for some kind of reason it started to work now.

jazzblu, 2022-01-11:
I concur that turning on the Pi's AP solves the problem. Hooray!

hoopsta1423, 2022-01-11:
> 

I think this actually worked for me as well. A few days ago I turned on the AP mode to try and get logs when it was happening but didn't have time to do it. I just noticed yesterday that it was behaving as it should be =)

marcone, 2022-01-11:
I suspect that just the act of running setup again made things work again, since AP mode has nothing to do with whether or not the drive is presented to the car, and when I said earlier that I couldn't reproduce the issue, that was without AP.

adjego, 2022-01-12:
Well, i can say for sure that running the setup again fixed it, i’m just describing what i’m seeing. Before enabling AP i think i did almost 5 or 6 setup runs and every time it did not work, only when i reverted back to 1370509.

What i do need to mention is that using 1370509 i still needed to do the adjustment on the connect function to first remove the kernel module, with a sleep of 5 seconds before it started to work, but i had that already in place before. 

Maybe for some kind of reason there is some different behaviour regarding how the car handles the mount.

marcone, 2022-01-12:
Closing this since OP no longer has the problem, and I could never reproduce it.

closed 2022-01-12 by marcone