marcone/teslausb #715: Teslausb dismounts when parked at work

aaronac8, 2022-01-12:
Please see attached diagnostic file:
[diagnostics.txt](https://github.com/marcone/teslausb/files/7858523/diagnostics.txt)

When I leave in the morning for work the teslausb is mounted, but when I get out of work (i.e. 1/12/2022 at 5pm EST) it is not mounted.  However, when I arrive home I can access the teslausb.

How do I prevent the teslausb from being inaccessible on my drive home?

Thank you

aaronac8, 2022-01-12:
Just did an update to the teslausb, and now my diagnostic file is:
[diagnostics (1).txt](https://github.com/marcone/teslausb/files/7858617/diagnostics.1.txt)

aaronac8, 2022-01-14:
The music folder works fine at all times.

aaronac8, 2022-01-14:
My Teslausb was not mounted today (14 January 2022) at 0720 when I left my home, but did mount when I left work at 1700.  Please see diagnostics:

[diagnostics.txt](https://github.com/marcone/teslausb/files/7873576/diagnostics.txt)

marcone, 2022-01-16:
The fact that the music drive works fine either indicates corruption of the "cam" filesystem image, or some problem on the car's side. The diagnostics show nothing obviously wrong. This seems like an another instance of https://github.com/marcone/teslausb/issues/654

closed 2022-01-16 by marcone

aaronac8, 2022-01-16:
How do I correct a corrupt "cam" file system image?

marcone, 2022-01-16:
> How do I correct a corrupt "cam" file system image?

TeslaUSB does that automatically, which is why I'm leaning towards "problem on the car's side".
You could connect the Pi to a PC and see if the drive shows up correctly there.

aaronac8, 2022-01-16:
After connecting the Pi to a PC, the drive does show up correctly there.
If it is a problem "on the car side" then is there anything that can be
done?

marcone, 2022-01-17:
Sorry, no