marcone/teslausb #718: Archiveloop fail to start with kernel 5.10

remygrandin, 2022-01-27:
### Describe the problem

Hi,

I followed the manual upgrade stepas, and on the last Raspberry Pi OS Lite, the teslausb.service fail to start with this error :

```
Jan 27 00:17:14 teslausb systemd[1]: teslausb.service: Failed with result 'exit-code'.
Jan 27 00:17:15 teslausb systemd[1]: teslausb.service: Scheduled restart job, restart counter is at 7.
Jan 27 00:17:15 teslausb systemd[1]: Stopped TeslaUSB archiveloop service.
Jan 27 00:17:15 teslausb systemd[1]: Started TeslaUSB archiveloop service.
Jan 27 00:17:15 teslausb bash[443]: /usr/bin/tvservice is not supported when using the vc4-kms-v3d driver.
Jan 27 00:17:15 teslausb bash[443]: Similar features are available with standard linux tools
Jan 27 00:17:15 teslausb bash[443]: such as modetest from libdrm-tests.
Jan 27 00:17:15 teslausb systemd[1]: teslausb.service: Main process exited, code=exited, status=1/FAILURE
Jan 27 00:17:15 teslausb systemd[1]: teslausb.service: Failed with result 'exit-code'.
Jan 27 00:17:16 teslausb systemd[1]: teslausb.service: Scheduled restart job, restart counter is at 8.
Jan 27 00:17:16 teslausb systemd[1]: Stopped TeslaUSB archiveloop service.
Jan 27 00:17:16 teslausb systemd[1]: Started TeslaUSB archiveloop service.
Jan 27 00:17:16 teslausb bash[457]: /usr/bin/tvservice is not supported when using the vc4-kms-v3d driver.
Jan 27 00:17:16 teslausb bash[457]: Similar features are available with standard linux tools
```

It seem to be linked to the upgrade to the 5.10 kernel sometime in 2021 https://forums.raspberrypi.com/viewtopic.php?t=288234

We should try to find an alternate command to the `/usr/bin/tvservice -o` in the archive loop service to be compatible for this version onward.

### Device

Raspberry Pi Zero W

### Car Model

Model 3

### USB connection

Center console

remygrandin, 2022-01-27:
commenting the `/usr/bin/tvservice -o` line in the service seem to fix the fail loop

marcone, 2022-01-27:
> I followed the manual upgrade steps

It is highly recommended to use the prebuilt image, to avoid problems like this.
There is no inherent problem with kernel 5.10 and tvservice:
```
root@teslausb:~# uname -a
Linux teslausb 5.10.52+ #1441 Tue Aug 3 18:08:18 BST 2021 armv6l GNU/Linux
root@teslausb:~# /usr/bin/tvservice -o
Powering off HDMI
root@teslausb:~# 
```
If you insist on using manual setup, maybe you can select a different video driver, as hinted at by the error message you're getting?

closed 2022-01-27 by marcone

remygrandin, 2022-01-27:
I prefer using an official image, as I prefered to rather audit just the extra scripts from here rather than diff the whole image , it looked less time consumming :-)

I'm not sure how to set the video driver to another one, I'll look if I can find something in this regard.

marcone, 2022-01-27:
> I prefer using an official image, as I prefered to rather audit just the extra scripts from here rather than diff the whole image , it looked less time consumming :-)

Fair enough. I also provide the [necessary code/diffs to build your own image](https://github.com/marcone/teslausb/tree/main-dev/pi-gen-sources) from official sources/repos, so that's an option too.

remygrandin, 2022-01-27:
Found it,

Commenting the following 2 lines in the boot config (/boot/config.txt) did the trick

```
# Enable DRM VC4 V3D driver
dtoverlay=vc4-kms-v3d
max_framebuffers=2
```