marcone/teslausb #719: mutable partition corrupt

RichieB2B, 2022-01-27:
### Describe the problem

10 days after a clean install the disk would not mount in my Telsa anymore. Rebooting the Pi did not help. TeslaCam WiFi access point was still present. When connected to my home WiFi the Pi would ping but ssh and CIFS ports were not open (connection refused). Upon inspection of the SD card in a Linux machine I noticed partition 4 (mutable) would not mount. A manual `fsck.ext4` showed it was corrupted and was able to fix it.

Might be related to #654

### Device

Raspberry Pi Zero W

### Car Model

Model S

### USB connection

Center console

### Logs

No logs since `/mutable` was not mounted.

### Additional information

```
sdl                            119.1G
├─sdl1 /mnt/usb   boot           250M DF8B-3862
├─sdl2 /mnt/usb2  rootfs           2G a274b77c-9810-4d4f-a743-3fcb332490e8
├─sdl3 /mnt/usb3  backingfiles 116.8G 04a47e50-0645-4f75-a767-19b70c884847
└─sdl4            mutable        100M fc6751ec-9d09-4613-8017-a3f3ccc0f630
# fsck.ext4 /dev/sdl4
e2fsck 1.44.5 (15-Dec-2018)
ext2fs_open2: Superblock checksum does not match superblock
fsck.ext4: Superblock invalid, trying backup blocks...
mutable was not cleanly unmounted, check forced.
Pass 1: Checking inodes, blocks, and sizes
Pass 2: Checking directory structure
Pass 3: Checking directory connectivity
Pass 4: Checking reference counts
Pass 5: Checking group summary information
Free blocks count wrong for group #0 (6360, counted=1585)
etc
```

RichieB2B, 2022-01-28:
It is very likely I can test this in about 10 days. 

According to https://raspberrypi.stackexchange.com/questions/61723/raspberry-pi-3-and-raspbian-jessie-how-to-run-fsck-at-boot#61739 you also need `fsck.mode=yes`

RichieB2B, 2022-01-28:
I suppose the kernel parameters are just to force an fsck on the root file system? For other file systems set `Maximum mount count` to 1 and in `/etc/fstab` set the last parameter to 2. See https://linuxconfig.org/how-to-force-fsck-to-check-filesystem-after-system-reboot-on-linux

RichieB2B, 2022-01-28:
Alternatively, perhaps using XFS the for `mutable` partition could make this issue go away?
```
Jan 27 22:16:09 linux-box kernel: [22235166.158696]  sdl: sdl1 sdl2 sdl3 sdl4
Jan 27 22:16:09 linux-box kernel: [22235166.172824] sd 7:0:0:0: [sdl] Attached SCSI removable disk
Jan 27 22:18:20 linux-box kernel: [22235297.803238] XFS (sdl3): Mounting V5 Filesystem
Jan 27 22:18:21 linux-box kernel: [22235298.154325] XFS (sdl3): Starting recovery (logdev: internal)
Jan 27 22:18:21 linux-box kernel: [22235298.167856] XFS (sdl3): Ending recovery (logdev: internal)
Jan 27 22:18:21 linux-box kernel: [22235298.181900] xfs filesystem being mounted at /mnt/usb3 supports timestamps until 2038 (0x7fffffff)
Jan 27 22:18:24 linux-box kernel: [22235301.799429] EXT4-fs (sdl4): VFS: Found ext4 filesystem with invalid superblock checksum.  Run e2fsck?
Jan 27 22:21:38 linux-box kernel: [22235495.539784] EXT4-fs (sdl4): VFS: Found ext4 filesystem with invalid superblock checksum.  Run e2fsck?
Jan 27 22:21:44 linux-box kernel: [22235501.940994] EXT4-fs (sdl4): VFS: Found ext4 filesystem with invalid superblock checksum.  Run e2fsck?
Jan 27 22:22:32 linux-box kernel: [22235549.907737] ext4 filesystem being mounted at /mnt/usb4 supports timestamps until 2038 (0x7fffffff)
Jan 27 22:25:52 linux-box kernel: [22235749.227267] XFS (sdl3): Unmounting Filesystem
Jan 27 22:26:16 linux-box kernel: [22235773.926657] usb 1-5: USB disconnect, device number 36
```

marcone, 2022-01-28:
> It is very likely I can test this in about 10 days.
> 
> According to https://raspberrypi.stackexchange.com/questions/61723/raspberry-pi-3-and-raspbian-jessie-how-to-run-fsck-at-boot#61739 you also need `fsck.repair=yes`

Yes, I had both: `fsck.mode=force fsck.repair=yes`, but while that did increase the time to boot (so it did _something_), there were no fsck-related messages in the kernel log, and the "Last checked" field that `tune2fs -l` prints was unchanged.

marcone, 2022-01-28:
> I suppose the kernel parameters are just to force an fsck on the root file system? For other file systems set `Maximum mount count` to 1 and in `/etc/fstab` set the last parameter to 2. See https://linuxconfig.org/how-to-force-fsck-to-check-filesystem-after-system-reboot-on-linux

I tried this first, and it appears to have zero effect.

RichieB2B, 2022-01-28:
I have set the `Maximum mount count` to 1  and the `Check interval` to 1 second. 
```
root@teslausb:~#  tune2fs -i 1s /dev/mmcblk0p4
tune2fs 1.44.5 (15-Dec-2018)
Setting interval between checks to 1 seconds
```
This does trigger the kernel to notice the need for an `fsck`:
```
[   21.119298] EXT4-fs (mmcblk0p4): warning: maximal mount count reached, running e2fsck is recommended
[   21.125028] EXT4-fs (mmcblk0p4): mounted filesystem with ordered data mode. Opts: (null)
[   21.125119] ext4 filesystem being mounted at /mutable supports timestamps until 2038 (0x7fffffff)
```
It does however not seem to perform one:
```
Filesystem created:       Sat Jan 15 19:22:06 2022
Last mount time:          Thu Feb 14 11:12:15 2019
Last write time:          Thu Feb 14 11:12:15 2019
Mount count:              63
Maximum mount count:      1
Last checked:             Thu Jan 27 22:22:15 2022
Check interval:           1 (0:00:01)
Next check after:         Thu Jan 27 22:22:16 2022
```

RichieB2B, 2022-01-28:
This RedHat bug gave me some pointers: https://bugzilla.redhat.com/show_bug.cgi?id=1105877
```
root@teslausb:~# journalctl SYSLOG_IDENTIFIER=systemd-fsck -b
-- Logs begin at Thu 2019-02-14 11:12:02 CET, end at Fri 2022-01-28 16:43:11 CET. --
Feb 14 11:12:04 teslausb systemd-fsck[90]: Please pass 'fsck.mode=skip' rather than 'fastboot' on the kernel command line.
Feb 14 11:12:15 teslausb systemd-fsck[180]: Please pass 'fsck.mode=skip' rather than 'fastboot' on the kernel command line.
Feb 14 11:12:15 teslausb systemd-fsck[181]: Please pass 'fsck.mode=skip' rather than 'fastboot' on the kernel command line.
Feb 14 11:12:15 teslausb systemd-fsck[183]: Please pass 'fsck.mode=skip' rather than 'fastboot' on the kernel command line.
```
So in `/boot/cmdline.txt` I replaced `fastboot` with `fsck.mode=auto` and `fsck.repair=yes` was already there. 
Now it is checked:
```
Filesystem created:       Sat Jan 15 19:22:06 2022
Last mount time:          Thu Feb 14 11:12:30 2019
Last write time:          Thu Feb 14 11:12:30 2019
Mount count:              1
Maximum mount count:      1
Last checked:             Thu Feb 14 11:12:26 2019
Check interval:           1 (0:00:01)
```
I changed the `Check interval` back to 0 and it still works:
```
Filesystem created:       Sat Jan 15 19:22:06 2022
Last mount time:          Thu Feb 14 11:12:20 2019
Last write time:          Thu Feb 14 11:12:20 2019
Mount count:              1
Maximum mount count:      1
Last checked:             Thu Feb 14 11:12:16 2019
Check interval:           0 (<none>)
root@teslausb:~#  journalctl SYSLOG_IDENTIFIER=systemd-fsck -b
-- Logs begin at Thu 2019-02-14 11:12:02 CET, end at Fri 2022-01-28 16:59:06 CET. --
Feb 14 11:12:03 teslausb systemd-fsck[73]: e2fsck 1.44.5 (15-Dec-2018)
Feb 14 11:12:03 teslausb systemd-fsck[73]: Superblock last mount time is in the future.
Feb 14 11:12:03 teslausb systemd-fsck[73]:         (by less than a day, probably due to the hardware clock being incorrectly set)
Feb 14 11:12:03 teslausb systemd-fsck[73]: Superblock last write time is in the future.
Feb 14 11:12:03 teslausb systemd-fsck[73]:         (by less than a day, probably due to the hardware clock being incorrectly set)
Feb 14 11:12:03 teslausb systemd-fsck[73]: rootfs: clean, 47628/131328 files, 346131/524288 blocks
Feb 14 11:12:16 teslausb systemd-fsck[185]: /usr/sbin/fsck.xfs: XFS file system.
Feb 14 11:12:16 teslausb systemd-fsck[184]: e2fsck 1.44.5 (15-Dec-2018)
Feb 14 11:12:16 teslausb systemd-fsck[184]: Superblock last mount time is in the future.
Feb 14 11:12:16 teslausb systemd-fsck[184]:         (by less than a day, probably due to the hardware clock being incorrectly set)
Feb 14 11:12:16 teslausb systemd-fsck[184]: Superblock last write time (Fri Jan 28 16:57:46 2022,
Feb 14 11:12:16 teslausb systemd-fsck[184]:         now = Thu Feb 14 11:12:16 2019) is in the future.
Feb 14 11:12:16 teslausb systemd-fsck[184]: Fix? yes
Feb 14 11:12:16 teslausb systemd-fsck[184]: Pass 1: Checking inodes, blocks, and sizes
Feb 14 11:12:18 teslausb systemd-fsck[186]: fsck.fat 4.1 (2017-01-24)
Feb 14 11:12:18 teslausb systemd-fsck[186]: /dev/mmcblk0p1: 406 files, 101787/504090 clusters
Feb 14 11:12:19 teslausb systemd-fsck[184]: Pass 2: Checking directory structure
Feb 14 11:12:20 teslausb systemd-fsck[184]: Pass 3: Checking directory connectivity
Feb 14 11:12:20 teslausb systemd-fsck[184]: Pass 4: Checking reference counts
Feb 14 11:12:20 teslausb systemd-fsck[184]: Pass 5: Checking group summary information
Feb 14 11:12:20 teslausb systemd-fsck[184]: mutable: 2546/12272 files (0.7% non-contiguous), 12508/102400 blocks
```
So it seems `fastboot` disables running `fsck` which is pretty bad IMHO. Also: during boot my Pi seems to think it is `Thu Feb 14 11:12:15 2019`

marcone, 2022-01-28:
> Also: during boot my Pi seems to think it is `Thu Feb 14 11:12:00 2019`

Oh, that's a good point. The Pi doesn't have a hardware clock, so even if it updates the "Last checked" field during boot, it will always be wrong.

The "fastboot" option skipping fsck might be working as intended (though I'm not sure, since it isn't listed on the [official list of kernel parameters](https://www.kernel.org/doc/html/v4.14/admin-guide/kernel-parameters.html)), but the fact that teslausb adds the fastboot option is a holdover from the original teslausb project. If it's preventing fsck from running it might be better to remove it.

RichieB2B, 2022-01-28:
Yes, it sure is preventing `fsck` and probably the root cause of #654

marcone, 2022-01-29:
https://github.com/marcone/teslausb/commit/9281f720cbbf8bc7e60b7fc462582c314af6bd31 removes the fastboot parameter and changes the mount count setting for rootfs and mutable

closed 2022-01-29 by marcone