marcone/teslausb #720: Raspberry pi zero 2 w

MartinSymphonyIT, 2022-02-03:
### Describe the problem

Having issues with NTP working, after installation date and is 16 Nov 2021. i have checked the other solutions and non of them perminately resolve the problem.

i was getting pool dns errors

I have however come across a solution i think needs looking into more, and thats if you edit
/etc/systemd/system/ntp.service

and change PrivateTmp=true to PrivateTmp=false

has something to do with the ntp service not liking to run in readonly mode.

after rebooting ntp sync'd straight away no problem. and no further dns pool errors in the log.

### Device

Raspberry Pi Zero 2 W

### Car Model

Model 3

### USB connection

Glove box

marcone, 2022-02-03:
> /etc/systemd/system/ntp.service

This file does not exist on a standard TeslaUSB install using the prebuilt image (it is in /lib, not in /etc). What OS version did you try to install TeslaUSB on?

marcone, 2022-02-03:
What's the actual problem you were having? The time/date being wrong shouldn't normally be an issue, since it is set with `sntp` before archiving.
(and if your installs are anything like mine, those "error resolving pool" messages in the log only happen while away from wifi, and will continue to happen while away from wifi, even with the PrivateTmp change)

MartinSymphonyIT, 2022-02-03:
i was getting the error resolving pool  messages whilst on wifi, and the time would not automatically set, it meant i could run the rclone install, change the password, install update etc. and wouldnt perform an initial archive becouse the time and date were so far out.

marcone, 2022-02-03:
archiving doesn't depend on the ntp service though. It uses sntp, which queries a given ntp server directly (teslausb uses time.google.com)

MartinSymphonyIT, 2022-02-03:
strange then, i was having problems perhaps more probably at the setup stage but changing that file just made life easier, and i don't like seeing errors in log files.

thought it might be of some use to others.

marcone, 2022-02-03:
OK. A diagnostics file might have provided more info.
For what it's worth, the "official" way to make changes like this is not to manually edit the service definition file, but to use an override. You can use `systemctl edit ntp` to create one.

closed 2022-02-03 by marcone