marcone/teslausb #723: 2032.4.5.21 teslausb no longer recognized as flash drive

discojon, 2022-04-13:
### Describe the problem

My model 3 just updated. The raspberry pi 0 is flashing green rapidly and not recognized as a flash drive.

### Device

Raspberry Pi Zero W

### Car Model

Model 3

### USB connection

Center console

discojon, 2022-04-13:
Edit, it may just be in an archive loop.

rmyadsk, 2022-04-13:
If you're able to SSH to your Pi, enter the following at the terminal: 

tail -f /mutable/archiveloop.log

This will tell you what teslausb thinks it's doing in realtime (press CTRL-C to exit out of the screen dump and return to a terminal prompt).

Also in /tmp you may find useful log files. For example, I use rsync to transfer files, so this is useful for me to track archive progress:

tail -f /tmp/rsynclog

discojon, 2022-04-13:
After sitting all night it was working this morning.  I have a 400GB sd card in it and I believe something triggered a full backup that took a while.

closed 2022-04-21 by discojon