marcone/teslausb #724: Tesla doesn't recognize USB drive anymore, I can connect via wifi to teslausb, solid green light

Jodox1976, 2022-04-17:
### Describe the problem

Starting yesterday, my teslausb on my Pi Zero W is not recognized as USB stick anymore - or the tesla says "not enough space". 

I've one - with a combination of car and Pi resets managed that the car recognized it again. I then deleted the dashcam clips via the car menu. Now, teslausb is not being recognized anymore. 

When I connect it via USB to the PC, the music and cam drive are not being mounted and the LED shows a solid green light (not blinking) 

As I'm at my in-law's my possibilities are a bit limited. I can connect to the Pi via wifi with my smartphone. I also have Putty on my smartphone. 

In the Archiveloop log it says: Warning: low space for new snapshots, but only one snapshot exists. Please use a lager storage medium or reduce CAM_SIZE

Can someone give me a hint what I can try to get teslausb running again?

### Device

Raspberry Pi Zero W

### Car Model

Model 3

### USB connection

Center console

### Logs

====== summary ======
hardware: Raspberry Pi Zero W Rev 1.1
OS: Raspbian GNU/Linux 10 (buster)
CIFS archiving selected
2 snapshots mounted
USB state (158.9 seconds ago): unknown:
====== disk / images ======
Model: SD SF128 (sd/mmc)
Disk /dev/mmcblk0: 128GB
Sector size (logical/physical): 512B/512B
Partition Table: msdos
Disk Flags: 
Number  Start   End     Size    Type     File system  Flags
 1      4194kB  273MB   268MB   primary  fat32        lba
 2      273MB   2181MB  1908MB  primary  ext4
 3      2181MB  128GB   126GB   primary  xfs
 4      128GB   128GB   105MB   primary  ext4
cam disk image has 17949 extents
Model:  (file)
Disk /backingfiles/cam_disk.bin: 75.2GB
Sector size (logical/physical): 512B/512B
Partition Table: msdos
Disk Flags: 
Number  Start   End     Size    Type     File system  Flags
 1      1049kB  75.2GB  75.2GB  primary  fat32        lba
music disk image has 18 extents
Model:  (file)
Disk /backingfiles/music_disk.bin: 42.9GB
Sector size (logical/physical): 512B/512B
Partition Table: msdos
Disk Flags: 
Number  Start   End     Size    Type     File system  Flags
 1      1049kB  42.9GB  42.9GB  primary  fat32        lba
====== network ======
ap0: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet 192.168.66.1  netmask 255.255.255.0  broadcast 192.168.66.255
        inet6 fe80::ba27:ebff:fe07:3a50  prefixlen 64  scopeid 0x20<link>
        ether b8:27:eb:07:3a:50  txqueuelen 1000  (Ethernet)
        RX packets 1741  bytes 111243 (108.6 KiB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 2761  bytes 3527131 (3.3 MiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0
lo: flags=73<UP,LOOPBACK,RUNNING>  mtu 65536
        inet 127.0.0.1  netmask 255.0.0.0
        inet6 ::1  prefixlen 128  scopeid 0x10<host>
        loop  txqueuelen 1000  (Local Loopback)
        RX packets 100  bytes 6784 (6.6 KiB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 100  bytes 6784 (6.6 KiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0
wlan0: flags=4099<UP,BROADCAST,MULTICAST>  mtu 1500
        ether b8:27:eb:07:3a:50  txqueuelen 1000  (Ethernet)
        RX packets 0  bytes 0 (0.0 B)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 0  bytes 0 (0.0 B)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

marcone, 2022-04-17:
The solid green led indicates the archive loop didn't run or crashed. The full diagnostics or archiveloop.log might show why. If you can still connect to the web interface you should be able to download those from there.

Jodox1976, 2022-04-17:
Thanks for the reply, @marcone!
Here's the archiveloop.log
[archiveloop.log](https://github.com/marcone/teslausb/files/8501950/archiveloop.log)

Jodox1976, 2022-04-17:
I managed to connect via Putty on my smartphone, now. Just don't know any usefull commands to ask teslausb to perform ;-)

marcone, 2022-04-17:
Did you get the diagnostics file you originally pasted here from clicking the download button in the web interface, or did you copy/paste the page?
If you downloaded it, that's surprising, since it should include a lot more.
Your archiveloop.log also seems to be missing at least the last few lines, but otherwise shows teslausb is connecting the drive to the car, so it *should* be working, and I would expect the LED to be flashing. If you can ssh into the Pi could run `dmesg` and see if there are any I/O errors on the sd card that might be preventing things from working correctly. You could also run `/root/bin/setup-teslausb diagnose` manually and see if/why it ends early.

Jodox1976, 2022-04-17:
Here is the dmesg return. 

[   13.595016] bcm2835-isp bcm2835-isp: Register output node 0 with media con
troller
[   13.595056] bcm2835-isp bcm2835-isp: Register capture node 1 with media co
ntroller
[   13.595084] bcm2835-isp bcm2835-isp: Register capture node 2 with media co
ntroller
[   13.595111] bcm2835-isp bcm2835-isp: Register capture node 3 with media co
ntroller
[   13.608219] bcm2835-isp bcm2835-isp: Loaded V4L2 bcm2835-isp
[   13.612156] bcm2835_codec: module is from the staging directory, the quali
ty is unknown, you have been warned.
[   13.672707] bcm2835-codec bcm2835-codec: Device registered as /dev/video10
[   13.672796] bcm2835-codec bcm2835-codec: Loaded V4L2 decode
[   13.706243] bcm2835-codec bcm2835-codec: Device registered as /dev/video11
[   13.706327] bcm2835-codec bcm2835-codec: Loaded V4L2 encode
[   13.717813] bcm2835-codec bcm2835-codec: Device registered as /dev/video12
[   13.717896] bcm2835-codec bcm2835-codec: Loaded V4L2 isp
[   15.022060] cfg80211: Loading compiled-in X.509 certificates for regulator
y database
[   15.567268] cfg80211: Loaded X.509 cert 'sforshee: 00b28ddf47aef9cea7'
[   16.202940] brcmfmac: F1 signature read @0x18000000=0x1541a9a6
[   16.232856] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43430-sdi
o for chip BCM43430/1
[   16.233327] usbcore: registered new interface driver brcmfmac
[   16.265754] brcmfmac mmc1:0001:1: Direct firmware load for brcm/brcmfmac43
430-sdio.raspberrypi,model-zero-w.txt failed with error -2
[   16.525301] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43430-sdi
o for chip BCM43430/1
[   16.525523] brcmfmac: brcmf_c_process_clm_blob: no clm_blob available (err
=-2), device may have limited channels available
[   16.526806] brcmfmac: brcmf_c_preinit_dcmds: Firmware: BCM43430/1 wl0: Oct
 22 2019 01:59:28 version 7.45.98.94 (r723000 CY) FWID 01-3b33decd
[   20.117886] EXT4-fs (mmcblk0p4): recovery complete
[   20.120237] EXT4-fs (mmcblk0p4): mounted filesystem with ordered data mode
. Opts: (null)
[   20.120331] ext4 filesystem being mounted at /mutable supports timestamps
until 2038 (0x7fffffff)
[   21.276748] SGI XFS with ACLs, security attributes, realtime, no debug ena
bled
[   21.381080] XFS (mmcblk0p3): Mounting V5 Filesystem
[   21.719900] XFS (mmcblk0p3): Starting recovery (logdev: internal)
[   21.795920] XFS (mmcblk0p3): Ending recovery (logdev: internal)
[   21.816140] xfs filesystem being mounted at /backingfiles supports timesta
mps until 2038 (0x7fffffff)
[   21.956996] fuse: init (API version 7.31)
[   25.210493] random: crng init done
[   25.210515] random: 7 urandom warning(s) missed due to ratelimiting
[   27.046000] IPv6: ADDRCONF(NETDEV_CHANGE): ap0: link becomes ready
[   27.349540] brcmfmac: brcmf_cfg80211_set_power_mgmt: power save enabled
[   38.608556]  loop0: p1
[ 2093.707659] loop: Write error at byte offset 1081344, length 4096.
[ 2093.708182] blk_update_request: I/O error, dev loop0, sector 2112 op 0x1:(
WRITE) flags 0x100000 phys_seg 1 prio class 0
[ 2093.708209] Buffer I/O error on dev loop0p1, logical block 8, lost async p
age write
[ 2093.709842] loop: Write error at byte offset 1540096, length 4096.
[ 2093.709969] loop: Write error at byte offset 2912256, length 4096.
[ 2093.710082] loop: Write error at byte offset 7634944, length 4096.
[ 2093.710191] loop: Write error at byte offset 9134080, length 4096.
[ 2093.710499] blk_update_request: I/O error, dev loop0, sector 3008 op 0x1:(
WRITE) flags 0x100000 phys_seg 1 prio class 0
[ 2093.710525] Buffer I/O error on dev loop0p1, logical block 120, lost async
 page write
[ 2093.710581] blk_update_request: I/O error, dev loop0, sector 5688 op 0x1:(
WRITE) flags 0x100000 phys_seg 1 prio class 0
[ 2093.710597] Buffer I/O error on dev loop0p1, logical block 455, lost async
 page write
[ 2093.710631] blk_update_request: I/O error, dev loop0, sector 14912 op 0x1:
[... 19 lines cut ...]
c page write
[ 2093.714704] blk_update_request: I/O error, dev loop0, sector 23608 op 0x1:
(WRITE) flags 0x100000 phys_seg 1 prio class 0
[ 2093.714720] Buffer I/O error on dev loop0p1, logical block 2695, lost asyn
c page write
[ 2093.714753] blk_update_request: I/O error, dev loop0, sector 32832 op 0x1:
(WRITE) flags 0x100000 phys_seg 1 prio class 0
[ 2093.714768] Buffer I/O error on dev loop0p1, logical block 3848, lost asyn
c page write
[ 2093.714801] blk_update_request: I/O error, dev loop0, sector 35760 op 0x1:
(WRITE) flags 0x100000 phys_seg 1 prio class 0
[ 2093.714816] Buffer I/O error on dev loop0p1, logical block 4214, lost asyn
c page write
[ 2094.150592]  loop0: p1
[ 2102.265321] Mass Storage Function, version: 2009/09/11
[ 2102.265343] LUN: removable file: (no medium)
[ 2102.265532] LUN: removable file: /backingfiles/music_disk.bin
[ 2102.265657] LUN: removable file: /backingfiles/cam_disk.bin
[ 2102.265670] Number of LUNs=2
[ 2102.270558] g_mass_storage gadget: Mass Storage Gadget, version: 2009/09/1
1
[ 2102.270578] g_mass_storage gadget: g_mass_storage ready
[ 2102.270598] dwc2 20980000.usb: bound driver g_mass_storage
[ 2102.487125] dwc2 20980000.usb: new device is high-speed
pi@ElMo-TeslaCam:~ $

Jodox1976, 2022-04-17:
There's probably something missing on the top. I don't know how to redirect the output to a locally saved file on PuTTY.

The diagnose comman, I get "permission denied".

Jodox1976, 2022-04-17:
> Did you get the diagnostics file you originally pasted here from clicking the download button in the web interface, or did you copy/paste the page?

I copy/pasted it, but here is the downloaded one
[diagnostics-1.txt](https://github.com/marcone/teslausb/files/8502069/diagnostics-1.txt)
.

Jodox1976, 2022-04-17:
Okay, I typed `sudo -i` and then `/root/bin/setup-teslausb diagnose`, but the return didn't contain any further information.

marcone, 2022-04-17:
What does `cat /etc/fstab` say?

Jodox1976, 2022-04-17:
pi@ElMo-TeslaCam:~ $ cat /etc/fstab
proc            /proc           proc    defaults          0       0
PARTUUID=3f98dc8b-01  /boot           vfat    defaults,ro          0       2
PARTUUID=3f98dc8b-02  /               ext4    defaults,noatime,ro  0       1
# a swapfile is not a swap partition, no line here
#   use  dphys-swapfile swap[on|off]  for that
LABEL=backingfiles /backingfiles xfs auto,rw,noatime 0 2
LABEL=mutable /mutable ext4 auto,rw 0 2
tmpfs /var/run/samba tmpfs nodev,nosuid 0 0
tmpfs /var/cache/samba tmpfs nodev,nosuid 0 0
tmpfs /var/tmp tmpfs nodev,nosuid 0 0
tmpfs /tmp    tmpfs nodev,nosuid 0 0
tmpfs /var/spool tmpfs nodev,nosuid 0 0
/backingfiles/cam_disk.bin /mnt/cam teslausb noauto 0 0
/backingfiles/music_disk.bin /mnt/music teslausb noauto 0 0
//192.168.150.21/TeslaCam /mnt/archive cifs noauto,credentials=/root/.teslaCa
mArchiveCredentials,iocharset=utf8,file_mode=0777,dir_mode=0777,, 0
//192.168.150.21/TeslaMusic /mnt/musicarchive cifs noauto,credentials=/root/.
teslaCamArchiveCredentials,iocharset=utf8,file_mode=0777,dir_mode=0777,, 0
tmpfs /var/log/nginx tmpfs nodev,nosuid 0 0
tmpfs /var/lib/nginx tmpfs nodev,nosuid 0 0
mount.ctts#/mutable/TeslaCam /var/www/html/TeslaCam fuse defaults,nofail,x-sy
stemd.requires=/mutable 0 0
pi@ElMo-TeslaCam:~ $

marcone, 2022-04-17:
Weird, that should have been the next section in the diagnostics, after networking, but your diagnostics are cut off right after the network section

marcone, 2022-04-17:
Oh, I know why diagnostics don't work. Basically when the Pi is not associated with a Wi-Fi access point, then diagnostics will fail. I will fix that.

Jodox1976, 2022-04-17:
I've given my father-in-law's guest network the SSID of my home network, so the diagnostics would work.

Here's the complete file.
[diagnostics-2.txt](https://github.com/marcone/teslausb/files/8502296/diagnostics-2.txt)

marcone, 2022-04-17:
To be clear though: the diagnostics not working isn't why your setup isn't working. That's because of something else. You do have some I/O errors in the kernel log:
```
[ 2093.707659] loop: Write error at byte offset 1081344, length 4096.
[ 2093.708182] blk_update_request: I/O error, dev loop0, sector 2112 op 0x1:(
WRITE) flags 0x100000 phys_seg 1 prio class 0
[ 2093.708209] Buffer I/O error on dev loop0p1, logical block 8, lost async p
age write
[ 2093.709842] loop: Write error at byte offset 1540096, length 4096.
[ 2093.709969] loop: Write error at byte offset 2912256, length 4096.
```
(etc)
but they're from the loopback device, so not sure if that means the sd card is bad or not.

Jodox1976, 2022-04-17:
Since I have an access point (sort of), I'll let teslausb run an upgrade. Maybe that helps.

marcone, 2022-04-17:
Try running `journalctl -u teslausb` and see if that shows any obvious errors.

Jodox1976, 2022-04-17:
pi@ElMo-TeslaCam:~ $ journalctl -u teslausb
-- Logs begin at Thu 2019-02-14 11:12:02 CET, end at Sun 2022-04-17 22:18:00
Apr 17 21:51:21 ElMo-TeslaCam systemd[1]: Started TeslaUSB archiveloop servic
Apr 17 21:51:22 ElMo-TeslaCam bash[219]: Powering off HDMI
Apr 17 21:51:22 ElMo-TeslaCam bash[219]: /root/bin/setup-teslausb was sourced
Apr 17 21:51:24 ElMo-TeslaCam bash[219]: mkdir: cannot create directory ‘/bac
Apr 17 21:51:31 ElMo-TeslaCam bash[219]: /root/bin/make_snapshot.sh: line 187
Apr 17 21:51:32 ElMo-TeslaCam bash[219]: diff: /backingfiles/snapshots/snap-0
Apr 17 21:51:32 ElMo-TeslaCam bash[219]: umount: /backingfiles/snapshots/snap

Jodox1976, 2022-04-17:
Well I guess, I'll buy a high endurance SD card and do a new setup when I return home.

marcone, 2022-04-17:
This looks suspicious:

> Apr 17 21:51:24 ElMo-TeslaCam bash[219]: mkdir: cannot create directory ‘/bac

That looks like it might be from https://github.com/marcone/teslausb/blob/main-dev/run/make_snapshot.sh#L11, however that normally shouldn't fail. You could try running `sudo mkdir -p /backingfiles/snapshots` manually and see if it still prints an error. It could be there was an error on the backingfiles filesystem which caused it to be mounted read-only.

Jodox1976, 2022-04-18:
No errors, but no improvement, also.
Thanks a lot for your efforts, but I'll have to get a new SD card, I guess.

marcone, 2022-04-19:
The solid green light is puzzling, since archiveloop is running so it should be blinking normally, but I guess anything's possible with a bad sd card.

closed 2022-04-19 by marcone

Jodox1976, 2022-04-24:
Hello @marcone,

I could use some more help. I've bought a new 256GB SD card used Etcher to copy the image to it, copied the changed conf-file to the boot drive, but the pi still just shows the solid green light. 

When I pull the SD card an put it into my PC, there are no log files.

Is there anything I can try?

marcone, 2022-04-24:
Can you still ssh into the Pi? If so I would start by figuring out why it's not connecting to your network. It seems likely there is a mistake in the network config. Start by looking at /etc/wpa-supplicant/wpa-supplicant.conf and make sure it is correct. The "iw" tools (iwlist, iwconfig, etc) might also help here.

Jodox1976, 2022-04-24:
I can SSH into the Pi with the old SD card, the one that's possibly defective.
If I use the new SD card, I can't connect.

Jodox1976, 2022-04-25:
I do have to uncomment the exFAT part for the 256GB SD card, right?

marcone, 2022-04-25:
> I do have to uncomment the exFAT part for the 256GB SD card, right?

No, why do you think that? You can certainly use it if you want, but you don't _have_ to, and I [recommend you don't](https://github.com/marcone/teslausb/blob/c26f80624357a9106827918837d96eb3714723f9/pi-gen-sources/00-teslausb-tweaks/files/teslausb_setup_variables.conf.sample#L76).

Jodox1976, 2022-04-25:
I'm just trying to find the misconfiguration in my config-file that prevents the headless setup from working with the new card. Since it's way bigger than the old one, the images are bigger, too. So I thought, I might have to use ex FAT.

Jodox1976, 2022-04-26:
Whenever I want to try something new in the conf-file, do I have to do the Etcher part (deleting all partitions on the SD card, copying the image with Etcher) all over again or is it sufficient to just drop the new conf-file into boot?

Jodox1976, 2022-04-26:
Does the conf-file have to be coded in a certain way? ANSI? I used UTF-8.

Jodox1976, 2022-05-01:
Okay, I've managed to get a little further now. 

I've used PuTTY and the less command to look at the config-file in the root of the old (defective) SD card and compared it to my new config file that was not working. The only difference I noticed was that "export musicsharename=" in the old config file was "export MUSIC_SHARE_NAME=" in the new config file. I changed it to  "export musicsharename=" and now the Pi is connecting to Wifi with the new SD card, also.

But when I try to access its web server, I only get a greeting-page 

Welcome to nginx!

If you see this page, the nginx web server is successfully installed and working. Further configuration is required.

For online documentation and support please refer to [nginx.org](http://nginx.org/).
Commercial support is available at [nginx.com](http://nginx.com/).

Thank you for using nginx.

Jodox1976, 2022-05-01:
I ran `bin/setup-teslausb upgrade` on PuTTY to see what's going on and it finished with `upgrade finished`. 
The headless setup log reads `all done` at the end.
The LED is flashing rapidly now.

Jodox1976, 2022-05-02:
Now, everything is working like it should, @marcone.  Looks like the the wrong "export MUSIC_SHARE_NAME=" caused the headless setup to halt.

rmyadsk, 2022-05-02:
Glad you got it resolved @Jodox1976. One of the idiosyncrasies of this program is that it can't check for an updated [teslausb_setup_variables.conf](https://github.com/marcone/teslausb/blob/main-dev/pi-gen-sources/00-teslausb-tweaks/files/teslausb_setup_variables.conf.sample) - so unless the image date is very recent, the file contained within the image is likely to be out of date.

Jodox1976, 2022-05-02:
The wrong line is also in here! 
https://github.com/marcone/teslausb/blob/c26f80624357a9106827918837d96eb3714723f9/pi-gen-sources/00-teslausb-tweaks/files/teslausb_setup_variables.conf.sample#L86

marcone, 2022-05-02:
That line isn't wrong. The setup script will accept either `musicsharename` or `MUSIC_SHARE_NAME`. In fact when it sees `musicsharename` it will instead define `MUSIC_SHARE_NAME` and then undefine `musicsharename`.
I suspect that there was a different problem in your config file, that you just happened to fix when you changed that line.