marcone/teslausb #727: Unsecured wifi unsupported

chrsturbo, 2022-05-03:
### Describe the problem

With no wifipass or encryption rc.local bails out - I removed the test for wifipass and hand edited the wpa file for no encryption - its working great now.  Not sure if this would be a bug or feature:)

### Device

Raspberry Pi Zero W

### Car Model

Model S

### USB connection

Glove box

chrsturbo, 2022-05-05:
I can make a diff to rc.local if that helps

marcone, 2022-05-06:
Having an unsecured network at home (where you'd normally be setting up teslausb) seems like a very niche use case.

If you did want to use an open wifi network for setup, you could modify the [wpa_supplicant.conf.sample](https://github.com/marcone/teslausb/blob/main-dev/pi-gen-sources/00-teslausb-tweaks/files/wpa_supplicant.conf.sample) file in the boot partition of the prebuilt image, since that is used as a template. Taking out the `psk` line, changing `key_mgmt` to NONE, and specifying a dummy password in the config file should do it, I think.

chrsturbo, 2022-05-06:
Yeap thats what I did as I said probably a feature not a bug thanks for looking

closed 2022-05-06 by chrsturbo