marcone/teslausb #728: rsync error: received SIGINT, SIGTERM, or SIGHUP (code 20)...

miles267, 2022-05-11:
### Describe the problem

Am receiving the following 2 error rows in my archiveloop.log when attempting to sync RecentClips to smb share.  Upon encountering these errors, the teslausb stops syncing and the car will be permitted to sleep:

SentryClips/2022-05-10_12-26-32/2022-05-10_12-15-50-back.mp4
SentryClips/2022-05-10_12-26-32/2022-05-10_12-15-50-front.mp4
SentryClips/2022-05-10_12-26-32/2022-05-10_12-15-50-left_repeater.mp4
SentryClips/2022-05-10_12-26-32/2022-05-10_12-15-50-right_repeater.mp4
SentryClips/2022-05-10_12-26-32/2022-05-10_12-16-50-back.mp4
**rsync error: received SIGINT, SIGTERM, or SIGHUP (code 20) at io.c(503) [generator=v3.2.3]
rsync error: received SIGINT, SIGTERM, or SIGHUP (code 20) at rsync.c(701) [sender=v3.2.3]**
Wed 11 May 09:01:46 CDT 2022: cleaning cam mount
Wed 11 May 09:01:47 CDT 2022: done cleaning cam mount
Wed 11 May 09:01:47 CDT 2022: Trimming free space in /mnt/cam, which has 169 extents
Wed 11 May 09:01:47 CDT 2022: Trim complete, image now has 169 extents
Wed 11 May 09:01:47 CDT 2022: Unmounting /mnt/cam...
Wed 11 May 09:01:47 CDT 2022: Unmounted /mnt/cam.
Wed 11 May 09:01:47 CDT 2022: Finished archiving.

### Device

Raspberry Pi Zero 2 W

### Car Model

Model 3

### USB connection

Glove box

### Logs

[teslausb-headless-setup.log](https://github.com/marcone/teslausb/files/8670695/teslausb-headless-setup.log)
[diagnostics.txt](https://github.com/marcone/teslausb/files/8670696/diagnostics.txt)
[archiveloop (1).log](https://github.com/marcone/teslausb/files/8670697/archiveloop.1.log)

marcone, 2022-05-11:
This is because of:
> Wed 11 May 09:01:45 CDT 2022: connection dead, killing archive-clips

Basically what happens is that copying the files saturates the network connection to the point that it crowds out the watchdog thread that checks whether the archive server is still reachable, and so that watchdog thread ends up killing the copy process.

miles267, 2022-05-11:
OK.  I've never seen this before on my network.  Everything is the same.  Access point, connection, RPi 0 configuration, etc.  Are there any known remedies for this?  It seems to be happening consistently and my Ri Pi0 2 is transferring at it's full 72 Mbps.  It's as if the connectivity is almost too good.

miles267, 2022-05-11:
@marcone Are there any known fixes for this? Since this only just started despite no network or Rpi config changes, should I reimage the teslausb or should I be switching to a Pi4 then in this case?

It seems to suddenly be happening consistently and my Ri Pi0 2 has near perfect connectivity, transferring at it's full 72 Mbps. It's as if the connectivity is almost too good. I've never seen this before on my network. Everything is the same. Access point, connection, RPi 0 configuration, etc.

marcone, 2022-05-11:
> Are there any known fixes for this?

You could try adding something like `--bwlimit=4000` to the rsync command in archive-clips.sh, to make rsync limit its throughput.

> It seems to suddenly be happening consistently and my Ri Pi0 2 has near perfect connectivity, transferring at it's full 72 Mbps

How did you arrive at 72 Mbps?

> Wed 11 May 09:01:46 CDT 2022: Error during archiving. Archived 276 event files in 20m51s
> ...
> 2022/05/11 09:01:45 [13893] sent 6477293485 bytes  received 8807 bytes  total size 7667622692

means it sent a little under 5 MB/s, or about 39 Mbps. There will be some additional overhead on top of that, but I'd be surprised if that overhead amounts to 33 Mbps.

miles267, 2022-05-11:
> You could try adding something like `--bwlimit=4000` to the rsync command in archive-clips.sh, to make rsync limit its throughput.
>
>
> How did you arrive at 72 Mbps?
>
>
> means it sent a little under 5 [...]

Thanks.  I'll try that.  Unifi registers the Tx and Rx rates for the teslausb RPi Zero 2 at 72 Mbps.

marcone, 2022-05-11:
> Unifi registers the Tx and Rx rates for the teslausb RPi Zero 2 at 72 Mbps.

That's the link speed for a 20 MHz 1x1 channel, not the actual throughput. The 39 Mbps actual throughput I mentioned above is  probably about as good as you can expect in a real world environment with the PiZeroW-2.
This is why I suggested using `--bwlimit=4000`, as it'll stay about 20% under what you're currently achieving. That does mean it won't be optimally using the network connection though. Ideally, the connection checker should be updated to use a different method, like looking at actual throughput through the network interface.

miles267, 2022-05-12:
If it would be possible to submit a request to have the connection checker upgraded in a future release, that would be much appreciated.

In the meantime, I've tried to place the --bwlimit=4000 in archive-clips.sh without success.  Does it go here?

while [ -n "${1+x}" ]
do
  if ! (rsync -avhRL --remove-source-files --temp-dir="$rsynctmp" --no-perms --omit-dir-times --stats \
        **--bwlimit=4000**
        --log-file=/tmp/archive-rsync-cmd.log --ignore-missing-args \
        --files-from="$2" "$1/" "$ARCHIVE_MOUNT" &> /tmp/rsynclog || [[ "$?" = "24" ]] )
  then
    cat /tmp/archive-rsync-cmd.log /tmp/rsynclog > /tmp/archive-error.log

marcone, 2022-05-12:
Yes, you can put it there. Don't forget to end the line with a `\`

miles267, 2022-05-12:
I've since re-imaged this SD card completely with the latest teslausb to determine whether perhaps a recent update (i.e. apt-get update) has caused this.  It's a peculiar issue that seems to have occurred recently without otherwise there being any other changes on my network.  Will confirm back.

miles267, 2022-05-13:
@marcone providing more feedback as I troubleshoot.  I've since reimaged same card to get it onto teslausb tested, base image.  issue still occurring.  Have added the --bwlimit=4000 only in one location of the archive-clips.sh as shown below.  Is that the correct method?  I wasn't sure whether needed it on the --log-file and --files-from rows also?  I can then play around with the limit value.

while [ -n "${1+x}" ]
do
  if ! (rsync -avhRL --bwlimit=4000 --remove-source-files --temp-dir="$rsynctmp" --no-perms --omit-dir-times --stats \
        --log-file=/tmp/archive-rsync-cmd.log --ignore-missing-args \
        --files-from="$2" "$1/" "$ARCHIVE_MOUNT" &> /tmp/rsynclog || [[ "$?" = "24" ]] )
  then
    cat /tmp/archive-rsync-cmd.log /tmp/rsynclog > /tmp/archive-error.log
    exit 1

marcone, 2022-05-13:
> I wasn't sure whether needed it on the --log-file and --files-from rows also?

No, those are all still part of the same command. Everything between the parentheses is a single command line.

miles267, 2022-05-13:
Thanks again.  Despite different trial/error attempts, I'm finding myself suddenly needing to further restrict --bwlimit=4000 to 3000 and potentially lower.  On my Unifi network I've already done a series of other things and it appears it could be connectivity related.  However once the SIGTERM occurs and teslausb disconnects from my network, it never attempts to reconnect unless it loses power and needs to reboot/reconnect.  Is there any way to have it reconnect periodically if dropped?

So far I've created a specific 2.4 ghz SSID entirely for my teslausb with legacy support enabled (802.11b), boosted the signal level on the nearby AP which is dedicated to the Pi, tried different 2.4 ghz channels (non-overlapping) and removed all interference.  It seems to improve the duration of the RPi Zero 2's connection, but eventually it seems to hit a "connection dead" before all of the teslausb clips have archived.

marcone, 2022-05-23:
I wonder if using a different queuing algorithm on the network interface would help. It looks like Raspberry Pi OS uses "pfifo_fast" by default, whereas other modern linux distributions have apparently switched to "fq_codel".
You could try adding `tc qdisc add dev wlan0 root fq_codel || true` to the teslausb config file and see if that helps. (reboot afterwards, then run `ip link show` to check that it's using fq_codel)

miles267, 2022-07-07:
> I've started experiencing this particular problem on my Pi Zero (1) just recently. I thought it was to do with the fact that the Zero 1 W is just too low spec to handle teslausb, but reading through this thread seems to indicate it has something to do with the wifi drivers.
> 
> I'd love to upgrade to a Pi 4, but it's pretty much out of stock everywhere in my country...
> 
> I'll give the `bwlimit` flag a go and hope it will stabilise things for the time being.

Hi @bt in my case, it proved to be caused when I changed from VHT40 to VHT80 channels on my home network for 5 ghz.  Although the RPi Zero 2 W and RPi Zero don't support 5 Ghz it still seemed to be the culprit despite various troubleshooting methods.  Finally I just reverted back to VHT40 channels on my Unifi 5 ghz SSIDs.

lemkeant, 2022-12-08:
I seem to be having the same issue, as Im receiving the same error, on a Pi Zero 2. I do have a Unifi setup, but I dont have any 80 mhz channels on my 5ghz network at all. So I can eliminate that option. 

In order to edit the archive-clips.sh file, how do I do that post install? Where is it? Sorry, a little on the rookie side when it comes to linux and pi's. Assuming I can run the nano command to edit it if I need to ssh? I know it's been 6 months, no dirver update in sight? Happy to help test something if there is.

Thanks

lemkeant, 2022-12-23:
Ok, missed the instructions about running `bin/remountfs_rw` to make an edit to a file. Did that, added the `bwlimit `command, still no luck. Everything works ok except the archiving back to my server. It's able to write the file to the .teslausbtmp folder that it creates, but from there is not able to move it to the correct subfolder. Im lost

THX723, 2025-04-26:
I started to notice this error myself on an RPi4. Only started to do this recently.

DennisGarvey, 2025-05-08:
I also have this issue

darkmagi2, 2025-05-08:
Just started having this issue as well. Pi Zero 2 W + Unraid Share + Unifi network.