marcone/teslausb #733: teslausb daemon.err ntpd[441]: error resolving pool 3.debian.pool.ntp.org: Temporary failure in name resolution (-3)

aaronac8, 2022-07-01:
### Describe the problem

I have the following error:
 teslausb daemon.err ntpd[441]: error resolving pool 3.debian.pool.ntp.org: Temporary failure in name resolution (-3)

How do I resolve this error, in simple terms, as I am a newbie?

Thank you

### Device

Raspberry Pi Zero W

### Car Model

Model Y

### USB connection

Center console

### Logs

[diagnostics.txt](https://github.com/marcone/teslausb/files/9028259/diagnostics.txt)

marcone, 2022-07-01:
That's just a message in a log file, which it will log when it's not connected to the network. Are you having any actual problems?

aaronac8, 2022-07-01:
Raspberry pi (Tesla USB) in car was not found for a full day even with a
soft and hard reset of the car.

marcone, 2022-07-01:
Which day? (the diagnostics covers multiple days, so it would be helpful if to know which day something didn't work)

The diagnostics don't show anything wrong at first glance, except for your wifi being pretty slow. I'm guessing your car is parked far away from your access point or something.

aaronac8, 2022-07-01:
June 30, Thursday

marcone, 2022-07-02:
From the log it looks like the Pi was simply without power for most of that day.
It booted up on June 28, around 22:45:
```
Tue 28 Jun 22:17:03 EDT 2022: Starting archiveloop at 24.75 seconds uptime...
...
Tue 28 Jun 22:18:54 EDT 2022: Trying to set time...
Tue 28 Jun 22:47:33 EDT 2022: Time adjusted by 1719.692488 seconds after 0.310000 seconds
```
it then ran continuously until the evening of June 29, when it apparently lost power some time between 22:17 and 22:50.

When it next booted up, it was the late evening of June 30, just after 23:00:
```
Wed 29 Jun 22:17:03 EDT 2022: Starting archiveloop at 25.86 seconds uptime...
...
Wed 29 Jun 22:19:00 EDT 2022: Trying to set time...
Thu 30 Jun 23:05:53 EDT 2022: Time adjusted by 89212.759517 seconds after 0.350000 seconds
```

What brand/model sd card are you using by the way? Some of the operations I see in the log are much slower than on my own setup (a Raspberry Pi Zero W with a Sandisk Ultra 256 GB sd card)

aaronac8, 2022-07-02:
SanDisk 256GB High Endurance Video microSDXC Card with Adapter for Dash Cam
and Home Monitoring systems - C10, U3, V30, 4K UHD, Micro SD Card -
SDSQQNR-256G-GN6IA https://a.co/d/98XdBJc

Don't know why it would lose power.  It is always plugged into my center
console USB port with sentry mode on at all times.

aaronac8, 2022-07-02:
Got in my car just now (July 2, 7:30 am). Dashcam unavailable....rebooted
Teslausb and still unavailable

====== summary ======
hardware: Raspberry Pi Zero W Rev 1.1
OS: Raspbian GNU/Linux 10 (buster)
CIFS archiving selected
lun0 connected, from file /backingfiles/cam_disk.bin
lun1 connected, from file /backingfiles/music_disk.bin
2 snapshots mounted
Gadget state: not attached. 237.1 seconds ago: mass storage ready, but not connected to host (check cable)
====== disk / images ======
Model: SD SH256 (sd/mmc)
Disk /dev/mmcblk0: 256GB
Sector size (logical/physical): 512B/512B
Partition Table: msdos
Disk Flags: 
Number  Start   End     Size    Type     File system  Flags
 1      4194kB  273MB   268MB   primary  fat32        lba
 2      273MB   2261MB  1988MB  primary  ext4
 3      2261MB  256GB   254GB   primary  xfs
 4      256GB   256GB   105MB   primary  ext4
cam disk image has 3841 extents
Model:  (file)
Disk /backingfiles/cam_disk.bin: 68.7GB
Sector size (logical/physical): 512B/512B
Partition Table: msdos
Disk Flags: 
Number  Start   End     Size    Type     File system  Flags
 1      1049kB  68.7GB  68.7GB  primary  fat32        lba
music disk image has 171 extents
Model:  (file)
Disk /backingfiles/music_disk.bin: 53.7GB
Sector size (logical/physical): 512B/512B
Partition Table: msdos
Disk Flags: 
Number  Start   End     Size    Type     File system  Flags
 1      1049kB  53.7GB  53.7GB  primary  fat32        lba
Filesystem      Size  Used Avail Use% Mounted on
/dev/mmcblk0p1  253M   49M  204M  20% /boot
/dev/root       1.8G  1.2G  588M  66% /
/dev/mmcblk0p3  236G  192G   45G  82% /backingfiles
/dev/mmcblk0p4   93M   17M   70M  19% /mutable
====== network ======
lo: flags=73<UP,LOOPBACK,RUNNING>  mtu 65536
        inet 127.0.0.1  netmask 255.0.0.0
        inet6 ::1  prefixlen 128  scopeid 0x10<host>
        loop  txqueuelen 1000  (Local Loopback)
        RX packets 224  bytes 18144 (17.7 KiB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 224  bytes 18144 (17.7 KiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0
wlan0: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet 192.168.1.211  netmask 255.255.255.0  broadcast 192.168.1.255
        inet6 fe80::735e:d276:5126:5413  prefixlen 64  scopeid 0x20<link>
        ether b8:27:eb:a6:3e:90  txqueuelen 1000  (Ethernet)
        RX packets 7770  bytes 1967248 (1.8 MiB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 11574  bytes 13708154 (13.0 MiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0
          Link Quality=42/70  Signal level=-68 dBm  
====== fstab ======
proc            /proc           proc    defaults          0       0
PARTUUID=27cf9d8e-01  /boot           vfat    defaults,ro          0       2
PARTUUID=27cf9d8e-02  /               ext4    defaults,noatime,ro  0       1
# a swapfile is not a swap partition, no line here
#   use  dphys-swapfile swap[on|off]  for that
LABEL=backingfiles /backingfiles xfs auto,rw,noatime 0 2
LABEL=mutable /mutable ext4 auto,rw 0 2
tmpfs /var/log tmpfs nodev,nosuid 0 0
[... 547 lines cut ...]
Wed 14 Apr 17:27:46 EDT 2021 : verify-and-configure-archive: The archive share is mountable using: mount -t cifs '//192.168.1.103/TeslaMusic' '/tmp/archivetestmount' -o 'credentials=/tmp/teslaCamArchiveCredentials,iocharset=utf8,file_mode=0777,dir_mode=0777,,'
Wed 14 Apr 17:27:46 EDT 2021 : verify-and-configure-archive: Configuring the archive...
Wed 14 Apr 17:27:46 EDT 2021 : verify-and-configure-archive: Configured the archive.
Wed 14 Apr 17:27:52 EDT 2021 : curl -o /tmp/configure-automount.sh https://raw.githubusercontent.com/marcone/teslausb/7cfe05e9f1bf30c63ef2a0a6a1d9687425521464/setup/pi/configure-automount.sh
Wed 14 Apr 17:27:53 EDT 2021 : Downloaded /tmp/configure-automount.sh ...
Wed 14 Apr 17:27:58 EDT 2021 : curl -o /root/bin/auto.teslausb https://raw.githubusercontent.com/marcone/teslausb/7cfe05e9f1bf30c63ef2a0a6a1d9687425521464/run/auto.teslausb
Wed 14 Apr 17:27:59 EDT 2021 : Downloaded /root/bin/auto.teslausb ...
Wed 14 Apr 17:27:59 EDT 2021 : configure-automount: converting snapshot mountpoints to links
Wed 14 Apr 17:27:59 EDT 2021 : configure-automount: configured automount
Wed 14 Apr 17:27:59 EDT 2021 : make-root-fs-readonly: start
Wed 14 Apr 17:27:59 EDT 2021 : make-root-fs-readonly: Disabling unnecessary service...
Wed 14 Apr 17:28:05 EDT 2021 : make-root-fs-readonly: Removing unwanted packages...
Wed 14 Apr 17:28:16 EDT 2021 : make-root-fs-readonly: Installing ntp and busybox-syslogd...
Wed 14 Apr 17:28:21 EDT 2021 : make-root-fs-readonly: Configuring system...
Wed 14 Apr 17:28:22 EDT 2021 : make-root-fs-readonly: done
Wed 14 Apr 17:28:22 EDT 2021 : Skipping package upgrade.
Wed 14 Apr 17:28:28 EDT 2021 : All done.
Wed 14 Apr 17:28:29 EDT 2021 : upgrade finished
Sat 26 Jun 12:54:03 EDT 2021 : curl -L https://github.com/marcone/teslausb/archive/main-dev.tar.gz
Sat 26 Jun 12:54:04 EDT 2021 : Downloaded /tmp/20599/setup-teslausb ...
Sat 26 Jun 12:54:04 EDT 2021 : /tmp/setup-teslausb is up to date
Sat 26 Jun 12:54:04 EDT 2021 : Downloaded /tmp/20599/rc.local ...
Sat 26 Jun 12:54:04 EDT 2021 : rc.local is up to date
Sat 26 Jun 12:54:04 EDT 2021 : Updating package index files...
Sat 26 Jun 12:54:53 EDT 2021 : Verifying that the r

marcone, 2022-07-02:
Can you attach the actual diagnostics file instead of copy/pasting? (the copy/paste above is incomplete)

aaronac8, 2022-07-02:
Lost connection again on 7/2/2022 at 3pm...see attached diagnostic file

====== summary ======
hardware: Raspberry Pi Zero W Rev 1.1
OS: Raspbian GNU/Linux 10 (buster)
CIFS archiving selected
lun0 connected, from file /backingfiles/cam_disk.bin
lun1 connected, from file /backingfiles/music_disk.bin
1 snapshots mounted
Gadget state: not attached. 270.9 seconds ago: mass storage ready, but not connected to host (check cable)
====== disk / images ======
Model: SD SH256 (sd/mmc)
Disk /dev/mmcblk0: 256GB
Sector size (logical/physical): 512B/512B
Partition Table: msdos
Disk Flags: 
Number  Start   End     Size    Type     File system  Flags
 1      4194kB  273MB   268MB   primary  fat32        lba
 2      273MB   2261MB  1988MB  primary  ext4
 3      2261MB  256GB   254GB   primary  xfs
 4      256GB   256GB   105MB   primary  ext4
cam disk image has 3841 extents
Model:  (file)
Disk /backingfiles/cam_disk.bin: 68.7GB
Sector size (logical/physical): 512B/512B
Partition Table: msdos
Disk Flags: 
Number  Start   End     Size    Type     File system  Flags
 1      1049kB  68.7GB  68.7GB  primary  fat32        lba
music disk image has 171 extents
Model:  (file)
Disk /backingfiles/music_disk.bin: 53.7GB
Sector size (logical/physical): 512B/512B
Partition Table: msdos
Disk Flags: 
Number  Start   End     Size    Type     File system  Flags
 1      1049kB  53.7GB  53.7GB  primary  fat32        lba
Filesystem      Size  Used Avail Use% Mounted on
/dev/mmcblk0p1  253M   49M  204M  20% /boot
/dev/root       1.8G  1.2G  588M  66% /
/dev/mmcblk0p3  236G  192G   45G  82% /backingfiles
/dev/mmcblk0p4   93M   17M   70M  19% /mutable
====== network ======
lo: flags=73<UP,LOOPBACK,RUNNING>  mtu 65536
        inet 127.0.0.1  netmask 255.0.0.0
        inet6 ::1  prefixlen 128  scopeid 0x10<host>
        loop  txqueuelen 1000  (Local Loopback)
        RX packets 14288  bytes 1157328 (1.1 MiB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 14288  bytes 1157328 (1.1 MiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0
wlan0: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet 192.168.1.211  netmask 255.255.255.0  broadcast 192.168.1.255
        inet6 fe80::735e:d276:5126:5413  prefixlen 64  scopeid 0x20<link>
        ether b8:27:eb:a6:3e:90  txqueuelen 1000  (Ethernet)
        RX packets 147617  bytes 21704071 (20.6 MiB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 79629  bytes 29675136 (28.3 MiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0
          Link Quality=44/70  Signal level=-66 dBm  
====== fstab ======
proc            /proc           proc    defaults          0       0
PARTUUID=27cf9d8e-01  /boot           vfat    defaults,ro          0       2
PARTUUID=27cf9d8e-02  /               ext4    defaults,noatime,ro  0       1
# a swapfile is not a swap partition, no line here
#   use  dphys-swapfile swap[on|off]  for that
LABEL=backingfiles /backingfiles xfs auto,rw,noatime 0 2
LABEL=mutable /mutable ext4 auto,rw 0 2
tmpfs /var/log tmpfs nodev,nosuid 0 0
tmpfs /var/tmp tmpfs nodev,nosuid 0 0
[... 558 lines cut ...]
Wed 14 Apr 17:28:16 EDT 2021 : make-root-fs-readonly: Installing ntp and busybox-syslogd...
Wed 14 Apr 17:28:21 EDT 2021 : make-root-fs-readonly: Configuring system...
Wed 14 Apr 17:28:22 EDT 2021 : make-root-fs-readonly: done
Wed 14 Apr 17:28:22 EDT 2021 : Skipping package upgrade.
Wed 14 Apr 17:28:28 EDT 2021 : All done.
Wed 14 Apr 17:28:29 EDT 2021 : upgrade finished
Sat 26 Jun 12:54:03 EDT 2021 : curl -L https://github.com/marcone/teslausb/archive/main-dev.tar.gz
Sat 26 Jun 12:54:04 EDT 2021 : Downloaded /tmp/20599/setup-teslausb ...
Sat 26 Jun 12:54:04 EDT 2021 : /tmp/setup-teslausb is up to date
Sat 26 Jun 12:54:04 EDT 2021 : Downloaded /tmp/20599/rc.local ...
Sat 26 Jun 12:54:04 EDT 2021 : rc.local is up to date
Sat 26 Jun 12:54:04 EDT 2021 : Updating package index files...
Sat 26 Jun 12:54:53 EDT 2021 : Verifying that the requested configuration is valid...
Sat 26 Jun 12:54:53 EDT 2021 : Downloaded /tmp/verify-configuration.sh ...
Sat 26 Jun 12:54:53 EDT 2021 : USB_DRIVE is not set. SD card will be used.
Sat 26 Jun 12:54:53 EDT 2021 : Verifying that there is sufficient space available on the MicroSD card...
Sat 26 Jun 12:54:53 EDT 2021 : There is sufficient space available.
Sat 26 Jun 12:54:57 EDT 2021 : Downloading additional setup scripts.
Sat 26 Jun 12:54:57 EDT 2021 : Downloaded /root/bin/remountfs_rw ...
Sat 26 Jun 12:54:57 EDT 2021 : Downloaded /root/bin/make_snapshot.sh ...
Sat 26 Jun 12:54:57 EDT 2021 : Downloaded /root/bin/mount_snapshot.sh ...
Sat 26 Jun 12:54:57 EDT 2021 : Downloaded /root/bin/release_snapshot.sh ...
Sat 26 Jun 12:54:58 EDT 2021 : Downloaded /root/bin/force_sync.sh ...
Sat 26 Jun 12:54:58 EDT 2021 : Downloaded /tmp/create-backingfiles-partition.sh ...
Sat 26 Jun 12:54:58 EDT 2021 : Downloaded /

marcone, 2022-07-02:
> Lost connection again on 7/2/2022 at 3pm...see attached diagnostic file

Looks like you didn't actually attach the diagnostics file, but copy/pasted it, so it's incomplete again.

aaronac8, 2022-07-02:
[diagnostics.txt](https://github.com/marcone/teslausb/files/9033779/diagnostics.txt)

marcone, 2022-07-02:
It looks you "lost connection" well before that. You mentioned earlier that the Pi is always on because you have Sentry mode always on, correct? In that case, the hourly snapshot that teslausb takes of the drive image should contain new files every time, however:
```
Sat  2 Jul 07:38:43 EDT 2022: Waiting for archive to be unreachable...
Sat  2 Jul 07:40:09 EDT 2022: Sleeping before retry...
Sat  2 Jul 07:40:10 EDT 2022: Retrying...
Sat  2 Jul 08:34:26 EDT 2022: waiting up to 90 seconds for idle interval
Sat  2 Jul 08:36:08 EDT 2022: couldn't determine idle interval
Sat  2 Jul 08:36:09 EDT 2022: taking snapshot of cam disk in /backingfiles/snapshots/snap-009843
Sat  2 Jul 08:36:22 EDT 2022: took snapshot
Sat  2 Jul 08:36:27 EDT 2022: comparing new snapshot with /backingfiles/snapshots/snap-009842/snap.bin
Sat  2 Jul 08:36:27 EDT 2022: new snapshot is identical to previous one, discarding
Sat  2 Jul 09:34:38 EDT 2022: waiting up to 90 seconds for idle interval
Sat  2 Jul 09:36:20 EDT 2022: couldn't determine idle interval
Sat  2 Jul 09:36:21 EDT 2022: taking snapshot of cam disk in /backingfiles/snapshots/snap-009843
Sat  2 Jul 09:36:34 EDT 2022: took snapshot
Sat  2 Jul 09:36:38 EDT 2022: comparing new snapshot with /backingfiles/snapshots/snap-009842/snap.bin
Sat  2 Jul 09:36:39 EDT 2022: new snapshot is identical to previous one, discarding
Sat  2 Jul 10:34:48 EDT 2022: waiting up to 90 seconds for idle interval
Sat  2 Jul 10:36:30 EDT 2022: couldn't determine idle interval
Sat  2 Jul 10:36:30 EDT 2022: taking snapshot of cam disk in /backingfiles/snapshots/snap-009843
Sat  2 Jul 10:36:44 EDT 2022: took snapshot
Sat  2 Jul 10:36:48 EDT 2022: comparing new snapshot with /backingfiles/snapshots/snap-009842/snap.bin
Sat  2 Jul 10:36:48 EDT 2022: new snapshot is identical to previous one, discarding
Sat  2 Jul 11:34:52 EDT 2022: waiting up to 90 seconds for idle interval
Sat  2 Jul 11:36:34 EDT 2022: couldn't determine idle interval
Sat  2 Jul 11:36:34 EDT 2022: taking snapshot of cam disk in /backingfiles/snapshots/snap-009843
Sat  2 Jul 11:36:48 EDT 2022: took snapshot
Sat  2 Jul 11:36:53 EDT 2022: comparing new snapshot with /backingfiles/snapshots/snap-009842/snap.bin
Sat  2 Jul 11:36:53 EDT 2022: new snapshot is identical to previous one, discarding
Sat  2 Jul 12:34:55 EDT 2022: waiting up to 90 seconds for idle interval
Sat  2 Jul 12:36:37 EDT 2022: couldn't determine idle interval
Sat  2 Jul 12:36:38 EDT 2022: taking snapshot of cam disk in /backingfiles/snapshots/snap-009843
Sat  2 Jul 12:36:50 EDT 2022: took snapshot
Sat  2 Jul 12:36:55 EDT 2022: comparing new snapshot with /backingfiles/snapshots/snap-009842/snap.bin
Sat  2 Jul 12:36:55 EDT 2022: new snapshot is identical to previous one, discarding
Sat  2 Jul 13:34:57 EDT 2022: waiting up to 90 seconds for idle interval
Sat  2 Jul 13:36:40 EDT 2022: couldn't determine idle interval
Sat  2 Jul 13:36:40 EDT 2022: taking snapshot of cam disk in /backingfiles/snapshots/snap-009843
Sat  2 Jul 13:36:53 EDT 2022: took snapshot
Sat  2 Jul 13:36:57 EDT 2022: comparing new snapshot with /backingfiles/snapshots/snap-009842/snap.bin
Sat  2 Jul 13:36:57 EDT 2022: new snapshot is identical to previous one, discarding
```
indicates that nothing was being recorded during that entire time.
Then you apparently went for a drive around 13:47:
```
Sat  2 Jul 13:47:44 EDT 2022: Sleeping before retry...
Sat  2 Jul 13:47:45 EDT 2022: Retrying...
Sat  2 Jul 13:47:50 EDT 2022: Sleeping before retry...
Sat  2 Jul 13:47:51 EDT 2022: Retrying...
Sat  2 Jul 13:47:56 EDT 2022: Sleeping before retry...
Sat  2 Jul 13:47:57 EDT 2022: Retrying...
Sat  2 Jul 13:48:02 EDT 2022: Sleeping before retry...
Sat  2 Jul 13:48:03 EDT 2022: Retrying...
Sat  2 Jul 13:48:03 EDT 2022: Sleeping before retry...
Sat  2 Jul 13:48:04 EDT 2022: Retrying...
Sat  2 Jul 13:48:05 EDT 2022: Sleeping before retry...
Sat  2 Jul 13:48:06 EDT 2022: Retrying...
Sat  2 Jul 13:48:06 EDT 2022: Sleeping before retry...
Sat  2 Jul 13:48:07 EDT 2022: Retrying...
Sat  2 Jul 13:48:07 EDT 2022: Sleeping before retry...
Sat  2 Jul 13:48:08 EDT 2022: Retrying...
Sat  2 Jul 13:48:08 EDT 2022: Sleeping before retry...
Sat  2 Jul 13:48:09 EDT 2022: Retrying...
Sat  2 Jul 13:48:09 EDT 2022: Sleeping before retry...
Sat  2 Jul 13:48:10 EDT 2022: Retrying...
Sat  2 Jul 13:48:10 EDT 2022: Attempts exhausted.
Sat  2 Jul 13:48:10 EDT 2022: Archive is unreachable.
```
(if you didn't go for a drive, then something else happened that caused the archive server to become unreachable)

[... 54 lines cut ...]
Sat  2 Jul 15:36:59 EDT 2022: Trimming free space in /mnt/cam, which has 3841 extents
Sat  2 Jul 15:37:00 EDT 2022: Trim complete, image now has 3841 extents
Sat  2 Jul 15:37:00 EDT 2022: Unmounting /mnt/cam...
Sat  2 Jul 15:37:01 EDT 2022: Unmounted /mnt/cam.
Sat  2 Jul 15:37:01 EDT 2022: Finished archiving.
Sat  2 Jul 15:37:01 EDT 2022: Copying music...
Sat  2 Jul 15:37:01 EDT 2022: Starting music sync...
Sat  2 Jul 15:37:01 EDT 2022: Ensuring music backing file is mounted...
Sat  2 Jul 15:37:01 EDT 2022: Mounting /mnt/music...
Sat  2 Jul 15:37:01 EDT 2022: Mounted /mnt/music.
Sat  2 Jul 15:37:01 EDT 2022: Ensured music drive is mounted.
Sat  2 Jul 15:37:01 EDT 2022: Syncing music from archive...
Sat  2 Jul 15:37:40 EDT 2022: Copied 0 music file(s), deleted 0, skipped 5693 previously-copied files, and encountered 0 errors.
Sat  2 Jul 15:37:40 EDT 2022: Trimming free space in /mnt/music, which has 171 extents
[... cut ...]
Sat  2 Jul 15:37:42 EDT 2022: Connected usb to host.
Sat  2 Jul 15:37:47 EDT 2022: Waiting for archive to be unreachable...
```
The only thing out of the ordinary is that the car isn't actually writing anything to the drive. I don't know why it would be doing that (though it certainly has nothing to do with the "temporary failure in name resolution" message). If it's a bad cable you wouldn't be able to access the USB music files in the car and the dashcam viewer in the car would show nothing, so that's something to check.

aaronac8, 2022-07-02:
I brought the teslausb into my home and connected to my PC and the music and cam drives are recognized with the same cable I use in the car, even though they were not recognized in the car.  Attached new diagnostic file:
[diagnostics.txt](https://github.com/marcone/teslausb/files/9033840/diagnostics.txt)

aaronac8, 2022-07-02:
I just tried a:
sudo -i
/root/bin/setup-teslausb upgrade

to see if that will help.  Not sure what else to do

aaronac8, 2022-07-03:
I just did a hard reset on the car and now the Teslausb is working.

marcone, 2022-07-06:
sorry, meant to keep this open as a reminder to try and do something about the "temporary failure in name resolution" message, since while benign, it makes people think something is wrong.

closed 2022-07-06 by marcone

SpeedlimitsB, 2024-01-08:
Hi @marcone ,
many thanks for you effort , 

Did you fix already the error resolving pool 3.debian.pool.ntp.org: Temporary failure in name resolution ?

regards

marcone, 2024-01-09:
No, it hasn't been a priority because it's not an actual problem.