marcone/teslausb #735: Can't upgrade/change settings: The microSD card is too small

SergioRius, 2022-07-19:
### Describe the problem

I changed my samba share for teslausb, so I wanted to update the configuration in the teslaUsb device.
I ssh'd and sudo'd.
I remounted as rw and modified the conf file. I copied it at /boot.
Executed: `/root/bin/setup-teslausb selfupdate`
Then executed: `/root/bin/setup-teslausb`
But it complained about the sd card being small.

I noticed that the console says to use a different command to upgrade. I tried it but the result is the same.

### Device

Raspberry Pi Zero W

### Car Model

Model 3

### USB connection

Center console

### Logs

Nothing relevant on logs, aside from the error trying to access the samba share.

### Additional information

root@teslausb:~# bin/setup-teslausb upgrade
preparing for full upgrade
/tmp/setup-teslausb is up to date
rc.local is up to date
Updating package index files...
Hit:1 http://raspbian.raspberrypi.org/raspbian buster InRelease                                    
Get:2 http://archive.raspberrypi.org/debian buster InRelease [32.6 kB]                             
Get:3 http://archive.raspberrypi.org/debian buster/main armhf Packages [391 kB]
Fetched 423 kB in 6s (65.5 kB/s)                                                                                                                                                                                                    
Reading package lists... Done
Verifying that the requested configuration is valid...
Downloaded /tmp/verify-configuration.sh ...
USB_DRIVE is not set. SD card will be used.
Verifying that there is sufficient space available on the MicroSD card...
STOP: The MicroSD card is too small: 3145728 bytes available.
Model: SD SH128 (sd/mmc)
Disk /dev/mmcblk0: 128GB
Sector size (logical/physical): 512B/512B
Partition Table: msdos
Disk Flags: 

Number  Start   End     Size    Type     File system  Flags
 1      4194kB  266MB   262MB   primary  fat32        boot, lba
 2      266MB   2414MB  2147MB  primary  ext4
 3      2414MB  128GB   125GB   primary  xfs
 4      128GB   128GB   105MB   primary  ext4

marcone, 2022-07-19:
Whoops, my bad. A recent change broke this. I've reverted the change, so please try again.

SergioRius, 2022-07-21:
> Whoops, my bad. A recent change broke this. I've reverted the change, so please try again.

Jejeje... Sometimes shit happens.

So to change the config, is the required procedure still?:
sudo -i
make it rw
change config file in user folder
copy to /boot
execute script (I guess only the one in the welcome text)
reboot

marcone, 2022-07-22:
You don't need to copy the config file back to /boot, you can edit it in-place in /root