marcone/teslausb #736: ERROR: rsync: [Receiver] ERROR: cannot stat destination "/mnt/archive": Stale file handle (116)

hikewv, 2022-07-23:
### Describe the problem

When TeslaUSB connects to my wifi i get this rsync error first every time.

rsync: [Receiver] ERROR: cannot stat destination "/mnt/archive": Stale file handle (116)
rsync error: errors selecting input/output files, dirs (code 3) at main.c(765) [Receiver=v3.2.3]

If then i click under tools Trigger archive/sync it then works no problem and copies all the files with no errors.

### Device

Raspberry Pi Zero 2 W

### Car Model

Model 3

### USB connection

Center console

### Logs

2022/07/20 18:20:33 [27408] building file list
2022/07/20 18:20:33 [27408] done
building file list ... done
rsync: [Receiver] ERROR: cannot stat destination "/mnt/archive": Stale file handle (116)
rsync error: errors selecting input/output files, dirs (code 3) at main.c(765) [Receiver=v3.2.3]
Wed 20 Jul 18:20:34 EDT 2022: Ensuring cam file is mounted...
Wed 20 Jul 18:20:34 EDT 2022: Disconnecting usb from host...
Wed 20 Jul 18:20:34 EDT 2022: Running fsck on /backingfiles/cam_disk.bin...
Wed 20 Jul 18:20:34 EDT 2022: | fsck from util-linux 2.33.1
Wed 20 Jul 18:20:37 EDT 2022: | fsck.fat 4.1 (2017-01-24)
Wed 20 Jul 18:20:37 EDT 2022: | 0x41: Dirty bit is set. Fs was not properly unmounted and some data may be corrupt.
Wed 20 Jul 18:20:37 EDT 2022: |  Automatically removing dirty bit.
Wed 20 Jul 18:20:37 EDT 2022: | Performing changes.
Wed 20 Jul 18:20:37 EDT 2022: | /dev/loop2p1: 346 files, 303224/5886542 clusters
Wed 20 Jul 18:20:37 EDT 2022: Finished fsck on /backingfiles/cam_disk.bin.
Wed 20 Jul 18:20:37 EDT 2022: Running fsck on /backingfiles/music_disk.bin...
Wed 20 Jul 18:20:37 EDT 2022: | fsck from util-linux 2.33.1
Wed 20 Jul 18:20:38 EDT 2022: | fsck.fat 4.1 (2017-01-24)
Wed 20 Jul 18:20:38 EDT 2022: | /dev/loop2p1: 4 files, 3/1471610 clusters
Wed 20 Jul 18:20:39 EDT 2022: Finished fsck on /backingfiles/music_disk.bin.
Wed 20 Jul 18:20:39 EDT 2022: Disconnected usb from host.
Wed 20 Jul 18:20:39 EDT 2022: Mounting /mnt/cam...
Wed 20 Jul 18:20:39 EDT 2022: Mounted /mnt/cam.
Wed 20 Jul 18:20:39 EDT 2022: Ensured cam file is mounted.
Wed 20 Jul 18:20:39 EDT 2022: cleaning cam mount
Wed 20 Jul 18:20:40 EDT 2022: done cleaning cam mount
Wed 20 Jul 18:20:40 EDT 2022: Trimming free space in /mnt/cam, which has 313 extents
Wed 20 Jul 18:20:41 EDT 2022: Trim complete, image now has 176 extents
Wed 20 Jul 18:20:41 EDT 2022: Unmounting /mnt/cam...
Wed 20 Jul 18:20:41 EDT 2022: Unmounted /mnt/cam.
Wed 20 Jul 18:20:41 EDT 2022: Finished archiving.
Wed 20 Jul 18:20:41 EDT 2022: Copying music...
Wed 20 Jul 18:20:41 EDT 2022: Starting music sync...
Wed 20 Jul 18:20:41 EDT 2022: Ensuring music backing file is mounted...
Wed 20 Jul 18:20:41 EDT 2022: Disconnecting usb from host...
Wed 20 Jul 18:20:41 EDT 2022: Disconnected usb from host.
Wed 20 Jul 18:20:41 EDT 2022: Mounting /mnt/music...
Wed 20 Jul 18:20:41 EDT 2022: Mounted /mnt/music.
Wed 20 Jul 18:20:41 EDT 2022: Ensured music drive is mounted.
Wed 20 Jul 18:20:41 EDT 2022: Syncing music from archive...
Wed 20 Jul 18:20:42 EDT 2022: Copied 0 music file(s), deleted 0, skipped 0 previously-copied files, and encountered 0 errors.
Wed 20 Jul 18:20:42 EDT 2022: Trimming free space in /mnt/music, which has 3 extents
Wed 20 Jul 18:20:42 EDT 2022: Trim complete, image now has 3 extents
Wed 20 Jul 18:20:42 EDT 2022: Unmounting /mnt/music...
Wed 20 Jul 18:20:42 EDT 2022: Unmounted /mnt/music.
Wed 20 Jul 18:20:42 EDT 2022: Finished copying music.
Wed 20 Jul 18:20:42 EDT 2022: unmounting /mnt/archive
Wed 20 Jul 18:20:42 EDT 2022: Connecting usb to host...
Wed 20 Jul 18:20:42 EDT 2022: unmounting /mnt/musicarchive
Wed 20 Jul 18:20:43 EDT 2022: Connected usb to host.
Wed 20 Jul 18:20:48 EDT 2022: Waiting for archive to be unreachable...
Wed 20 Jul 18:40:39 EDT 2022: Simulating archive being unreachable.
Wed 20 Jul 18:40:39 EDT 2022: Waiting for archive to be reachable...
Wed 20 Jul 18:40:39 EDT 2022: Archive is reachable.
Wed 20 Jul 18:40:39 EDT 2022: Trying to set time...
Wed 20 Jul 18:40:39 EDT 2022: Time adjusted by 0.108566 seconds after 0.100000 seconds
Wed 20 Jul 18:40:39 EDT 2022: not keeping car awake.
Wed 20 Jul 18:40:59 EDT 2022: taking snapshot of cam disk in /backingfiles/snapshots/snap-000018
Wed 20 Jul 18:41:03 EDT 2022: took snapshot
Wed 20 Jul 18:41:05 EDT 2022: comparing new snapshot with /backingfiles/snapshots/snap-000017/snap.bin
Wed 20 Jul 18:41:05 EDT 2022: making links for /tmp/snapshots/snap-000018, retargeted to /backingfiles/snapshots/snap-000018/mnt
Wed 20 Jul 18:41:07 EDT 2022: made all links for /tmp/snapshots/snap-000018
Wed 20 Jul 18:41:07 EDT 2022: Archiving...
Wed 20 Jul 18:41:07 EDT 2022: Ensuring cam archive is mounted...
Wed 20 Jul 18:41:07 EDT 2022: Mounting /mnt/archive...
Wed 20 Jul 18:41:07 EDT 2022: Mounted /mnt/archive.
Wed 20 Jul 18:41:07 EDT 2022: Ensured cam archive is mounted.
Wed 20 Jul 18:41:07 EDT 2022: Ensuring music archive is mounted...
Wed 20 Jul 18:41:07 EDT 2022: Mounting /mnt/musicarchive...
Wed 20 Jul 18:41:08 EDT 2022: Mounted /mnt/musicarchive.
Wed 20 Jul 18:41:08 EDT 2022: Ensured music archive is mounted.
Wed 20 Jul 18:41:08 EDT 2022: Checking saved folder count...
Wed 20 Jul 18:41:08 EDT 2022: There are 1 event folder(s) with 46 file(s) and 0 track mode file(s) to move.  1 short recording(s) will be skipped.
Wed 20 Jul 18:41:08 EDT 2022: Starting recording archiving
Wed 20 Jul 18:41:08 EDT 2022: Archiving 46 file(s) including 1 event folder(s) starting at Wed 20 Jul 18:41:08 EDT 2022
Wed 20 Jul 18:41:08 EDT 2022: Sending Slack message.
Wed 20 Jul 18:47:04 EDT 2022: Archiving completed successfully. Archived 46 files in 5m56s
Wed 20 Jul 18:47:04 EDT 2022: Sending Slack message.

marcone, 2022-07-24:
Please attach full diagnostics (can be downloaded from the diagnostics tab in the web ui)

hikewv, 2022-07-24:
[diagnostics.txt](https://github.com/marcone/teslausb/files/9176828/diagnostics.txt)

marcone, 2022-07-24:
The logs just prior to the part you quoted earlier are:
```
Wed 20 Jul 18:20:32 EDT 2022: Ensuring cam archive is mounted...
Wed 20 Jul 18:20:32 EDT 2022: Mounting /mnt/archive...
Wed 20 Jul 18:20:32 EDT 2022: Mounted /mnt/archive.
Wed 20 Jul 18:20:32 EDT 2022: Ensured cam archive is mounted.
Wed 20 Jul 18:20:32 EDT 2022: Ensuring music archive is mounted...
Wed 20 Jul 18:20:32 EDT 2022: Mounting /mnt/musicarchive...
Wed 20 Jul 18:20:32 EDT 2022: Mounted /mnt/musicarchive.
Wed 20 Jul 18:20:32 EDT 2022: Ensured music archive is mounted.
Wed 20 Jul 18:20:32 EDT 2022: Checking saved folder count...
Wed 20 Jul 18:20:32 EDT 2022: There are 1 event folder(s) with 46 file(s) and 0 track mode file(s) to move.  1 short recording(s) will be skipped.
Wed 20 Jul 18:20:33 EDT 2022: Starting recording archiving
Wed 20 Jul 18:20:33 EDT 2022: Archiving 46 file(s) including 1 event folder(s) starting at Wed 20 Jul 18:20:33 EDT 2022
Wed 20 Jul 18:20:33 EDT 2022: Sending Slack message.
Wed 20 Jul 18:20:33 EDT 2022: Error during archiving. Archived 0 files in 0s
Wed 20 Jul 18:20:33 EDT 2022: Sending Slack message.
2022/07/20 18:20:33 [27408] building file list
2022/07/20 18:20:33 [27408] done
building file list ... done
rsync: [Receiver] ERROR: cannot stat destination "/mnt/archive": Stale file handle (116)
rsync error: errors selecting input/output files, dirs (code 3) at main.c(765) [Receiver=v3.2.3]
```
so basically teslausb mounted the file share, and then literally a second later it got a "stale file handle" for that same share.
I don't know what would cause that, but I'm inclined to think it's a problem on the server side. Have you checked for errors on the server?

hikewv, 2022-07-24:
No errors that I can see in my server log.  Whats weird is i can do a Trigger archive/sync and it works no problems.  Hum.

hikewv, 2022-07-24:
Well i did some further reading a setting to try.
What I did on my unraid server was turn off Use cache pool (for new files/directories) for my TeslaCam Share, it was set to yes.  I was reading where that could cause the issue of the Stale file handle. 
I then made a recording, unplugged the teslausb, plugged it back it, it connected and copied the files of the saved clip no issues.

I'll keep giving it a try later this week when i run errands and i'll make some other clips and see what happens once i get home and it connects back to the wifi.  But I have a feeling this might have resolved the issue,

hikewv, 2022-07-30:
This was resolved by me updating the configuration on unraid file share, not to use a cache pool for new files/directories for the share.
It has been working perfectly since.

closed 2022-07-30 by hikewv

jxjftw, 2023-02-18:
> This was resolved by me updating the configuration on unraid file share, not to use a cache pool for new files/directories for the share.
> It has been working perfectly since.

Thanks brother, I was having the same issue on my unfair setup.

THESHULK, 2023-03-09:
@hikewv @jxjftw I'm trying to get TeslaUSB setup with my Unraid server but it's having trouble mounter the share:

`Thu  9 Mar 17:48:01 CST 2023 : verify-and-configure-archive: Trying mount command-line:
Thu  9 Mar 17:48:01 CST 2023 : verify-and-configure-archive: mount -t cifs '//unraid-server.local/mnt/user/backup/TeslaCam' '/tmp/archivetestmount' -o 'credentials=/tmp/teslaCamArchiveCredentials,iocharset=utf8,file_mode=0777,dir_mode=0777,vers=default,'
`

I have a share called "backup" with a sub-folder called "TeslaCam" that I want the recording transferred to. How do you guys have it configured in Unraid and the conf file?

lwndow, 2025-03-16:
> This was resolved by me updating the configuration on unraid file share, not to use a cache pool for new files/directories for the share. It has been working perfectly since.

Thanks for posting this, never would have thought that using the cache would cause this issue, now working fine here too.

lwndow, 2025-03-16:
> [@hikewv](https://github.com/hikewv) [@jxjftw](https://github.com/jxjftw) I'm trying to get TeslaUSB setup with my Unraid server but it's having trouble mounting the share:
> 
> Thu 9 Mar 17:48:01 CST 2023 : verify-and-configure-archive: Trying mount command-line: Thu 9 Mar 17:48:01 CST 2023 : verify-and-configure-archive: mount -t cifs '//unraid-server.local/mnt/user/backup/TeslaCam' '/tmp/archivetestmount' -o 'credentials=/tmp/teslaCamArchiveCredentials,iocharset=utf8,file_mode=0777,dir_mode=0777,vers=default,' ... Thu 9 Mar 17:48:07 CST 2023 : verify-and-configure-archive: STOP: no working combination of vers and sec mount options worked
> 
> I have a share called "backup" with a sub-folder called "TeslaCam" that I want the recordings transferred to. How do you guys have it configured in Unraid and the conf file?
> 
> Here are my conf file settings: export ARCHIVE_SYSTEM=cifs export ARCHIVE_SERVER=unraid-server.local export SHARE_NAME=mnt/user/backup/TeslaCam

Realize this is old, but if you never managed to get it working, my advice would be to make a share from the Shares config and a dedicated user, then have your config be:
```
export ARCHIVE_SYSTEM=cifs
export ARCHIVE_SERVER=$SERVER
export SHARE_NAME='$SHARENAME/$VEHICLE'
export SHARE_USER=$CAR_USER
export SHARE_PASSWORD=$CAR_PASSWORD
```

lwndow, 2025-03-16:
For anyone coming later syncing to an Unraid server, this fix is better if you want the speed of the cache drive. https://github.com/marcone/teslausb/issues/941