marcone/teslausb #737: configure: Skipping tesla_api.py install because no credentials were provided

chilman408, 2022-08-09:
### Describe the problem

Getting this error in the Setup logs when I provided my Tesla credentials...

configure: Skipping tesla_api.py install because no credentials were provided

### Device

Raspberry Pi Zero 2 W

### Car Model

Model 3

### USB connection

Center console

marcone, 2022-08-09:
I assume you specified your email and password? That doesn't work anymore. You need to specify a refresh token, as described in [the latest version of the config file](https://github.com/marcone/teslausb/blob/main-dev/pi-gen-sources/00-teslausb-tweaks/files/teslausb_setup_variables.conf.sample#L192)

chilman408, 2022-08-09:
Yes, used email and pwd.  But I reas that the token expires in 8hrs now, will Teslausb be able to auto update the token?

marcone, 2022-08-09:
it should automatically refresh/update when needed

closed 2022-08-19 by marcone