marcone/teslausb #739: Feature Request: "Map" RPi bluetooth to the Tesla usb for wireless game controllers

chrisryan10, 2022-09-04:
### Describe the problem

The use case here is to connect a wireless game controller (e.g. xbox controller) to the Tesla for games. With the changes to the Model 3/Y in recent times, the only USB data port available is in the glovebox. If I'm already using a RPi for tesla_dashcam, it would ideal if I could take advantage of the bluetooth available in the RPi as well and connect the game controller to it, which in turn would make it available to the Tesla.

The current solution is to get a usb splitter/hub and a usb-bluetooth dongle, but that's at least 2 more pieces of hardware (and at least 1 more cable) required.

### Device

Raspberry Pi 4

### Car Model

Model 3

### USB connection

Glove box

### Logs

N/A

marcone, 2022-09-04:
It's a neat idea, but a bit beyond the scope of teslausb. It also seems like it might ultimately be wasted effort, since Tesla is rumored to be adding support for bluetooth controllers. You're welcome to contribute code to do this though.

chrisryan10, 2022-09-04:
I had not heard that - awesome.  Will go with the workaround for now and await an update.

closed 2022-09-04 by chrisryan10