marcone/teslausb #741: Pi 4  usb SSD update error

mitchcecg, 2022-09-30:
### Describe the problem

below you will see the ssd is already formatted but the update script has a problem with that 
Fri 30 Sep 12:30:21 EDT 2022 : Verifying that there is sufficient space available on the USB drive ...
Fri 30 Sep 12:30:21 EDT 2022 : STOP: The provided drive seems to be a partition. Please specify path to the disk.
------------------------
a slightly larger clip

  File "/root/bin/tzupdate.py", line 322, in <module>
    main()
  File "/root/bin/tzupdate.py", line 284, in main
    timezone = get_timezone(args.ip, timeout=args.timeout, services=services)
  File "/root/bin/tzupdate.py", line 72, in get_timezone
    "No usable response from any API in {} seconds".format(timeout)
__main__.TimezoneAcquisitionError: No usable response from any API in 5.0 seconds
Fri 30 Sep 12:30:21 EDT 2022 : Verifying that the requested configuration is valid...
Fri 30 Sep 12:30:21 EDT 2022 : Downloaded /tmp/verify-configuration.sh ...
Fri 30 Sep 12:30:21 EDT 2022 : Checking XFS support
Fri 30 Sep 12:30:21 EDT 2022 : XFS supported
Fri 30 Sep 12:30:21 EDT 2022 : USB_DRIVE is set to /dev/sda. This will be used for /mutable and backingfiles.
Fri 30 Sep 12:30:21 EDT 2022 : Verifying that there is sufficient space available on the USB drive ...
Fri 30 Sep 12:30:21 EDT 2022 : STOP: The provided drive seems to be a partition. Please specify path to the disk.

### Device

Raspberry Pi 4

### Car Model

Model Y

### USB connection

Glove box

### Logs

Fri 30 Sep 12:30:21 EDT 2022 : Verifying that there is sufficient space available on the USB drive ...
Fri 30 Sep 12:30:21 EDT 2022 : STOP: The provided drive seems to be a partition. Please specify path to the disk.

marcone, 2022-10-06:
I cannot reproduce this, installation works for me regardless of whether the external drive is already formatted or not.
The current setup script should give a somewhat more meaningful error message, so maybe try again.
If it still fails, and you can ssh into the Pi, please run `lsblk /dev/sda` and paste its output here.

closed 2022-10-08 by marcone

mitchcecg, 2022-10-08:
Will try again during the week been sick

THANX
MITCH

Closed #741<https://github.com/marcone/teslausb/issues/741> as completed.