marcone/teslausb #747: USB drive not found

Str1atum, 2022-12-03:
### Describe the problem

Latest teslausb on Pi Zero 2 W connected to 2022 Model Y in the glovebox

256 GB Sandisk with Camsize 128G

Setup worked fine, wifi connection established and Samba share available in my home network. When connected to the Tesla it boots correctly and flashes once per second but the car doesn’t find the USB drive at all. 

Any ideas?

### Device

Raspberry Pi Zero 2 W

### Car Model

Model Y

### USB connection

Glove box

marcone, 2022-12-03:
Have you looked at https://github.com/marcone/teslausb/wiki/Troubleshooting ?
(specifically the parts about trying different cables, and connecting to a PC)

marcone, 2022-12-10:
Closing due to no response

closed 2022-12-10 by marcone