marcone/teslausb #751: Teslausb on Armbian Bullseye not mounted by car

miles267, 2022-12-22:
### Describe the problem

@marcone Have Teslausb installed and running on Armbian. It’s connected to the glovebox USB3 port using a splitter with the car powering the Rock Pi 4C+ over USB-C and a data USB3 cable connected to the USB3 of the RPi4C+. The device will boot up and connect to Wi-Fi showing with hostname Teslausb, but the Pi LED flashes twice every second but the car doesn’t detect it (car shows unavailable). Have you experienced this or does your car work fine with the RPi4C+? Was thinking it would mount over USB3 and the slow usb device errors would no longer appear in the car. Thanks. 

### Device

Rock Pi 4C plus

### OS Image

Armbian

### Car Model

Model 3

### USB connection

Glove box

marcone, 2022-12-22:
Are you using the top USB3 port? (that's the OTG port by default, the bottom one is host and won't work for this)
Where did you get the Armbian image?

miles267, 2022-12-22:
Yes, I’ve tried both USB3 ports. I got the image from here: https://github.com/radxa-build/rock-4c-plus/releases and have applied all available updates.

marcone, 2022-12-23:
The minimal bullseye image from that link, specifically [this one](https://github.com/radxa-build/rock-4c-plus/releases/download/20221216-1256/Armbian_23.02.0-trunk_Rockpi-4cplus_bullseye_current_5.15.83_minimal.img.xz), works for me.
Since that page recommends using the official Armbian image instead I tried [that one](https://mirror-us-phx1.armbian.airframes.io/dl/rockpi-4cplus/archive/Armbian_22.11.1_Rockpi-4cplus_bullseye_current_5.15.80_minimal.img.xz) too, and it worked too.

Y-cables usually have one leg that is power-only and another that is power+data. Have you verified that you're connecting the power+data leg to the Rock Pi's USB3 port?
Have you tried connecting to a PC instead of the car, and/or using two separate cables instead of a Y-cable?

miles267, 2022-12-23:
Yes, I was able to get it to work briefly today with the latest Jammy minimal build although I’m sure the bullseye one would work too. I have found that the Broadcom wireless chipset of the Rock Pi 4C+ is rather pathetic compared to the RPi4 I use reliably. It seems to have much more limited range and the throughout is slow. Perhaps that will improve with future drivers or it’s just a hardware limitation of these early builds. Have you found that the slow USB/glovebox errors have gone away using yours with the RPi4C+ over USB3?

marcone, 2022-12-23:
> Yes, I was able to get it to work briefly today with the latest Jammy minimal build although I’m sure the bullseye one would work too.

Can you elaborate? How were you able to get it to work? Why only briefly? What made it stop working?

> I have found that the Broadcom wireless chipset of the Rock Pi 4C+ is rather pathetic compared to the RPi4 I use reliably. It seems to have much more limited range and the throughout is slow.

What speeds are you seeing? I'm seeing up to 18 MB/sec throughput with TeslaUSB's built in network speed test, compared to up to 10 MB/sec with a Raspberry Pi 4 connected to the same base station at the same location.
Did you attach an antenna to the Rock Pi? (mine came with one in the box, but I don't know if every seller includes one)

> Perhaps that will improve with future drivers or it’s just a hardware limitation of these early builds. Have you found that the slow USB/glovebox errors have gone away using yours with the RPi4C+ over USB3?

My car does not have a glovebox USB port, and I never had the "slow USB" errors even with a Raspberry Pi Zero W.

miles267, 2022-12-23:
@marcone i need to spend some more time with it. Seems I had it working, connected to USB3 (top port) the camera volume was connected and I was playing FLAC audio from USB3. Upon switching to the radio, the Teslausb became Unavailable from the car UI.

Also it appears I may have a Wi-Fi driver or firmware issue according to the dmesg log above. I’ve posted to the Armbian forum but no know fix yet.

marcone, 2022-12-23:
I'm seeing those same messages in dmesg, but wifi works anyway.

miles267, 2022-12-23:
OK. I’ll ignore those Broadcom firmware messages. Have flashed with same Bullseye image you’re using. Unfortunately mine did not include an external antenna. Compared to my RPi4 which gets approx 90 Mbps from the same wireless AP above the car in the garage from within the glovebox, the Rock Pi 4C+ is getting about 30 Mbps or 30% of the throughout.

I’ll continue to try it out and see how it performs under daily use using USB3. Are you noticing better write performance with your Rock Pi based on USB3 emulation?

marcone, 2023-01-02:
> Are you noticing better write performance with your Rock Pi based on USB3 emulation?

Absolutely. Over 100 MB/sec sequential write speed.
(I don't know at what point the car would complain about the drive being "too slow", but the car only writes about 2 MB/sec on average)

miles267, 2023-01-02:
OK. I may not have the right power source/OTG cable as it seems mine is disconnecting. Also, am looking for an external antenna as my Wi-Fi signal on this RPi4C+ is rather weak compared to both the Zero 2 W and my current RPi4.

marcone, 2025-01-07:
Closing this issue as it's over 2 years old and there isn't anything actionable here.

closed 2025-01-07 by marcone