marcone/teslausb #756: Setup fails to complete when SSH_ROOT_PUBLIC_KEY is set (no /root/.ssh/ directory)

JinbaIttai, 2023-04-11:
### Describe the problem

Hi, I just installed teslausb for the first time, and noticed the setup process never finished and the Pi did not reboot.

After logging in I saw the last entry in /boot/teslausb-headless-setup.log was "configuring ssh" and the setup process was no longer running.

I ran /etc/rc.local as per the troubleshooting guide and it stopped at:

```
Downloaded /tmp/configure-ssh.sh ...
configuring ssh
/tmp/configure-ssh.sh: line 8: /root/.ssh/authorized_keys: No such file or directory
```

The reason it failed is /root/.ssh/ directory does not exist.

After running "mkdir -p /root/.ssh"  and starting /etc/rc.local again the setup completed successfully.

I recommend adding the mkdir command to the configure-ssh.sh script to fix this bug.

### Device

Raspberry Pi Zero 2 W

### OS Image

Prebuilt TeslaUSB image

### Car Model

Model Y

### USB connection

Glove box

marcone, 2023-04-11:
@tomrund

tomrund, 2023-04-11:
Ahh sorry about that @JinbaIttai, thank you for the recommendation, https://github.com/marcone/teslausb/pull/757 should resolve this issue.