marcone/teslausb #759: Feature: Regular Notifications with sync progress

Fredyy90, 2023-04-15:
### Describe the problem

Hey,

I often have 10-20 Sentry Events per day and pretty poor WIFI reception for my teslausb so the daily-sync takes a lot of time.
I would really like a function in the teslausb notification setup, that is configurable and sends a "progress" message during sync on a fixed schedule, like either every x % or every x minutes.

Currently, there are only notifications for start, stop and error of sync. But during the sometimes 5-6h of syncing over here, there is no simple way to see the sync-progress.

Once the progress is tracked, it would also be nice, to show it on the web interface.

### Device

Raspberry Pi Zero W

### OS Image

Prebuilt TeslaUSB image

### Car Model

Model Y

### USB connection

Glove box

thesourcerer, 2024-07-01:
+1. Also archive as long as the network is available, do not wait for reconnection. when archiving recent video, it will never finish to archive otherwise with the bandwidth that pi zero w can provide...

oaquique, 2025-04-20:
+1 on this. Having a progress indicator (even in the web UI) about replication would be great. If we can also include having an optional OLED screen with update info would be a cherry on the top. :)