marcone/teslausb #760: Setting up a new Raspberry Pi 4 for a new car (a 2nd car) without success

miles267, 2023-04-17:
### Describe the problem

Hi - we've added a second vehicle to household on same wifi network.  First car is successfully running on a Raspberry Pi 4 with latest image without issue.  Am now assembling a second Raspberry 4 device (identical hardware) on same network.  I've copied the original config file to the new device only modifying the following variables but it isn't synching with the network:

export TESLAUSB_HOSTNAME
export tesla_vin

Everything else remains identical.  Yet the new RPi4 is not synching as it should.  Do I need to be using a new TESLA_REFRESH_TOKEN?  I don't want to break the original, operational device while adding this new one.

Thank you.

### Device

Raspberry Pi 4

### OS Image

Prebuilt TeslaUSB image

### Car Model

Model Y

### USB connection

Glove box

### Logs

[diagnostics.txt](https://github.com/marcone/teslausb/files/11256417/diagnostics.txt)

miles267, 2023-04-17:
The new raspberry pi mounts as expected in the car without issue, however it isn't syncing any data to or from the smb file share whereas the other, nearly identical device does without issue.

miles267, 2023-04-18:
I'm able to get camera footage to sync without issue, however music is not syncing despite setting the following on a 256 GB SD card:

export musicsize=64G
export musicsharename=teslamusic (this is the name of my share)

It just seems to ignore the music part of the sync.

Mon 17 Apr 20:04:56 CDT 2023: Copying music...
Mon 17 Apr 20:04:56 CDT 2023: Starting music sync...
Mon 17 Apr 20:04:56 CDT 2023: Ensuring music backing file is mounted...
Mon 17 Apr 20:04:56 CDT 2023: Disconnecting usb from host...
Mon 17 Apr 20:04:56 CDT 2023: Disconnected usb from host.
Mon 17 Apr 20:04:56 CDT 2023: Mounting /mnt/music...
Mon 17 Apr 20:04:56 CDT 2023: Mounted /mnt/music.
Mon 17 Apr 20:04:56 CDT 2023: Ensured music drive is mounted.
Mon 17 Apr 20:04:56 CDT 2023: Trimming free space in /mnt/music, which has 1 extents
Mon 17 Apr 20:04:56 CDT 2023: Trim complete, image now has 1 extents
Mon 17 Apr 20:04:56 CDT 2023: Unmounting /mnt/music...
Mon 17 Apr 20:04:56 CDT 2023: Unmounted /mnt/music.
Mon 17 Apr 20:04:56 CDT 2023: Finished copying music.
Mon 17 Apr 20:04:56 CDT 2023: unmounting /mnt/archive
Mon 17 Apr 20:04:56 CDT 2023: Stopping wake background task.
Mon 17 Apr 20:04:56 CDT 2023: Connecting usb to host...
Mon 17 Apr 20:04:56 CDT 2023: unmounting /mnt/musicarchive

marcone, 2023-04-19:
This is fixed now

closed 2023-04-19 by marcone