marcone/teslausb #762: stack smashing detected

harmstorf, 2023-04-27:
### Describe the problem

I am running Armbian 23.02.2 Jammy with Linux 5.15.93-rockchip64 on a Rock 4C+ and followed the instructions on https://github.com/marcone/teslausb/wiki/Installation

The installation aborts:

`root@rockpi-4cplus:~# curl https://raw.githubusercontent.com/marcone/teslausb/main-dev/setup/generic/install.sh | bash
*** stack smashing detected ***: terminated`

### Device

Rock Pi 4C plus

### OS Image

Armbian

### Car Model

Model 3

### USB connection

Glove box

marcone, 2023-04-29:
Can you link the specific Armbian image you used?

harmstorf, 2023-05-02:
I used the 4C+ Image Armbian_23.02.2_Rockpi-4cplus_jammy_current_5.15.93_minimal.img.xz provided here: https://www.armbian.com/rockpi4/

marcone, 2023-05-06:
I used that same image but did not encounter the "stack smashing detected" problem. Is this reproducible for you?

harmstorf, 2023-05-06:
Thanks for the valid question which made me start from scratch again. This time, no issues with stack smashing.

closed 2023-05-06 by harmstorf