marcone/teslausb #765: Setup Fails when SSD has backingfiles from previous TeslaUSB

dariusdunlap, 2023-05-22:
### Describe the problem

reinstalling TeslaUSB on Pi4b with Samsung T5 SSD as cam and music store. 

Installation fails when I use previously configured T5 connected while Setup runs. This was verified when after I got it working, I borked something while installing Argon Fan Hat support and decided the simplest fix was to re-flash and run through setup again.

Initial symptom I saw was Setup failure showing error that the mount points  /mnt/cam and /mnt/music did not exist. Creating these manually and re-running did not fix it, so I suspect other parts of the setup script also did not run properly. 

Re-flashed, loaded my known good teslausb_setup_variables.conf, and this time formatted the SSD, and now the Startup ran straight through correctly. 

Here's a cleansed version (placeholder SSIDs, passwords, etc.) of the teslausb_setup_variables.conf file:
[teslausb_setup_variables-cleansed.conf.txt](https://github.com/marcone/teslausb/files/11533661/teslausb_setup_variables-cleansed.conf.txt)

This can be recreated by these steps: 

- Flash TeslaUSB image to SD card
- load teslausb_setup_variables.conf in /boot (perhaps editing so that cifs music share works with your setup)
- attach freshly formatted T5
- Boot and run through setup. Should be successful
- Wait for Archive/Sync to run completely
- shutdown
- re-flash TeslaUSB image to SD card
- load teslausb_setup_variables.conf in /boot
- DO NOT re-format T5
- Boot and let setup run. It will fail.

Verification:
- Flash TeslaUSB image to SD card
- load teslausb_setup_variables.conf in /boot (perhaps editing so that cifs music share works with your setup)
- Re-format t5 and attach freshly formatted T5
- Boot and run through setup. Should be successful

### Device

Raspberry Pi 4

### OS Image

Prebuilt TeslaUSB image

### Car Model

Model 3

### USB connection

Center console

marcone, 2023-06-04:
> * Boot and let setup run. It will fail.

Fail in what way?

dariusdunlap, 2023-06-05:
From the original report: 

"Setup failure showing error that the mount points /mnt/cam and /mnt/music did not exist. Creating these manually and re-running did not fix it, so I suspect other parts of the setup script also did not run properly." 

If I had understood what was happening, I might have been able to look back at the installation log to see what else might have gone wrong. If you have any trouble recreating the problem, I can put some time into it later this week to see if I can come up with a clean report. 

I'm guessing that the installation sees the already configured USB drive and so it skips over some sections of the setup, thereby missing some important steps – missing mount points at /mnt/cam and /mnt/music were right there in the error, but it seems there was more setup that was not done since creating the mount points didn't fix it. 

It all ran correctly with exactly the same .conf when I gave it a newly formatted USB drive. 

d.

timberjoegithub, 2023-06-20:
I am running into the exact same problem.   My pi had stopped syncing and I tried to do an update and it failed so I started a rebuild.  The rebuild fails in the exact method described.  /mnt/cam and /mnt/music don't get created and it fails on "failed to start /etc/rc.local" and the directories for /etc/cam and /etc/music missing.  Adding them and restarting failed as well.   I am using an external drive mounted on sda.  I killed the existing partitions and it still fails.

dariusdunlap, 2023-06-20:
timberjoegithub - I got it to work by starting the installation completely from scratch and providing a blank USB drive for the camera and music storage. I haven't had a chance to experiment further, once I got it working, or to look at the script details to see if I can find the error(s) myself.

timberjoegithub, 2023-06-22:
Just cleaning the device on /dev/sda did not work for me, I still had issues once I blew away the external drive.  I got frustrated and decided to make this whole processes even harder and move to bullseye while I was struggling with it (after wiping the external drive).  I had to hack up the install.sh script to make it work with /dev/sda otherwise it would boot loop trying to get the disk space on the sdcard.  once I got that sorted and got the partitions sorted I was able to get to a point where I am seeing issues that come back to fstab (which I think may have been an initial problem with the install).  I needed to upgrade fuse to fuse3 in order to support the nofail argument.  I am still getting errors in fstab that it is seeing a mount point as not empty (even though it appears to be) so I am going to play around with getting the nonempty directive into fstab.  I am getting close to having a working system again.  Not everything is mounting correctly yet, but it feels like that is just a matter of digging a bit more.

timberjoegithub, 2023-06-23:
All set and working again using the manual build process with all the partitioning stuff taken out of install.sh to allow for an external storage device on /dev/sda.   I did need to upgrade to fuse 3 and the mount errors seem to have been more warnings as the appropriate mounts seem to be happening both away from wifi range and within.

commit 63415fc2 (2024-10-28): setup: fix setup with existing external disk

Disk image mountpoints weren't created when setting up TeslaUSB
from scratch with an external drive from a prior install.
Fixes #765
closed 2024-10-28 by marcone