marcone/teslausb #766: Fri 26 May 16:49:46 BST 2023 : Mounting the partition for the backing files...

dansche, 2023-05-26:
### Describe the problem

Fresh install, everything done according to the Video with the User and the share, but it just does not Mount the drives.

What could be the issue?

### Device

Raspberry Pi Zero W

### OS Image

Prebuilt TeslaUSB image

### Car Model

Model 3

### USB connection

Center console

marcone, 2023-05-28:
There isn't enough information here.

closed 2023-05-28 by marcone

dansche, 2023-05-28:
Glad this is still active, I‘m not a pro ehrtet with Raspbeery nor with GitHub, if you tell me what you need I‘m providing everything, not urgent as I am on vacation next week.

Poncherelly, 2023-05-28:
Which center console do you have. If it's the version 2, the USB port may not have data. Try connecting a game controller to see if you can play games. If not, you probably have to use the USB port in the glove box or buy the data module for the center console.

dansche, 2023-05-28:
Well I am not even that far, the mount the drives is by installing the Raspberry Pi Zero, the routine starts but stucks allways by the line mounting the drives. So the raspberry is still under control by SSH and connected to the USB power from the power outlet.

marcone, 2023-05-28:
It's not clear to me what exactly failed. Did setup succeed? Is the car not seeing the drives? Did you try connecting to a PC, and does that PC see the drives? If setup finished you should be able to download various log files from the web interface. Attach those here.

dansche, 2023-05-29:
Hi Marcone, perfect I will attach whatever you need. As said I‘m on vacation this week and can give you some information on Thursday.

it‘s not finishing the first install, as it does not mount the drives. Let me post more infos when the issue will happen during install next Thursday when I am able to try it again. 

Thanks a lot

BR

Daniel

dansche, 2023-05-31:
so here now some more information as I am home today and can attach.

I do the installation on a 128GB SD Card on a Raspberry Zero W, I can flash the drive and I do edit the file "teslausb_setup_variables.conf" attached the file here as well.

If I enter "tail -f teslausb-headless-setup.log" to check the progress I get the following:

pi@teslausb:/boot $ tail -f teslausb-headless-setup.log
Wed 31 May 23:14:21 BST 2023 : Starting setup.
Wed 31 May 23:14:22 BST 2023 : curl -L https://github.com/marcone/teslausb/archive/main-dev.tar.gz
Wed 31 May 23:14:24 BST 2023 : Downloaded /root/bin/envsetup.sh ...
Wed 31 May 23:14:26 BST 2023 : Configuring the hostname...
Wed 31 May 23:14:30 BST 2023 : Configured hostname: teslausb
Wed 31 May 23:14:30 BST 2023 : Downloaded /tmp/506/setup-teslausb ...
Wed 31 May 23:14:30 BST 2023 : /root/bin/setup-teslausb is up to date
Wed 31 May 23:14:30 BST 2023 : Downloaded /tmp/506/rc.local ...
Wed 31 May 23:14:30 BST 2023 : rc.local is up to date
Wed 31 May 23:14:30 BST 2023 : Updating package index files...
Wed 31 May 23:17:24 BST 2023 : Verifying that the requested configuration is valid...
Wed 31 May 23:17:26 BST 2023 : Downloaded /tmp/verify-configuration.sh ...
Wed 31 May 23:17:26 BST 2023 : Checking XFS support
Wed 31 May 23:17:45 BST 2023 : XFS supported
Wed 31 May 23:17:45 BST 2023 : DATA_DRIVE is not set. SD card will be used.
Wed 31 May 23:17:45 BST 2023 : Verifying that there is sufficient space available on the MicroSD card...
Wed 31 May 23:17:45 BST 2023 : There is sufficient space available.
Wed 31 May 23:17:49 BST 2023 : Downloading common runtime scripts.
Wed 31 May 23:17:49 BST 2023 : Downloaded /root/bin/remountfs_rw ...
Wed 31 May 23:17:49 BST 2023 : Downloaded /root/bin/make_snapshot.sh ...
Wed 31 May 23:17:50 BST 2023 : Downloaded /root/bin/release_snapshot.sh ...
Wed 31 May 23:17:50 BST 2023 : Downloaded /root/bin/force_sync.sh ...
Wed 31 May 23:17:50 BST 2023 : Downloaded /root/bin/mountoptsforimage ...
Wed 31 May 23:17:50 BST 2023 : Downloaded /root/bin/mountimage ...
Wed 31 May 23:17:50 BST 2023 : Downloaded /root/bin/enable_gadget.sh ...
Wed 31 May 23:17:50 BST 2023 : Downloaded /root/bin/disable_gadget.sh ...
Wed 31 May 23:17:50 BST 2023 : /boot/cmdline.txt is up to date
Wed 31 May 23:17:55 BST 2023 : Downloaded /tmp/create-backingfiles-partition.sh ...
Wed 31 May 23:17:55 BST 2023 : create-backingfiles-partition: Checking existing partitions...
Wed 31 May 23:17:56 BST 2023 : create-backingfiles-partition: Modifying partition table for backing files partition...
Wed 31 May 23:18:30 BST 2023 : create-backingfiles-partition: Modifying partition table for mutable (writable) partition for script usage...
Wed 31 May 23:18:34 BST 2023 : create-backingfiles-partition: Writing updated partitions to fstab and /boot/cmdline.txt
Wed 31 May 23:18:35 BST 2023 : create-backingfiles-partition: Formatting new partitions...
Wed 31 May 23:19:36 BST 2023 : Mounting the partition for the backing files...

On Mounting it does not go further, I have attached the PI with a DataCable to the PC where I have made the shared folders and User for Tesla. So I made a Tesla user with Username: Tesla
Password: Tesla1234
TeslaCam Folder
TeslaMusic Folder
Gave both free to the user Tesla.

If I type in the Explorer \\192.168.1.205\ I can see the two folders as shared folder on the Network... as well the Tesla User has access.

The IP adress internal from the PC is 192.168.1.205

The Pi is connected to the WLAN otherwise I would not be able to connect to it with PuTTy.

I did delete the WLAN Password because of security reasons.
[teslausb_setup_variables.conf.txt](https://github.com/marcone/teslausb/files/11619282/teslausb_setup_variables.conf.txt)

I did attach the txt ending only to be able to attach the file, I upload it as .conf without the txt ending.

So there do I stand now.

Any Idea what I do wrong or not working?

For the installation I used the Tutorial on youtube from https://youtu.be/fnO7Md6TGHA

I want only music and teslacam as my 2021 has no boombox.

If you need more just let me know.