marcone/teslausb #767: shares not mounted

merlinfive, 2023-06-10:
### Describe the problem

during setup the shares seems to get mounted

Sat 10 Jun 14:09:03 CEST 2023 : verify-and-configure-archive: Verifying that the archive server 192.168.168.78 is reachable...
Sat 10 Jun 14:09:03 CEST 2023 : verify-and-configure-archive: The archive server is reachable.
Sat 10 Jun 14:09:04 CEST 2023 : verify-and-configure-archive: Verifying that the archive share is mountable...
Sat 10 Jun 14:09:04 CEST 2023 : verify-and-configure-archive: Trying mount command-line:
Sat 10 Jun 14:09:04 CEST 2023 : verify-and-configure-archive: mount -t cifs '//192.168.168.78/TeslaCam' '/tmp/archivetestmount' -o 'credentials=/tmp/teslaCamArchiveCredentials,iocharset=utf8,file_mode=0777,dir_mode=0777,vers=default,'
Sat 10 Jun 14:09:06 CEST 2023 : verify-and-configure-archive: The archive share is mountable using: mount -t cifs '//192.168.168.78/TeslaCam' '/tmp/archivetestmount' -o 'credentials=/tmp/teslaCamArchiveCredentials,iocharset=utf8,file_mode=0777,dir_mode=0777,vers=default,'
Sat 10 Jun 14:09:06 CEST 2023 : verify-and-configure-archive: Verifying that the archive share is mountable...
Sat 10 Jun 14:09:06 CEST 2023 : verify-and-configure-archive: Trying mount command-line:
Sat 10 Jun 14:09:06 CEST 2023 : verify-and-configure-archive: mount -t cifs '//192.168.168.78/TeslaMusic' '/tmp/archivetestmount' -o 'credentials=/tmp/teslaCamArchiveCredentials,iocharset=utf8,file_mode=0777,dir_mode=0777,vers=default,'
Sat 10 Jun 14:09:06 CEST 2023 : verify-and-configure-archive: The archive share is mountable using: mount -t cifs '//192.168.168.78/TeslaMusic' '/tmp/archivetestmount' -o 'credentials=/tmp/teslaCamArchiveCredentials,iocharset=utf8,file_mode=0777,dir_mode=0777,vers=default,'
Sat 10 Jun 14:09:07 CEST 2023 : verify-and-configure-archive: Configuring the archive...
Sat 10 Jun 14:09:07 CEST 2023 : verify-and-configure-archive: Configured the archive.

but after reboot not

root@teslausb df 
Filesystem     1K-blocks    Used Available Use% Mounted on
/dev/root        1998608 1181304    696064  63% /
devtmpfs          187200       0    187200   0% /dev
tmpfs             220176       0    220176   0% /dev/shm
tmpfs             220176    5968    214208   3% /run
tmpfs               5120       0      5120   0% /run/lock
tmpfs             220176       0    220176   0% /sys/fs/cgroup
tmpfs             220176       4    220172   1% /var/log/nginx
tmpfs             220176       0    220176   0% /var/lib/nginx
tmpfs             220176       0    220176   0% /var/lib/ntp
tmpfs             220176       0    220176   0% /var/tmp
tmpfs             220176       4    220172   1% /tmp
tmpfs             220176       0    220176   0% /var/spool
/dev/mmcblk0p4     95158    1564     86426   2% /mutable
/dev/mmcblk0p3 247766832 2327504 245439328   1% /backingfiles
/dev/mmcblk0p1    252045   49266    202779  20% /boot
tmpfs              44032       0     44032   0% /run/user/1000
root@teslausb :~#

Shares are located on Synology and correct accessable from Windows system with same credentials
[image]

### Device

Raspberry Pi Zero W

### OS Image

Prebuilt TeslaUSB image

### Car Model

Model S

### USB connection

Glove box

merlinfive, 2023-06-10:
tried with different user and password without special characters , same issue , ok from cli not during reboot

marcone, 2023-06-10:
Since you didn't include any logs/diagnostics I can't say for sure, but this sounds like: https://github.com/marcone/teslausb/wiki/FAQ#q-i-sshd-into-the-pi-but-am-not-seeing-my-network-share

merlinfive, 2023-06-10:
yes probably , I was reading just thru all closed issues and found 2 very similar , setup up from scratch right now

merlinfive, 2023-06-10:
ok , new setup done , and if I connect USB to my windows system the the "USB" drives , CAM,MUSIC and BOOMBOX are visible

My expectation was as soon as the Pi is online (wlan) he would check the network device and start syncing the music share to the pi which does not happen ?

merlinfive, 2023-06-10:
ok , just saw the sync starts after the next reboot 

Sat 10 Jun 19:30:52 CEST 2023: Ensured music drive is mounted.
Sat 10 Jun 19:30:52 CEST 2023: Syncing music from archive...
Sat 10 Jun 19:30:53 CEST 2023: rsync failed with error 0
Sat 10 Jun 19:30:54 CEST 2023: Copied 1 music file(s), deleted 0, skipped 0 previously-copied files, and encountered 0 errors.

eill check in car tomorrow (2023.12.300.1)

marcone, 2023-06-10:
> My expectation was as soon as the Pi is online (wlan) he would check the network device and start syncing the music share to the pi

That is how it should work, yes. My guess would be the Pi has a poor network connection. Check the web interface for network status and diagnostics.

closed 2023-06-10 by marcone