marcone/teslausb #773: Support for Light Show

KrzysztofHajdamowicz, 2023-08-17:
### Describe the problem

Tesla allows for having separate partition storing Light Shows.

It would be great to have ability to sync those shows from remote storage into USB Drive.
Also, it would be great to offer a switcher that manages to change file names from subdirectory into main directory.

Example file organization (as stated here: https://github.com/teslamotors/light-show):
```
# Directory structure needed by car functionality
.
./LightShow
./LightShow/lightshow.fseq
./LightShow/lightshow.{wav,mp3}
# Directory structure to store more shows and change active one via web interface
./CustomLightShows/
./CustomLightShows/RickRoll/lightshow.fseq
./CustomLightShows/RickRoll/lightshow.wav
```
Changing active LightShow should be done by copying files from chosen LightShow and overwriting files in `LightShow` directory.

### Device

Raspberry Pi 4

### OS Image

other

### Car Model

Model Y

### USB connection

Glove box

rmeden, 2023-11-02:
I wrote a script to do this (managed through a ssh command line).. works well.  I'll attach the file (remove *.txt extension needed for attachment) [set_show.txt](https://github.com/marcone/teslausb/files/13244731/set_show.txt)

On your archive host
- create a <musicshare>/lightshows directory (not /lightshow)
- copy this script to lightshows directory
- download some shows to <musicshare>/lightshows/showname/ (I use the zipfile name)
- Trigger Sync (via the webpage, or power on TeslaUSB)

SSH into TeslaUSB  (user pi, /home/pi home directory) (On Android, I use ConnectBot
- sudo /root/bin/remountfs_rw
- sudo mount /mnt/music
- cp /mnt/music/lightshows/set_show .
- chmod 755 set_show
- echo "alias sshow=/home/pi/set_show" >>~/.profile
- . .profile (<dot><space><dot>profile if you want to use the sshow without logging in again)

(hmmm typing these instructions, I wonder if I should have just called the script sshow :) )  It's changed over time.

To use it:
SSH into TeslaUSB  (user pi, /home/pi home directory)  (no root or rw needed)

pi@WhiteLightning:~ $ sshow
Select Show
0 ) disable lightshow
1 ) 2001 Space Odyssey
2 ) Disneyland's Main Street Electrical Parade - Man Collins
3 ) hells_bells - Florian Dageförde
4 ) Imperial March 2
5 ) Knight Rider lightshow - Leonardo Avella
6 ) mj---thriller
Choice?> 1
re-mounting drives
Successful
pi@WhiteLightning:~ $
`

KrzysztofHajdamowicz, 2023-12-07:
Tesla added support to multiple lightshows in holiday update -> Create directory `LightShow` and put shows files there.
https://twitter.com/notateslaapp/status/1732590927296757820

d-kar, 2023-12-21:
> Tesla added support to multiple lightshows in holiday update -> Create directory `LightShow` and put shows files there. https://twitter.com/notateslaapp/status/1732590927296757820

Just create LightShow folder in music sync folder and copy light show files in there (can have multiple .fseq and .wav file with different base name.
It works!

rmeden, 2023-12-21:
Of course they added this after I wrote my script :)

But good to know *how* to have the multiple shows.... the release notes didn't mention using a basename.(fseq|.wav)
I got the Christmas update yesterday, but haven't had time to test it yet.

marcone, 2024-01-15:
Closing since this can be done via music sync

closed 2024-01-15 by marcone