marcone/teslausb #774: Wiki update request: One-step Setup how to save a teslausb_setup_variables.conf using a mac

HugoCatarin0, 2023-08-22:
### Describe the problem

When using a mac to create the teslausb_setup_variables.conf file, it's impossible to save such file due to the lack of writing permissions. It would be great if the one-step setup guide could accomodate this scenario.

### Device

Raspberry Pi Zero 2 W

### OS Image

Prebuilt TeslaUSB image

### Car Model

Model Y

### USB connection

Glove box

### Logs

N/A

### Additional information

N/A

marcone, 2023-08-22:
What do you mean by "lack of writing permissions"? What _exactly_ are you trying to do that's not working?

HugoCatarin0, 2023-08-24:
Excuse me, just picked the topic now and realized I was simply using the wrong mac app to edit the teslausb_setup_variables.conf. I was using the standard text editor, which was preventing me from saving my file. However, after trying using visual studio code to edit the file it worked. :)

So this issue can be closed, it was just my noobness. :)

closed 2023-08-24 by HugoCatarin0