marcone/teslausb #776: [Meta] Radxa Rock Pi 4C Plus and TeslaUSB

KrzysztofHajdamowicz, 2023-09-17:
### Describe the problem

I have an idea to compile issues mentioned in multiple issues [example](https://github.com/marcone/teslausb/issues?q=is%3Aissue+rock) into single thread, allowing for faster searching.

### Device

Rock Pi 4C plus

### OS Image

Armbian

### Car Model

Model Y

### USB connection

Glove box

KrzysztofHajdamowicz, 2023-09-17:
So far, my progress is:
1. Get image https://github.com/radxa-build/rock-4c-plus/releases/download/20230312-1521/Armbian_23.05.0-trunk_Rockpi-4cplus_jammy_current_5.15.101_minimal.img.xz
2. Flash microSD card using that image
3. Plug ethernet cable to Rock Pi 4C+
4. Boot it
5. SSH into, using `root`/`1234` credentials
6. Run through initial configuration
7. use `nano` to create file thermals.dts
```
/dts-v1/;
/plugin/;

/ {
        compatible = "radxa,rockpi4c-plus", "radxa,rockpi4", "rockchip,rk3399";

        fragment@0 {
                target=<&tsadc>;
                __overlay__ {
                        status = "okay";
                        /* tshut mode 0:CRU 1:GPIO */
                        rockchip,hw-tshut-mode = <1>;
                        /* tshut polarity 0:LOW 1:HIGH */
                        rockchip,hw-tshut-polarity = <1>;
                };
        };

};
```
8. run `armbian-add-overlay thermals.dts`
9. reboot
10. Run TeslaUSB installer: `curl https://raw.githubusercontent.com/marcone/teslausb/main-dev/setup/generic/install.sh | bash`
11. Use /boot/teslausb_setup_variables.conf from TeslaUSB that was running in glovebox on Raspberry Pi 4B (so, 1:1 configuration migration)
12. Plug USB drive that was used by Raspberry Pi 4B from step 10.
12a. Iterate for a few times, fixing each error one by one:
12b. `mkdir /mnt/{cam,music,boombox}`
12c.
```
systemctl disable --now systemd-resolved
systemctl mask systemd-resolved
systemctl enable --now dnsmasq
```
12d. `mv /mutable/varlib/misc{,.old}`
12e. copy `/etc/wpa_supplicant/wpa_supplicant.conf` from Raspberry Pi 4B to Rock Pi 4C Plus
12f. Installer finally passes
13. Reboot
14.  `/backingfiles` and `/mutable` are mounted
15. unfortunately, USB OTG in upper USB3 port is not working. 
https://www.youtube.com/watch?v=ngMiCgQzwkI
dmesg:
```
[   46.419618] overlayfs: "xino" feature enabled using 2 upper inode bits.
[   46.803441] loop0: detected capacity change from 0 to 419428352
[   46.870567] exFAT-fs (loop0): Volume was not properly unmounted. Some data may be corrupt. Please run fsck.
[   47.067020] ------------[ cut here ]------------
[   47.067040] WARNING: CPU: 1 PID: 1528 at drivers/usb/dwc3/gadget.c:3649 dwc3_stop_active_transfer.part.0.constprop.0+0xb4/0xcc
[   47.067064] Modules linked in: nls_utf8 cifs cifs_arc4 cifs_md4 fscache netfs usb_f_mass_storage libcomposite exfat lz4hc lz4 zram binfmt_misc xfs sg btsdio snd_soc_hdmi_codec bluetooth snd_soc_simple_card snd_soc_rk817 snd_soc_simple_card_utils snd_soc_core brcmfmac snd_pcm_dmaengine hantro_vpu(C) rockchip_vdec(C) snd_pcm brcmutil rockchip_iep rockchip_rga v4l2_h264 snd_timer videobuf2_dma_contig dw_hdmi_i2s_audio dw_hdmi_cec v4l2_mem2mem videobuf2_dma_sg snd videobuf2_vmalloc cfg80211 videobuf2_memops videobuf2_v4l2 soundcore rfkill videobuf2_common videodev mc cpufreq_dt sch_fq_codel ramoops reed_solomon pstore_blk pstore_zone ip_tables x_tables autofs4 uas realtek dwmac_rk stmmac_platform stmmac pcs_xpcs
[   47.067247] CPU: 1 PID: 1528 Comm: disable_gadget. Tainted: G         C        5.15.101-rockchip64 #1
[   47.067256] Hardware name: Radxa ROCK Pi 4C+ (DT)
[   47.067261] pstate: 200000c5 (nzCv daIF -PAN -UAO -TCO -DIT -SSBS BTYPE=--)
[   47.067269] pc : dwc3_stop_active_transfer.part.0.constprop.0+0xb4/0xcc
[   47.067277] lr : dwc3_stop_active_transfer.part.0.constprop.0+0x78/0xcc
[   47.067285] sp : ffff80000d1abb20
[   47.067289] x29: ffff80000d1abb20 x28: ffff00000511ac40 x27: 0000000000000000
[   47.067300] x26: 0000000000000000 x25: 0000000000000000 x24: ffff000005237900
[   47.067312] x23: ffff0000052229b0 x22: 0000000000000000 x21: ffff000005222a78
[   47.067323] x20: 0000000000000000 x19: ffff0000053ad900 x18: ffffffffffffffff
[   47.067334] x17: 0000000000000000 x16: 0000000000000000 x15: 3d454d414e5f4344
[   47.067345] x14: ffff800009dcc4e0 x13: 0000000000000040 x12: 0000000000000000
[   47.067356] x11: ffffffffffff3f08 x10: ffff8000edff5000 x9 : ffffffffffff3f04
[   47.067368] x8 : 00004161a8e29e9d x7 : ffffffffffff3f00 x6 : ffffffffffff3f0c
[... 17 lines cut ...]
[   47.067539]  el0_svc_common.constprop.0+0x44/0xfc
[   47.067547]  do_el0_svc+0x28/0xa0
[   47.067554]  el0_svc+0x28/0x80
[   47.067564]  el0t_64_sync_handler+0xa4/0x130
[   47.067571]  el0t_64_sync+0x1a0/0x1a4
[   47.067579] ---[ end trace dfca4f5855ded11e ]---
[   47.186435] loop1: detected capacity change from 0 to 419430400
[   47.186735]  loop1: p1
[   47.263967] loop2: detected capacity change from 0 to 209715200
[   47.264289]  loop2: p1
[   47.481811] loop1: detected capacity change from 0 to 419428352
[   48.530664] Mass Storage Function, version: 2009/09/11
[   48.530689] LUN: removable file: (no medium)
[   48.542549] LUN: removable file: (no medium)
[   48.554082] LUN: removable file: (no medium)
[  107.841583] loop1: detected capacity change from 0 to 419428352
[  107.858035] exFAT-fs (loop1): Volume was not properly unmounted. Some data may be corrupt. Please run fsck.
[ 3588.566311] loop0: detected capacity change from 0 to 419430400
[ 3588.567791]  loop0: p1
[ 3588.771289] loop0: detected capacity change from 0 to 419428352
```
16. Wireless access point is not created

Diagnostics:
[diagnostics.txt](https://github.com/marcone/teslausb/files/12643420/diagnostics.txt)

marcone, 2023-09-17:
I haven't had time to try your steps yet, but I would suggest using a recent Armbian build with a newer kernel, e.g. Armbian_23.8.1_Rockpi-4cplus_jammy_current_6.1.50.img.xz from https://redirect.armbian.com/rockpi-4cplus/archive/

KrzysztofHajdamowicz, 2023-09-17:
I have tried it first. It does not see USB SSD (Samsung T7) and USB-NVME adapter with Samsung 980 Pro connected to USB 3.0 ports, only 2.0 ports see connected drive.

miles267, 2023-09-17:
I’ve just used apt-get update and apt-get upgrade as well as unfreezing kernel updates. That way it’ll ensure the kernel stays current regardless of the installed image. Also there is now a new beta coding UI you can install that’s a convenient option.

KrzysztofHajdamowicz, 2023-09-17:
> Also there is now a new beta coding UI you can install that’s a convenient option.

What do You mean? I'm new in Armbian world so discovering something new is fun :)

miles267, 2023-09-17:
Sorry. Please try this:

sudo apt update && sudo apt install armbian-config. You should then be able to run armbian-config to u freeze kernel installs and configure WLAN, etc. it’s the same as running nmtui.

KrzysztofHajdamowicz, 2023-09-17:
> Armbian_23.8.1_Rockpi-4cplus_jammy_current_6.1.50.img.xz from https://redirect.armbian.com/rockpi-4cplus/archive/

1. Flash image to microSD card
2. Boot
3. Do initial config
4. `apt update && apt install armbian-config`
5. use `armbian-config` to enable `dwc3-0-host` hardware overlay. SSD is now visible on lower USB3 port.
6. Also enable overlay `rockpi4cplus-usb-host`. After reboot, there is an error in dmesg
```
[    3.224357] dwc3 fe800000.usb: error -ETIMEDOUT: failed to initialize core
[    3.224499] dwc3: probe of fe800000.usb failed with error -110
```
7. Install TeslaUSB: `curl https://raw.githubusercontent.com/marcone/teslausb/main-dev/setup/generic/install.sh | bash`
8. Use /boot/teslausb_setup_variables.conf from TeslaUSB that was running in glovebox on Raspberry Pi 4B (so, 1:1 configuration migration)
9. `Mon Sep 18 12:11:09 AM CEST 2023 : STOP: this device (Radxa ROCK Pi 4C+) does not have a UDC driver` appeared in `/boot/teslausb-headless-setup.log`

Looks like #769 is present in current Armbian images.

marcone, 2023-09-17:
> 6\. Also enable overlay `rockpi4cplus-usb-host`. After reboot, there is an error in dmesg

The `rockpi4cplus-usb-host` overlay switches the top USB3 port to host mode (as described in /boot/dtb/rockchip/overlay/README.rockchip-overlays), thereby disabling the required OTG mode, which then leads to the error you see when trying to install TeslaUSB:

> 9\. `Mon Sep 18 12:11:09 AM CEST 2023 : STOP: this device (Radxa ROCK Pi 4C+) does not have a UDC driver` appeared in `/boot/teslausb-headless-setup.log`

KrzysztofHajdamowicz, 2023-09-18:
> The `rockpi4cplus-usb-host` overlay switches the top USB3 port to host mode (as described in /boot/dtb/rockchip/overlay/README.rockchip-overlays), thereby disabling the required OTG mode, which then [...]

overlay and SSD visibility:
[no-overlay.txt](https://github.com/marcone/teslausb/files/12645504/no-overlay.txt)
[dwc3-0-host rockpi4cplus-usb-host.txt](https://github.com/marcone/teslausb/files/12645508/dwc3-0-host.rockpi4cplus-usb-host.txt)
[dwc3-0-host.txt](https://github.com/marcone/teslausb/files/12645514/dwc3-0-host.txt)

dwc3-0-host overlay is the only overlay that gives SSD visibility in any of USB3 port (lower port, being specific).
Even with dwc3-0-host overlay, there is no UDC driver.
[image]

KrzysztofHajdamowicz, 2023-09-20:
https://imola.armbian.com/dl/rockpi-4cplus/archive/Armbian_23.8.1_Rockpi-4cplus_bookworm_current_6.1.50_minimal.img.xz

spiced with:
1. `apt install vim nano htop ncdu mc curl wget`
2. `wget https://github.com/radxa/overlays/raw/main/arch/arm64/boot/dts/rockchip/overlays/rk3399-dwc3-0-peripheral.dts -O rk3399-dwc3-0-peripheral.dts`
3. `armbian-add-overlay rk3399-dwc3-0-peripheral.dts`
4. `reboot`
5. after reboot:
```
# dmesg | grep -i dwc
[    2.730303] dwc3 fe900000.usb: error -ETIMEDOUT: failed to initialize core
[    2.730408] dwc3: probe of fe900000.usb failed with error -110
```

No SSD visibility in lower USB3 port, had to use USB2 port to make drive visible.
For testing purporses, teslausb installed using microSD card as backing storage to simplify device.

Unfortunately, whole rig is not visible in PC.
[diagnostics.txt](https://github.com/marcone/teslausb/files/12677593/diagnostics.txt)

KrzysztofHajdamowicz, 2023-09-22:
OK, I have managed to get something working:
1. Use image Armbian_23.8.1_Rockpi-4cplus_jammy_current_6.1.50.img.xz
2. `apt update && apt upgrade`
3. Run TeslaUSB installer: `curl https://raw.githubusercontent.com/marcone/teslausb/main-dev/setup/generic/install.sh | bash`
4. In teslausb_setup_variables.conf comment out SSID, WIFIPASS, AP_SSID, AP_PASS, AP_IP, 
[teslausb_setup_variables.conf.txt](https://github.com/marcone/teslausb/files/12703900/teslausb_setup_variables.conf.txt)
5. Perform configuration: `/etc/rc.local`

TeslaUSB works on Rock Pi.
Caveats:
1. After booting Rock Pi, devices appear and after few seconds disappear and re-appear after a second or two.
2. Lower USB3.0 port is not working, thus no USB3.0 for backing storage.

mjpcomp, 2024-03-08:
For anyone coming through here and noticing there's a Wiki now on how to install and get the 4C+ working:

Under the https://github.com/marcone/teslausb/wiki/Rock-Pi-4C--Installation#software-requirements Software Requirements section, the link to the image no longer works as that listed version has been archived. You can find that particular version here: https://imola.armbian.com/dl/rockpi-4cplus/archive/

mjpcomp, 2024-03-13:
> @mjpcomp the linked image was already archived, the problem is that that particular archive was deleted, as I'm sure the one you linked to will be deleted at some point too. This is because Armbian is switching to community images for the Rock Pi 4C+ due to lack of an official maintainer. The problem is that the community image does not have working OTG.

Ah, thanks for the heads up - that's definitely good to know... I'm going to make sure I keep an extra copy of this image around.. Wonder if it may need to be hosted on archive.org or something just in case it does go the way of the Dodo...

Crosenhain, 2024-03-13:
I found that the USB-OTG was unreliable, so I took a stab in the dark and removed the `usbstoragequirks` line from `/boot/armbianEnv.txt` and ran `update_initramfs -u`, rebooted, and it seems to be much more reliable from my initial testing. I don't use an external USB drive for storage, so ymmv. 

I still see the drives appear for a little bit, then go away, and then come back again. I suspect this might be due to the sync script running on startup but I haven't confirmed this yet.

ZhongDev, 2024-04-25:
> For anyone coming through here and noticing there's a Wiki now on how to install and get the 4C+ working:
>
> Under the [...]

The archive link path got changed.
The current like to the archive is now [https://imola.armbian.com/archive/rockpi-4cplus/archive/](https://imola.armbian.com/archive/rockpi-4cplus/archive/)
I wanted to leave it here in case someone else is looking for the version specified in the wiki.

ZhongDev, 2024-04-28:
@miles267 So far it seems to be mostly smooth for the past 3 days. (20 Drives, Sentry on, Disabled at home). There was one drive where the 4c+ failed to mount at the start of the drive, But I just replugged the USB in the glovebox and it has worked. I may try disabling "usbstoragequirks" like @Crosenhain so that hopefully it is more stable, and fewer failed mounts happen.

ZhongDev, 2024-04-28:
Also, Wifi signal has been okay, however I'm not sure if that is due to the 4c+ or my overbuilt Unifi wifi network.

gerculanum, 2024-04-28:
Older Rock 4C+ Armbian builds, including the one in the installation instructions, have a boot stability issue. This issue renders the board unusable in cars as it often fails to boot and register as USB storage. This has been fixed by https://github.com/armbian/build/pull/6414. Newer [Armbian builds](https://github.com/armbian/community/releases) should perform better. However, they require [this overlay](https://github.com/radxa/overlays/blob/fbd8024634a8c56c83bd7998e6a4ea1a0d2dad7a/arch/arm64/boot/dts/rockchip/overlays/rk3399-dwc3-0-peripheral.dts) to enable peripheral mode on the OTG port, which can be added using the `armbian-add-overlay` command.

TeslaUSB on a Rock 4C+ works well in my car, with no WiFi issues.

ZhongDev, 2024-04-28:
@gerculanum Thank you for that! I will try again soon using a new community build. I used an archived build because teslausb refused to install properly a community build. I think probably because i did not use the overlay.

Crosenhain, 2024-05-08:
> > I still see the drives appear for a little bit, then go away, and then come back again.
> 
> @Crosenhain , Does Tesla complain about this and show an alert/active alert?

I need to check this again, my car recently updated to 2024.14.3, which apparently did contain some fixes for USB drive problems according to a local Tesla service rep.

Crosenhain, 2024-05-11:
I do not have an alert - I think I did in the previous software version.

DriftingShadows, 2024-06-13:
> Older Rock 4C+ Armbian builds, including the one in the installation instructions, have a boot stability issue. This issue renders the board unusable in cars as it often fails to boot and register as USB storage. This has been fixed by [armbian/build#6414](https://github.com/armbian/build/pull/6414). Newer [Armbian builds](https://github.com/armbian/community/releases) should perform better. However, they require [this overlay](https://github.com/radxa/overlays/blob/fbd8024634a8c56c83bd7998e6a4ea1a0d2dad7a/arch/arm64/boot/dts/rockchip/overlays/rk3399-dwc3-0-peripheral.dts) to enable peripheral mode on the OTG port, which can be added using the `armbian-add-overlay` command.
> 
> TeslaUSB on a Rock 4C+ works well in my car, with no WiFi issues.

do you have a Model Y or 3? Are you also using the glovebox usb?

gerculanum, 2024-06-14:
> > TeslaUSB on a Rock 4C+ works well in my car, with no WiFi issues.
> 
> do you have a Model Y or 3? Are you also using the glovebox usb?

I'm using the glovebox USB in a Model 3.

DriftingShadows, 2024-06-18:
I can't seem to get my Rock Pi 4c+ to work with Model Y glovebox with USB3 (to avoid the warning about glovebox usb) in any configuration. The issue is that the Rockpi just doesnt get picked up by the Model Y (DashCam is unavailable in the settings). I've tried various combinations of the following with no luck:

- Using newer armbian builds with the overlay
- Using USB3 hub
- Using USB3 Y cable

There are ways to "trick" the car to pick up teslausb but these are mostly temporary as far as i can tell and eventually becomes unavailable again. For example, while the Rockpi is power, I plugged it into my laptop to make sure its mounted, then unplug from laptop and plug into USB hub connected to the glove box usb. Someone might be able to explain this phenomenon but it doesnt last long,  or maybe it never actually works and is just visual.

I can take a USB3 configuration that works 100% time on a Model 3 and plug it into a Model Y and it wont work. I've tested on 2 Model Y's and 2 Model 3's.

Only thing I've gotten to work reliably just using USB2, but then that kinda defeats the whole purpose of using a Rockpi and could just go with a smaller raspberry pi.

If anyone has any insight i'm all ears.

xybertech, 2024-07-14:
Have you checked another USB3 device, like an SSD to confirm it works? Does the RockPi power on or do you get nothing at all? 

Tesla Y manual says it needs to be USB2 or USB3 that is compatible with USB2 and sustains 4MB/s sustained. After I read that I abandoned the thought of a Rock Pi. 
https://www.tesla.com/ownersmanual/modely/en_us/GUID-F311BBCA-2532-4D04-B88C-DBA784ADEE21.html

jheyneman, 2024-07-28:
Figured I'd post here in case anyone else is running into the same issue... my RockPi 4C+ had been working fine since May until a Tesla update, it would only mount the drive if I did a scroll wheel reset.  I  figured I'd try to run the teslausb updater and it would no longer boot.

Using either images listed here or in the Wiki, the first time `apt upgrade` is run, it will no longer boot.  Connecting the RockPi to HDMI just shows a U-Boot screen. 

I flashed the latest Armbian image.  It appears NetworkManager isn't the default, instead interfaces are managed by systemd-networkd.   I was able to install NetworkManager, switch it to the default, and get WiFi working.   I also added the overlay a few comments above and was able to install teslausb.  

Hopefully this is a decent starting point for anyone encountering issues, if it works for me over the next week or so, I'll try to write up a more detailed guide.  If it doesn't work, I'll be sure to at least update this comment.

nmiller0113, 2024-10-01:
> Older Rock 4C+ Armbian builds, including the one in the installation instructions, have a boot stability issue. This issue renders the board unusable in cars as it often fails to boot and register [...]

@gerculanum hopefully you still see messages from this old thread. I'm curious if you still knew what release you were using. I tried getting things going with this latest trunk https://github.com/armbian/community/releases/tag/24.11.0-trunk.167, and I was able to install it and compile the overlay. Were there other steps you took before installing teslausb? Right now I cannot get wifi to authenticate, keeps timing out even though it see's my wifi no problem. This was not an issue with the version linked in the wiki...though I did have the boot stability issue on that version. If you wouldn't mind sharing any other steps you took before installing teslausb, that would be awesome. Or link me to where you found it. Thanks!

jheyneman, 2024-10-01:
@nmiller0113 I found that using the minimal image, there is some firmware or something missing.  I also could not authenticate, and then once I got the extra packages installed, it failed boot and showed a U-Boot screen. I think the boot partition ran out of space.  

About a month ago, I tried using the full image, obviously ran into issues with the install script since it's set to 3G partition.  I edited it to 10G, ran the script and followed the rest of the normal instructions.  It has been running and copying footage to my NAS with no issue for the better part of a month.

nmiller0113, 2024-10-01:
Thanks @jheyneman! What extra packages did you install? If I use the image listed wiki instructions I have boot issues as @gerculanum stated were fixed in later releases. I'm not sure why you're not having the same boot issues, as it doesn't seem to be a boot partition size problem from what was stated...but maybe I misunderstood.

This new version, albeit minimum install image, is super stable...I just can't get my wifi to authenticate and don't know which packages are missing to do so. That's why I was hoping that @gerculanum may see this and be able to assist as it sounds like he's got it running.

nmiller0113, 2024-10-01:
@jheyneman oh wait...I just noticed you edited your original response from full image (which I assumed was the one on the wiki) to desktop image. Where did you find that? Thanks again!