marcone/teslausb #786: [Feature Request] Keep Car Awake using TeslaFi API

THX723, 2023-10-14:
### Describe the problem

TeslaFi also provides APIs to keep the car alive (and sleep attempt) during archiving. For TeslaFi users, this is a safer alternative than using direct Tesla API and risk exposing the Tesla API in the config file.

The TeslaFi API are simple and as follows:

**Wake Vehicle**
https://www.teslafi.com/feed.php?token=INSERT_TESLAFI_TOKEN_HERE&command=wake

Start a sleep attempt
https://www.teslafi.com/feed.php?token=INSERT_TESLAFI_TOKEN_HERE&command=sleep

TeslaFi API Token can be generated from Settings -> Tesla API -> API Token (tab). It can be manually revoked at anytime.

### Device

Raspberry Pi 4

### OS Image

Prebuilt TeslaUSB image

### Car Model

Model S

### USB connection

Center console

THX723, 2023-10-14:
I did a quick peek at how TeslaUSB keep alive was implemented. Looks like it is simply enabling Sentry Mode during archive and then disabling Sentry Mode when complete. In that case, here's the TeslaFI API to coincide:

> **Sentry Mode ON**
> https://www.teslafi.com/feed.php?token=INSERT_TESLAFI_TOKEN_HERE&command=set_sentry_mode&sentryMode=true
> 
> **Sentry Mode OFF**
> https://www.teslafi.com/feed.php?token=INSERT_TESLAFI_TOKEN_HERE&command=set_sentry_mode&sentryMode=false
>

marcone, 2023-10-14:
It's a good idea, but I don't use TeslaFi or even the "keep awake" feature, so this likely won't get implemented unless someone contributes the code for it.

THX723, 2023-10-14:
Okay. I might take a stab at it. All others interested please feel free to reach out and help contribute as well.

THX723, 2023-10-16:
@marcone I'm done with the implentation. Tested okay for me. Shall I proceed with the Pull-request?

marcone, 2023-10-16:
Sure. Be sure to make it optional and backwards compatible (the 2 existing methods should still be available and work without any config changes needed) and document it.

THX723, 2023-10-16:
Yup. Did exactly that.

miles267, 2023-10-19:
Did I follow correctly that this new capability has already been merged into the main-dev?  How do you use it within the config file if updating an existing, working SD card image?

THX723, 2023-10-19:
> Did I follow correctly that this new capability has already been merged into the main-dev? How do you use it within the config file if updating an existing, working SD card image?

Yes, this capability has merged into main-dev and ready to rock 'n roll.
You'll need to re-configure your device using the latest .conf file [(Here)](https://github.com/marcone/teslausb/blob/main-dev/pi-gen-sources/00-teslausb-tweaks/files/teslausb_setup_variables.conf.sample). Follow the directions in the .conf file. Here's an excerpt.

```
# TeslaFi Subscriber?
#   If you're a TeslaFi user, you may consider keeping the car wake using
#   TeslaFi API instead. The advantage is not having to expose Tesla's Token
#   and other info in this file as plain text, which is a vulnerability (as
#   noted above. 'PIN to Drive' is not necessary in this case.
#   The TeslaFi API token can easily be revoked and a new one generated at
#   anytime. TeslaFi also allows selective command(s) enablement (e.g. disable
#   uneeded commands), further decreasing vulnerability.
# 
# GENERATE/REVOKE TESLAFI API TOKEN: Log into your TeslaFi account.
#   Navigate to Settings -> Tesla API -> API Token (tab)
#
# ENABLE/DISABLE TESLAFI API COMMANDS: Verify the 'Resume Polling' (wake)
#   command is enabled. Do not confused this with the 'Wake Vehicle' command.
#   Navigate to Settings -> Tesla API -> Commands (tab)
#
# NOTE: Both APIs should NOT be used at the same time. Choose only one.
#
# export TESLA_REFRESH_TOKEN='YOUR_REFRESH_TOKEN_GOES_HERE'
# export TESLAFI_API_TOKEN='YOUR_TESLAFI_API_TOKEN_GOES_HERE'

```

Be sure to comment the Tesla API variables, if you were using that previously.

```
# TESLA API ONLY (Do not configure if using TeslaFi API):
# If you have more than one Tesla vehicle on the account, you must also provide
# the name or VIN for the vehicle for which this TeslaUSB install will be used,
# so that TeslaUSB can keep the correct vehicle awake.
# If you only have one car on the account, you do not need to specify this.
# export TESLA_NAME="Elon's CyberTruck"
# export TESLA_VIN=5YJ3E1EA4JF000001
#
# TESLA API ONLY (Do not configure if using TeslaFi API):
# By default, TeslaUSB uses a combination of APIs to keep the car awake,
# similar to how to the Tesla app can wake up the car.
# If you find your car isn't staying awake, you can try setting the
# following to "sentry" to keep the car awake by temporarily turning
# on Sentry mode while archiving.
# export TESLA_WAKE_MODE=stream
```

marcone, 2023-10-20:
Closing issue since this is now implemented.

closed 2023-10-20 by marcone