marcone/teslausb #794: 'curl -L https://github.com/marcone/teslausb/archive/main-dev.tar.gz' failed, retrying

Roaders, 2023-10-22:
### Describe the problem

My teslausb-headless-setup.log log is just filled with failed attempts to curl the tar file from github.

### Device

Raspberry Pi Zero W

### OS Image

Prebuilt TeslaUSB image

### Car Model

Model Y

### USB connection

Glove box

### Logs

```
pi@teslausb:/boot $ sudo /root/bin/setup-teslausb diagnose
already running
```
```
pi@teslausb:/boot $ tail teslausb-headless-setup.log
Sun 22 Oct 16:28:15 BST 2023 : 'curl -L https://github.com/marcone/teslausb/archive/main-dev.tar.gz' failed, retrying
Sun 22 Oct 16:28:19 BST 2023 : 'curl -L https://github.com/marcone/teslausb/archive/main-dev.tar.gz' failed, retrying
Sun 22 Oct 16:28:23 BST 2023 : 'curl -L https://github.com/marcone/teslausb/archive/main-dev.tar.gz' failed, retrying
Sun 22 Oct 16:28:27 BST 2023 : 'curl -L https://github.com/marcone/teslausb/archive/main-dev.tar.gz' failed, retrying
Sun 22 Oct 16:28:31 BST 2023 : 'curl -L https://github.com/marcone/teslausb/archive/main-dev.tar.gz' failed, retrying
```

### Additional information

DNS seems to be working ok:

```
pi@teslausb:/boot $ nslookup github.com
Server:         192.168.86.140
Address:        192.168.86.140#53

Non-authoritative answer:
Name:   github.com
Address: 140.82.121.3
```

I read this thread:
https://github.com/marcone/teslausb/issues/211
which seems similar but that seemed to be an issue with the time being incorrect. The time on my pi seems fine:

```
pi@teslausb:/boot $ timedatectl status
               Local time: Sun 2023-10-22 16:30:55 BST
           Universal time: Sun 2023-10-22 15:30:55 UTC
                 RTC time: n/a
                Time zone: Europe/London (BST, +0100)
System clock synchronized: yes
              NTP service: active
          RTC in local TZ: no
```

closed 2023-10-22 by Roaders

marcone, 2023-10-22:
Not sure what happened, but there have been a few reports of this recently. I wonder if GitHub was doing maintenance or throttling downloads or something. I added some additional logs in case it happens again.

Roaders, 2023-10-22:
I did manage to download the file that it was trying to download on my windows pc whilst I was still getting the failure message on the pi. As I said it did start working, I then had to re-flash the SD card and start the setup again and that time the file downloaded first time I think.

Thanks for your support and responsiveness.

jtamboli, 2024-09-27:
I've gotten these errors a few times today while setting up a new install. I took a look at /tmp/curl.err, and the contents are:
```
curl: (23) Failure writing output to destination
```

The failure seems to be intermittent. Maybe some kind of race condition?

marcone, 2024-09-27:
@jtamboli What linux image were you using? Googling a bit for that error message seems to show it can be caused by using a wrong version of curl, and is solved by doing a manual "apt install curl" first.