marcone/teslausb #796: Tesla_apy.pi failing to install from fresh installation

TarkUK, 2023-10-26:
### Describe the problem

Fresh install of Tesla USB

in the teslausb_setup_variable.conf i have the below entered

export TESLA_EMAIL=
export TESLA_PASSWORD=
export TESLA_REFRESH_TOKEN=
export TESLA_WAKE_MODE=stream

i copy the conf file over to the sd card and insert in to the PI and power up. looking at the logs tesla_apy.pi fails to install or run

### Device

Raspberry Pi 4

### OS Image

Prebuilt TeslaUSB image

### Car Model

Model X

### USB connection

Center console

### Logs

Thu 26 Oct 14:03:26 BST 2023 : configure: Installing tesla_api.py
Thu 26 Oct 14:03:26 BST 2023 : Downloaded /root/bin/tesla_api.py ...
Thu 26 Oct 14:03:39 BST 2023 : configure: Failed to mount /mutable
Thu 26 Oct 14:05:31 BST 2023 : configure: tesla_api.py setup failed

if i ssh into the PI the mount is there

marcone, 2023-10-26:
> export TESLA_EMAIL=
> export TESLA_PASSWORD=

These are no longer used.

> export TESLA_REFRESH_TOKEN=

Did you actually put a refresh token there? Because the above isn't going to work. Please follow the [documentation in the config file](https://github.com/marcone/teslausb/blob/463f38218ef2ba84869010bb241c17d4bc45681d/pi-gen-sources/00-teslausb-tweaks/files/teslausb_setup_variables.conf.sample#L242C18-L242C18).

> export TESLA_WAKE_MODE=stream

You don't need to specify this as it's the default.

TarkUK, 2023-10-26:
Hello

i did put the refresh token there in addition to the  other fields.

the reason i have tried the above is the Tesla Mod X is not staying awake

i have tried export TESLA_WAKE_MODE=stream, and have tried the same but with sentry

looking at the diagnostic logs its adds a line Not Keep Tesla Awake so i am a bit stuck hence reading the other post here that suggests adding Tesla Email / Passoword.

At the moment my car does not stay away hence not mounting the drive to record

TarkUK, 2023-10-26:
event with only the token added with nothing else, tesla_apy.py still fails if that makes a difference

TarkUK, 2023-10-26:
will give that a shot now
with the quote are you specifing this   "   at the start and the end

TarkUK, 2023-10-26:
just done a fresh install still get the error during setup however after all is done i have run the command you specified and it does output the details of my car

TarkUK, 2023-10-26:
should i do another fresh install but the the quoation marks " start and end "

marcone, 2023-10-26:
Single quotes are usually better e.g.
```
export TESLA_REFRESH_TOKEN='eyJhbGciOiJ........2F1dGgyL3YzL3Rva2'
```
(the real token will be much longer, probably over a thousand characters)

marcone, 2023-10-26:
oh, and you shouldn't need to do a fresh install every time. You can just reboot and it should resume/retry installation.

TarkUK, 2023-10-26:
so if i do  

sudo -i
source bin/envsetup.sh
echo $TESLA_REFRESH_TOKEN
Then run bin/tesla_api.py list_vehicles

and the output i get is my car, am i correct in assuming that its working ?

TarkUK, 2023-10-26:
ok will looks like its working will go and plug in  to car and see how it goes

thank you for all your help, will let you know what happens

TarkUK, 2023-10-26:
Plugged the Pi in to the car and waited for it to boot and the recording icon to appear on the screen which it did.

Went away for dinner, came back to the car with out the keys say 15min later and the Pi was powered off, got the keys walked to the car and it woke up went in to the car and the PI was booting up, Car reported 1 sentry event however when i go in to the app there is no event recorded

marcone, 2023-10-26:
It's not clear to me whether you took the car to go to dinner or left the car at home while you went away.

In any case, TeslaUSB only uses the Tesla API to keep the car awake during archiving. When Sentry mode is enabled, the car itself should keep the USB ports powered, and therefore the Pi should be on as long as Sentry mode is on.

TarkUK, 2023-10-26:
Hi Sorry 

i should have send the car was parked at home with Sentry on

had dinner came back to car and the PI was powered off

TarkUK, 2023-10-26:
2020

marcone, 2023-10-26:
Hmm, I don't understand what's going on then. The car should keep the USB port powered while Sentry is on (because without power even a regular USB drive wouldn't work), and yours apparently doesn't.  That is unrelated to using tesla_api.py to keep the car awake during archiving.

TarkUK, 2023-10-26:
Defintly a very strange one unless something has been introduced in the latest tesla update to the car

merlinfive, 2023-10-27:
> so if i do
> 
> sudo -i source bin/envsetup.sh echo $TESLA_REFRESH_TOKEN Then run bin/tesla_api.py list_vehicles
> 
> and the output i get is my car, am i correct in assuming that its working ?

this is what I get :-( when I do the commands from top

root@teslausb:~# bin/tesla_api.py list_vehicles
Traceback (most recent call last):
  File "/root/bin/tesla_api.py", line 596, in <module>
    main()
  File "/root/bin/tesla_api.py", line 572, in main
    _get_id()
  File "/root/bin/tesla_api.py", line 185, in _get_id
    result = list_vehicles()
  File "/root/bin/tesla_api.py", line 269, in list_vehicles
    return _execute_request(base_url, None, None, False)
  File "/root/bin/tesla_api.py", line 78, in _execute_request
    json_response = _rest_request(url, method, data)
  File "/root/bin/tesla_api.py", line 105, in _rest_request
    'Authorization': 'Bearer {}'.format(_get_api_token()),
  File "/root/bin/tesla_api.py", line 163, in _get_api_token
    tesla.refresh_token()
  File "/usr/local/lib/python3.9/dist-packages/teslapy/__init__.py", line 249, in refresh_token
    super(Tesla, self).refresh_token(token_url, **kwargs)
  File "/usr/local/lib/python3.9/dist-packages/requests_oauthlib/oauth2_session.py", line 452, in refresh_token
    self.token = self._client.parse_request_body_response(r.text, scope=self.scope)
  File "/usr/local/lib/python3.9/dist-packages/oauthlib/oauth2/rfc6749/clients/base.py", line 427, in parse_request_body_response
    self.token = parse_token_response(body, scope=scope)
  File "/usr/local/lib/python3.9/dist-packages/oauthlib/oauth2/rfc6749/parameters.py", line 441, in parse_token_response
    validate_token_parameters(params)
  File "/usr/local/lib/python3.9/dist-packages/oauthlib/oauth2/rfc6749/parameters.py", line 448, in validate_token_parameters
    raise_from_error(params.get('error'), params)
  File "/usr/local/lib/python3.9/dist-packages/oauthlib/oauth2/rfc6749/errors.py", line 399, in raise_from_error
    raise cls(**kwargs)
oauthlib.oauth2.rfc6749.errors.LoginRequired: (login_required) Login required

TarkUK, 2023-10-27:
did you add the token refresh in to the setup file

merlinfive, 2023-10-27:
yes ,

export TESLA_REFRESH_TOKEN='eyJhbGciOiJSUzI1NiIsInR5cCI6Ikp....snip.....emrhH-hyGw'

and afterwards I started the bin/setup-teslausb , reboot to bring it into running config

TarkUK, 2023-10-27:
on my line i used these quotation marks "

then did a fresh install

merlinfive, 2023-10-27:
I saw a comment to use singe quotes but will give it a try

TarkUK, 2023-10-27:
yes true but i used " before i saw that comment but it worked

merlinfive, 2023-10-27:
same result :-(

TarkUK, 2023-10-27:
im see this line in the diagnostics

Fri 27 Oct 13:18:47 BST 2023: Failed to contact car using Tesla API, retrying in 5s...

TarkUK, 2023-10-27:
Another update, so i have remove the Tesla_api comments in the setup file and moved to Tesla FI API,

ran the install and i can see the wake call within Tesla FI web portal so its working, however after about 5 min the Pi Powers off even with Sentry being turned on.

not sure if this normal behaviour or not

marcone, 2023-10-27:
> ran the install and i can see the wake call within Tesla FI web portal so its working, however after about 5 min the Pi Powers off even with Sentry being turned on.
> 
> not sure if this normal behaviour or not

No, when Sentry mode is on the car is supposed to keep the USB ports powered. As I mentioned before, that's independent of using tesla_api.py to keep the car awake during archiving.

TarkUK, 2023-10-27:
Really don't know what the problem is, has to be a software issue with the tesla not sure if anyone else with a X is having the same

THX723, 2023-10-28:
@TarkUK What firmware version is your Model X current running?

As a side note (nothing to do with your issue), I've noticed my Model S on 2023.27.7 now goes to sleep after 12-15 minutes of park/idle, despite a 3rd-party service like TeslaFi polling away every second. The constant polling used to be enough to keep the car awake. I'm wondering if Tesla's made some changes recently, as they prepare for an official launch of their 3rd-party API support (which is currently under testing). With all that said, I'm going to have to make an update to the TeslaFi implementation to use the full 'wake_up' command to keep the car alive. I ran a private test version of that last night and it proved to be working well. Better even, since the car tends to go to sleep on the avg. 5 minutes after archive ends; before it could take as much as 30 minutes.

TarkUK, 2023-10-28:
I am running 2023.32.9

TarkUK, 2023-10-28:
I have tried soft reset make no difference and I have the pi directly connected to the car but I have also tried a backup battery pack and while that keeps the Pi alive the port in the car just stops over time

THX723, 2023-10-28:
What is your car's state of charge?

TarkUK, 2023-10-28:
Always above 20%  currently st 87

THX723, 2023-10-29:
Try using another USB-A to USB-C cable. I'm wondering if the port is shutting down, because you have one of those cheap/early USB-A to 0-C cable without the 56k pull-up resistor.

TarkUK, 2023-10-29:
Must have tried 6 usb cables. Whichbone you recommend from amazon

TarkUK, 2023-10-29:
Just ordered one comes Tuesday

THX723, 2023-10-30:
Actually, before you go swapping cable, try using a simple USB Stick and see if that works without any issues (e.g. not powered off and captures recordings). If that fails too, then it definitely you car, which you should flag for service.

THX723, 2023-10-31:
btw, this is the one I use: https://www.amazon.com/dp/B0BVQSC9PC?psc=1&ref=ppx_yo2ov_dt_b_product_details

miles267, 2024-02-08:
@marcone the issue may be a bit more complicated.  Opened a new bug ticket for Tesla API access failing with token refresh.  Seems they changed the Tesla API.

marcone, 2024-02-15:
> bin/tesla_api.py list_vehicles, it returned the error:

This should work again for now. I did have to delete /mutable/cache.json and /mutable/tesla_api.json, and create a new refresh token. Something like this:
```
sudo -i
source bin/envsetup.sh
rm /mutable/cache.json /mutable/tesla_api.json
export TESLA_REFRESH_TOKEN='<new token goes here>'
/root/bin/tesla_api.py list_vehicles
```

rworne, 2024-03-01:
> This should work again for now. I did have to delete /mutable/cache.json and /mutable/tesla_api.json, and create a new refresh token. Something like this:
>
> ```
> sudo -i
> source bin/envsetup.sh
> rm [...]

I tried this and it fixed the issue for me.  It's keeping the car awake, even after a reboot.

rworne, 2024-08-09:
I found a way to fix the Tesla API from failing to keep the car awake from the token expiring and not refreshing.

It's a hack, but it does the job.
in /root/bin/awake_start

find the function ping (line 11)

Change it from this:
```
function ping {
  while true
  do
    if [[ ( -n "${TESLA_REFRESH_TOKEN:+x}" ) ]]
    # Use Tesla API to keep car awake
    then
      if timeout 10 /root/bin/tesla_api.py streaming_ping
      then
        sleep 90
      else
        log "Failed to contact car using Tesla API, retrying in 5s..."
        sleep 5
      fi
    fi
```
to this:
```
function ping {
  while true
  do
    if [[ ( -n "${TESLA_REFRESH_TOKEN:+x}" ) ]]
    # Use Tesla API to keep car awake
    then
      if timeout 10 /root/bin/tesla_api.py streaming_ping
      then
        sleep 90
      else
        log "Failed to contact car using Tesla API, retrying in 5s..."
        sleep 5
        log "Clearing cache.json and tesla_api.json..."
        rm /mutable/cache.json /mutable/tesla_api.json
      fi
    fi
```

Now if the API call fails, it deletes the cached value - Tesla_API.py picks up on this and picks up a fresh token.  Car will now stay awake and back up all the files.

Log output.  Note the car stayed awake for 30+ minutes to finish the archiving (it's also charging):
```
Fri  9 Aug 11:51:10 PDT 2024: made all links for /tmp/snapshots/snap-000482
Fri  9 Aug 12:08:37 PDT 2024: Archive is reachable.
Fri  9 Aug 12:08:37 PDT 2024: Trying to set time...
Fri  9 Aug 12:08:37 PDT 2024: Time adjusted by 0.198245 seconds after 0.200000 seconds
Fri  9 Aug 12:08:37 PDT 2024: Starting background task to keep car awake.
**Fri  9 Aug 12:08:38 PDT 2024: Failed to contact car using Tesla API, retrying in 5s...
Fri  9 Aug 12:08:43 PDT 2024: Clearing cache.json and tesla_api.json...**
Fri  9 Aug 12:08:57 PDT 2024: taking snapshot of cam disk in /backingfiles/snapshots/snap-000483
Fri  9 Aug 12:08:59 PDT 2024: took snapshot
Fri  9 Aug 12:08:59 PDT 2024: comparing new snapshot with /backingfiles/snapshots/snap-000482/snap.bin
Fri  9 Aug 12:08:59 PDT 2024: making links for /tmp/snapshots/snap-000483, retargeted to /backingfiles/snapshots/snap-000483/mnt
Fri  9 Aug 12:09:22 PDT 2024: made all links for /tmp/snapshots/snap-000483
Fri  9 Aug 12:09:22 PDT 2024: low space, deleting /backingfiles/snapshots/snap-000439
Fri  9 Aug 12:09:24 PDT 2024: low space, deleting /backingfiles/snapshots/snap-000440
Fri  9 Aug 12:09:25 PDT 2024: Archiving...
Fri  9 Aug 12:09:25 PDT 2024: Ensuring cam archive is mounted...
Fri  9 Aug 12:09:25 PDT 2024: Mounting /mnt/archive...
Fri  9 Aug 12:09:27 PDT 2024: Mounted /mnt/archive.
Fri  9 Aug 12:09:27 PDT 2024: Ensured cam archive is mounted.
Fri  9 Aug 12:09:27 PDT 2024: Ensuring music archive is mounted...
Fri  9 Aug 12:09:27 PDT 2024: Mounting /mnt/musicarchive...
Fri  9 Aug 12:09:28 PDT 2024: Mounted /mnt/musicarchive.
[... 50 lines cut ...]
Fri  9 Aug 12:38:39 PDT 2024: Failed to mount /mnt/music.
Fri  9 Aug 12:38:39 PDT 2024: Sleeping before retry ...
Fri  9 Aug 12:38:40 PDT 2024: Retrying (1) 'ensure_mountpoint_is_mounted /mnt/music' ...
Fri  9 Aug 12:38:40 PDT 2024: Mounting /mnt/music...
mount: /mnt/music: failed to setup loop device for /backingfiles/music_disk.bin.
Fri  9 Aug 12:38:40 PDT 2024: Failed to mount /mnt/music.
Fri  9 Aug 12:38:40 PDT 2024: Sleeping before retry ...
Fri  9 Aug 12:38:41 PDT 2024: Retrying (2) 'ensure_mountpoint_is_mounted /mnt/music' ...
Fri  9 Aug 12:38:41 PDT 2024: Mounting /mnt/music...
Fri  9 Aug 12:38:41 PDT 2024: Mounted /mnt/music.
Fri  9 Aug 12:38:41 PDT 2024: Ensured music drive is mounted.
Fri  9 Aug 12:38:41 PDT 2024: Syncing music from archive...
Fri  9 Aug 12:39:08 PDT 2024: Copied 0 music file(s), deleted 0, skipped 3456 previously-copied files, and encountered 0 errors.
Fri  9 Aug 12:39:08 PDT 2024: Trimming free space in /mnt/music, which has 36 extents
Fri  9 Aug 12:39:08 PDT 2024: Trim complete, image now has 36 extents
Fri  9 Aug 12:39:08 PDT 2024: Unmounting /mnt/music...
Fri  9 Aug 12:39:08 PDT 2024: Unmounted /mnt/music.
Fri  9 Aug 12:39:08 PDT 2024: Finished copying music.
Fri  9 Aug 12:39:08 PDT 2024: unmounting /mnt/archive
Fri  9 Aug 12:39:08 PDT 2024: Truncating log...
**Fri  9 Aug 12:39:08 PDT 2024: Stopping wake car background task.**
Fri  9 Aug 12:39:08 PDT 2024: Connecting usb to host...
Fri  9 Aug 12:39:08 PDT 2024: Connected usb to host.
Fri  9 Aug 12:39:14 PDT 2024: Waiting for archive to be unreachable...
```

marcone, 2026-01-20:
The Tesla API no longer works so this issue is obsolete.