marcone/teslausb #803: Custom car lock sound

KrzysztofHajdamowicz, 2023-12-17:
### Describe the problem

2023 Holiday Update in Tesla firmware allows to change default "Honk on car close" sound to custom lock sound.
To do that, user have to put `LockChime.wav` file in `/` of any USB drive inserted to the car.
File have to be smaller than 1MiB.

Idea: add file upload form into web interface so file can be uploaded without detaching TeslaUSB from car.

### Device

Raspberry Pi 4

### OS Image

Prebuilt TeslaUSB image

### Car Model

Model Y

### USB connection

Glove box

BenWoodford, 2023-12-19:
Further idea: allow uploading multiple, and scheduling different sounds either every x minutes or using the time of day (I don’t want a screaming goat waking my neighbours at 1am)

d-kar, 2023-12-21:
You can do this by just copying LockChime.wav to music sync folder. Just tried it and it works!

everitj, 2023-12-22:
Yes and the Custom Light shows can be added in this way to the music folder under LightShow. Works a treat.

Helvio88, 2023-12-23:
I created something that I can run on crontab every hour or so, to randomize a WAV file for the next time it locks. I'm sure it can be refined, but here's a rough idea:

```bash
#!/bin/bash -eu

# Run disable_gadget.sh script
/root/bin/disable_gadget.sh &> /dev/null || :

# Mount /mnt/music
mount /mnt/music &> /dev/null || :

# Set the source and destination folders
source_folder="/mnt/music/"
destination_file="/mnt/music/LockChime.wav"

# Change to the source folder
cd "$source_folder" || exit

# Find all .wav files except LockChime.wav
files=($(find . -maxdepth 1 -type f -name "*.wav" ! -name "LockChime.wav"))

# Check if there are any eligible files
if [ ${#files[@]} -eq 0 ]; then
    echo "No eligible files found in $source_folder."
    exit 1
fi

# Get the total number of files
num_files=${#files[@]}

# Generate a random index
random_index=$((RANDOM % num_files))

# Choose a random file from the list
random_file=${files[random_index]}

# Copy the chosen file to the destination, overwriting the existing LockChime.wav
cp -f "$random_file" "$destination_file"

echo "Random file '$random_file' copied to '$destination_file'."

# Unmount /mnt/music
umount /mnt/music &> /dev/null || :

# Run enable_gadget.sh script
/root/bin/enable_gadget.sh &> /dev/null || :
```

Helvio88, 2023-12-23:
@marcone - I'm sure the script above can be improved, but it generally works. A few bugs sometimes, because (I think) the mount/unmount operations are too quick and kind of cut-off the file operations.

I'm sure that you, as the author, will know of more elegant ways to perform (and wait for completion of) those operations.

But, I hope it can be a neat little addition!

marcone, 2023-12-23:
The cause of the errors you're seeing script probably isn't that something is too quick, but that it is modifying the music drive while it is still connected to the car. In other words, the car would see disk sectors change out from under it, as if the drive is failing or corrupt.
The script needs to call `disable_gadget.sh` first, _then_ mount the drive on the Pi, modify it, unmount it, and finally call `enable_gadget.sh` to reconnect the updated drive to the car.

Helvio88, 2023-12-23:
I had the disable gadget at the top, but moving to the bottom kind of worked, too. Do you think you can "salvage" the script to make it into something official? Am I in the right track?

Another question, is it safe to run that script at boot time, or is there a routine that has to be awaited on?

Helvio88, 2023-12-23:
I noticed the crontabs are wiped on reboot lol - any other way to have this run at boot?

marcone, 2023-12-24:
I suppose the proper way to run it at boot would be to make it a systemd service, like TeslaUSB itself. If you make it so it runs before TeslaUSB, then you don't need to call disable_gadget/enable_gadget either.

d-kar, 2023-12-24:
> You can do this by just copying LockChime.wav to music sync folder. Just tried it and it works!
Works only for the first time. It does not overwrite existing LockChime.wav unfortunately
@marcone any way to detect that it's a different file and overwrite?

marcone, 2023-12-24:
> any way to detect that it's a different file and overwrite?

It should already detect this based on the file date.

d-kar, 2023-12-24:
> > any way to detect that it's a different file and overwrite?
> 
> It should already detect this based on the file date.

Problem is that the file I want to copy is created/modified earlier than the file that already exist on the TeslaUSB

d-kar, 2023-12-24:
Will it be a lot slower if you add -c (--checksum) option to rsync?

d-kar, 2023-12-25:
Reading rsync documentation it is supposed to copy files that have a different size, but it doesn't. When I do touch on a file in the musicarchive folder it copies it, but if it's older than the one in the /mnt/music then it's skipped even if file sizes are different.

`root@teslapi:/mnt/musicarchive# ls -al
total 192
drwxrwxrwx 2 root root   4096 Dec 23 16:55 .
drwxr-xr-x 6 root root   4096 Nov 29 11:37 ..
drwxrwxrwx 2 root root      0 Dec 21 14:22 LightShow
drwxrwxrwx 2 root root      0 Dec 23 17:34 LockChimes
-rwxrwxrwx 1 root root 185146 Dec 23 16:58 LockChime.wav

root@teslapi:/mnt/musicarchive# ls -al /mnt/music
total 516
drwxrwxrwx  5 root root  32768 Dec 31  1969  .
drwxr-xr-x  6 root root   4096 Nov 29 11:37  ..
drwxrwxrwx  2 root root  32768 Dec 21 14:25  LightShow
drwxrwxrwx 11 root root  32768 Dec 24 00:32  LockChimes
-rwxrwxrwx  1 root root 371226 Dec 24 17:48  LockChime.wav
-rwxrwxrwx  1 root root      0 Nov 29 21:02  .metadata_never_index
drwxrwxrwx  2 root root  32768 Nov 29 14:32 'System Volume Information'
`

`root@teslapi:/mnt/musicarchive# rsync -rum --no-human-readable --exclude=.fseventsd/*** --exclude=*.DS_Store --exclude=.metadata_never_index                 --exclude="System Volume Information/***"                 --delete --modify-window=2 --info=stats2 "/mnt/musicarchive/" "/mnt/music"

Number of files: 24 (reg: 12, dir: 12)
Number of created files: 0
Number of deleted files: 0
Number of regular files transferred: 0
Total file size: 34990145 bytes
Total transferred file size: 0 bytes
Literal data: 0 bytes
Matched data: 0 bytes
File list size: 0
File list generation time: 63.088 seconds
File list transfer time: 0.000 seconds
Total bytes sent: 798
Total bytes received: 19
`

d-kar, 2023-12-26:
Might want to add -t option to preserve original file dates.

marcone, 2024-01-12:
The web interface now has a music browser that supports uploading files to it, and a way to copy any .wav file to LockChime.wav, so you can copy a bunch of lock sounds to your music drive, then right-click any one of them in the web interface to mark them as the current lock sound.
Not sure how this interacts with actual USB music playback in the car though.

closed 2024-01-12 by marcone

hvyarms, 2024-08-03:
In case anyone else is looking for the "random" functionality where you can drop a bunch of wav files in a folder and forget about it, you can indeed install systemd service which will persist reboot to achieve this functionality.  The way I set it up is to use a subdirectory in the autosync music folder tied to a systemd service that is enabled to auto run and then set the script posted above in a loop.  Really rough details here:

systemd service (install in /etc/systemd/system directory):

```
[Unit]
Description=This service randomizes lockchimes

[Service]
User=root
WorkingDirectory=/home/pi/
ExecStart=/home/pi/CronLockChime.sh
Restart=always
RestartSec=60

[Install]
WantedBy=multi-user.target
```

Script:
Remember to add execute perms (chmod +x):
```
#!/bin/bash -eu

while true
do

# Mount /mnt/music
mount /mnt/music || :

# Set the source and destination folders
source_folder="/mnt/music/LockChimes/"
destination_file="/mnt/music/LockChime.wav"
curr_dir=$(pwd)

# Change to the source folder
cd "$source_folder" || exit

# Find all .wav files except LockChime.wav
files=($(find . -maxdepth 1 -type f -name "*.wav" ! -name "LockChime.wav"))

# Check if there are any eligible files
if [ ${#files[@]} -eq 0 ]; then
    echo "No eligible files found in $source_folder."
    exit 1
fi

# Get the total number of files
num_files=${#files[@]}

# Generate a random index
random_index=$((RANDOM % num_files))

# Choose a random file from the list
random_file=${files[random_index]}

# Copy the chosen file to the destination, overwriting the existing LockChime.wav
cp -f "$random_file" "$destination_file"

echo "Random file '$random_file' copied to '$destination_file'."

#
cd "$curr_dir"

# Unmount /mnt/music
umount /mnt/music || :

# Refresh USB gadget
/root/bin/disable_gadget.sh || :
/root/bin/enable_gadget.sh || :

sleep 60
done
```

Then the one-time commands to get it going:
```
sudo -i /root/bin/remountfs_rw
sudo systemctl daemon-reload
sudo systemctl start lockchime.service
sudo systemctl enable lockchime.service
```

marcone, 2024-08-03:
Note that that script will briefly remove the USB drive(s) from the car every minute, which will likely interrupt dashcam, sentry and music playback. The script also modifies the music drive while it is visible to the car, which to the car could look like disk corruption.

hvyarms, 2024-08-04:
In this case since the car itself takes care of the lockchime stuff when you exit... I'm wondering if I can just safely remove the umount / disable/enable part of the script and just do the copy (it probably will still work).

hvyarms, 2024-08-21:
Marcone - is there any method or existing script that detects if the car has been parked/idle for x amount of time?  The randomization I posted above works but it will "stick with" a song until the drive goes offline (as the filesystem isn't remounted).  With sentry mode and other things keeping the car awake, the randomization changes don't happen as frequently as I would like (maybe once a day).  Wondering if I could add logic to detect car has been parked/idle for let's say 30m, force a disconnect reconnect once in that situation.