marcone/teslausb #806: [Feature Request] Support mobile hotspot internet access

fiveangle, 2024-01-04:
### Describe the problem

Issue:
* TeslaUSB AP mode requires user to connect to TeslaUSB AP to access recordings (and any other future features). Device-specific implementations not withstanding, this prevents the users mobile device from accessing the Internet, so they must reconnect/disconnect to the TeslaUSB AP whenever they wish to view the recordings if they are not located at their primary WiFi location.

Feature Improvement:
* As many of us have either Personal Hotspot on our phone, or an outboard Mobile Hotspot device configured for our vehicles to access internet-connected features but don't subscribe to Tesla's Premium Connectivity, adding capability for TeslaUSB to attempt to connect to a specific secondary WiFi network would allow users to access TeslaUSB recording web interface at any time, rather than only when connecting to the TeslaUSB AP network. It would also allow the Tesla in-car browser to access TeslaUSB web interface (and any other new features developed in the future, as this would open up the TeslaUSB platform for greater possibilities.
* Add a new feature to TeslaUSB to make it aware of a second "mobile" WiFi connection that allows it to access the internet. Optionally, a setting could be created to inform TeslaUSB to not expect to find the archive CIFS share on this network. Just as it is now, other WiFi clients on this network will be able to access TeslaUSB via the IP address that it is assigned via DHCP to TeslaUSB. TeslaUSB should continue to publish mDNS for itself to the network (just as it does with the "home" WiFI) so that users can connect to "http://teslausb.lan" (or custom, if set) from devices connected to the same "mobile" WiFi.
* Essentially, this request consists of concisely, "Support 2 or more WiFi network configurations".
* I could see people who have a primary, and secondary "home" to also be able to use this feature, if they have an archive server located at each location.

### Device

Raspberry Pi Zero W

### OS Image

Prebuilt TeslaUSB image

### Car Model

Model 3

### USB connection

Center console

marcone, 2024-01-04:
This has been [requested before](https://github.com/marcone/teslausb/issues/387) and my position on this hasn't really changed. It's easy enough to add additional networks after the fact (and even easier to just give your phone hotspot the same SSID and password as your home wifi), so it doesn't need to be part of TeslaUSB setup.

closed 2024-01-04 by marcone

fiveangle, 2024-01-09:
Sorry for the additional spin. A cursory search came up empty, hence the request.GitHub really needs to step up their search capabilities.

Not privy to the mechanism to add additional WiFi networks, perhaps this procedure could be detailed (or linked to, if appropriate) in the TeslaUSB FAQ to help both users and reduce further requests like this ?

Aside: setting one's mobile hotspot to the same SSID/Shared Key as one's home WiFi is the absolute worst advice lest someone get a rude awakening when everyone in their household consumes their mobile tethering allotment quicker than Sentry Mode can fill up a USB stick 🤣

Ps- this project is truly wonderful so thanks for keeping it going so beautifully ! ♥️

marcone, 2024-01-09:
> Sorry for the additional spin. A cursory search came up empty, hence the request.GitHub really needs to step up their search capabilities.

Don't search Github for the answer to a general linux/Raspberry Pi OS question. A Google search for "raspberry pi multiple wifi connections" showed [this](https://raspberrypi.stackexchange.com/questions/11631/how-to-setup-multiple-wifi-networks) as one of the top results, which explains how to do it when using wpa_supplicant.

> Not privy to the mechanism to add additional WiFi networks, perhaps this procedure could be detailed (or linked to, if appropriate) in the TeslaUSB FAQ to help both users and reduce further requests like this ?

I really don't want to go down the path of documenting every little bit of linux knowledge, because that would be a full time job. The TeslaUSB and FAQ cover the basics. For anything else I suggest asking Google, the [TeslaUSB discussion forum](https://github.com/marcone/teslausb/discussions) or the [TeslaUSB Discord](https://discord.gg/b4MHf2x).

> Aside: setting one's mobile hotspot to the same SSID/Shared Key as one's home WiFi is the absolute worst advice lest someone get a rude awakening when everyone in their household consumes their mobile tethering allotment quicker than Sentry Mode can fill up a USB stick 🤣

Why would you turn on the hotspot at home though?