marcone/teslausb #809: Music Stopped Working

everitj, 2024-01-13:
### Describe the problem

Music sync has always worked fine but has recently stopped working and I cannot get it to work again. Drive appears when connected to a computer and the cam function operates fine. Car must be able to see drive as lightshow folder is seen and can access the lightshows from car and also the LockChime works. Have tied everything but wonder if this an issue with the latest software? 

### Device

Raspberry Pi Zero W

### OS Image

Prebuilt TeslaUSB image

### Car Model

Model 3

### USB connection

Glove box

### Logs

====== summary ======
hardware: Raspberry Pi Zero 2 Rev 1.0
OS: Raspbian GNU/Linux 10 (buster)
CIFS archiving selected
2 snapshots mounted
Gadget state: not attached. 17.7 seconds ago: connected to host
====== disk / images ======
Model: SD SD128 (sd/mmc)
Disk /dev/mmcblk0: 128GB
Sector size (logical/physical): 512B/512B
Partition Table: msdos
Disk Flags: 
Number  Start   End     Size    Type     File system  Flags
 1      4194kB  266MB   262MB   primary  fat32        boot, lba
 2      266MB   2414MB  2147MB  primary  ext4
 3      2414MB  128GB   125GB   primary  xfs
 4      128GB   128GB   105MB   primary  ext4
cam disk image has 10 extents
Model:  (file)
Disk /backingfiles/cam_disk.bin: 32.2GB
Sector size (logical/physical): 512B/512B
Partition Table: msdos
Disk Flags: 
Number  Start   End     Size    Type     File system  Flags
 1      1049kB  32.2GB  32.2GB  primary  fat32        lba
music disk image has 4 extents
Model:  (file)
Disk /backingfiles/music_disk.bin: 34.4GB
Sector size (logical/physical): 512B/512B
Partition Table: msdos
Disk Flags: 
Number  Start   End     Size    Type     File system  Flags
 1      1049kB  34.4GB  34.4GB  primary  fat32        lba
no boombox disk image found
Filesystem      Size  Used Avail Use% Mounted on
/dev/mmcblk0p1  247M   49M  199M  20% /boot
/dev/root       2.0G  1.2G  677M  64% /
/dev/mmcblk0p3  117G  3.7G  113G   4% /backingfiles
/dev/mmcblk0p4   95M  1.6M   86M   2% /mutable
====== network ======
lo: flags=73<UP,LOOPBACK,RUNNING>  mtu 65536
        inet 127.0.0.1  netmask 255.0.0.0
        inet6 ::1  prefixlen 128  scopeid 0x10<host>
        loop  txqueuelen 1000  (Local Loopback)
        RX packets 0  bytes 0 (0.0 B)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 0  bytes 0 (0.0 B)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0
[... 782 lines cut ...]
www-data   743   740  0 15:21 ?        00:00:00 /usr/sbin/fcgiwrap -c 4 -f
www-data   745   740  0 15:21 ?        00:00:00 /usr/sbin/fcgiwrap -c 4 -f
root       797     2 11 15:21 ?        00:00:00 [loop1]
root       800   308  2 15:21 ?        00:00:00 /bin/bash -eu /root/bin/copy-music.sh
root       811     1  0 15:21 ?        00:00:00 sleep 5
www-data   814   743  0 15:21 ?        00:00:00 /bin/bash /var/www/html/cgi-bin/diagnose.sh
root       815   814  2 15:21 ?        00:00:00 sudo /root/bin/setup-teslausb diagnose
root       834   815  0 15:21 ?        00:00:00 /bin/bash -eu /root/bin/setup-teslausb diagnose
root       840   834  0 15:21 ?        00:00:00 flock -en -E 99 /root/bin/setup-teslausb /root/bin/setup-teslausb diagnose
root       841   840  6 15:21 ?        00:00:00 /bin/bash -eu /root/bin/setup-teslausb diagnose
root       859   841  5 15:21 ?        00:00:00 /bin/bash -eu /root/bin/setup-teslausb diagnose
root       860   841  0 15:21 ?        00:00:00 tr \r \n
root       861   841  1 15:21 ?        00:00:00 sed /^ *$/d
root       862   841  0 15:21 ?        00:00:00 grep -a -v ^Reading package lists
root       863   841  0 15:21 ?        00:00:00 grep -a -v ^(Reading database
root       865   841  0 15:21 ?        00:00:00 grep -a -v ^Adding 'diversion of
root       866   841  0 15:21 ?        00:00:00 grep -a -v ^Removing 'diversion of
root       868   841  0 15:21 ?        00:00:00 sed -E s/\o033\[0;32m//
root       869   841  0 15:21 ?        00:00:00 sed -E s/\o033\[0m//
root       915   146  0 15:21 ?        00:00:00 /lib/systemd/systemd-udevd
root       916   146  0 15:21 ?        00:00:00 /lib/systemd/systemd-udevd
root       917   146  0 15:21 ?        00:00:00 /lib/systemd/systemd-udevd
root       936   859  0 15:21 ?        00:00:00 ps -eaf
teslausb has been up 0 minutes. System time is Sat 13 Jan 15:21:18 GMT 2024
====== end of diagnostics ======

marcone, 2024-01-13:
The logs seem to show that music sync was still in progress when you took the diagnostics, but that music sync did run and complete about 15 minutes prior to that. What exactly is the problem you're having?

everitj, 2024-01-13:
Yeah this was taken after a rebuild and multiple attempts to try different things. The issue is that the USB icon never appears in the car for music playback.. Used to work fine so not sure if what has changed to stop it from working

marcone, 2024-01-14:
If the lightshow and lock sound are on the music drive, then the car is clearly seeing the music drive. If it's not showing the music files you have on that drive, but you can see the music files when connected to a computer, then I think that's a car problem. TeslaUSB emulates the entire drive, it doesn't do anything special with the individual files on that drive.

closed 2024-01-15 by marcone

marcone, 2024-01-19:
> After further testing it would seem that as soon as you add lightshows to the drive it then won't recognise and play the music. Is it possible to add a third partition with the current config? Thanks

There already is support for a third drive, for boombox. Can lightshow share a drive with boombox?