marcone/teslausb #811: Error during update: Cam_disk.bin input/output error: unkown filesystem type ''.

muenchris, 2024-01-18:
### Describe the problem

I was synchronizing my TeslaUSB with my share on a NAS which took a very long time due to lots of footage. But at the very end, the "cam" USB drive was no longer visible in Windows. When I SSHed into the PI, I tried to get to the cam folder but could not mount the drive. Then I tried to update the pi using "/root/bin/setup-teslausb upgrade" but that failed with:

preparing for full upgrade
/tmp/setup-teslausb is up to date
updating rc.local
Updating package index files...
Get:1 http://archive.raspberrypi.org/debian buster InRelease [32.6 kB]
Get:2 http://raspbian.raspberrypi.org/raspbian buster InRelease [15.0 kB]
Get:3 http://raspbian.raspberrypi.org/raspbian buster/main armhf Packages [13.0 MB]
Get:4 http://archive.raspberrypi.org/debian buster/main armhf Packages [400 kB]
Get:5 http://raspbian.raspberrypi.org/raspbian buster/non-free armhf Packages [110 kB]
Fetched 13.6 MB in 43s (317 kB/s)
Reading package lists... Done
N: Repository 'http://raspbian.raspberrypi.org/raspbian buster InRelease' changed its 'Suite' value from 'oldstable' to 'oldoldstable'
N: Repository 'http://archive.raspberrypi.org/debian buster InRelease' changed its 'Suite' value from 'oldstable' to 'oldoldstable'
Verifying that the requested configuration is valid...
Downloaded /tmp/verify-configuration.sh ...
Checking XFS support
XFS supported
DATA_DRIVE is not set. SD card will be used.
Verifying that there is sufficient space available on the MicroSD card...
There is sufficient space available.
Downloading common runtime scripts.
Downloaded /root/bin/remountfs_rw ...
Downloaded /root/bin/make_snapshot.sh ...
Downloaded /root/bin/release_snapshot.sh ...
Downloaded /root/bin/force_sync.sh ...
Downloaded /root/bin/mountoptsforimage ...
Downloaded /root/bin/mountimage ...
Downloaded /root/bin/enable_gadget.sh ...
Downloaded /root/bin/disable_gadget.sh ...
~ ~
cmdline.txt is up to date
Downloaded /tmp/create-backingfiles-partition.sh ...
DATA_DRIVE not set. Proceeding to SD card setup
create-backingfiles-partition: using existing backingfiles and mutable partitions
updated /etc/fstab for /mnt/cam
updated /etc/fstab for /mnt/music
umount: /mnt/cam: not mounted.
already released
sfdisk: cannot open /backingfiles/cam_disk.bin: Input/output error
mount: /mnt/cam: unknown filesystem type ''.
STOP: Couldn't check image

What can I do to fix this?

### Device

Raspberry Pi Zero W

### OS Image

Prebuilt TeslaUSB image

### Car Model

Model X

### USB connection

Center console

marcone, 2024-01-18:
The "Input/output error" bit might indicate the sd card has gone bad.
If you haven't restarted the Pi since this happened, then running `dmesg` will show any I/O errors that happened.

muenchris, 2024-01-19:
Bummer, my wife had an accident two days ago and we were hoping that we have footage of the crash. But not the PI is not booting at all anymore :(

muenchris, 2024-01-19:
Does the sync process start with SentryClips or SavedClips? From my perspective SavedClips are more important as they are manually triggered and might have valuable streams. 
Maybe a great addition to the conf file would be to set the preference (Saved vs. Sentry) for synchronization.
In my case I think the card went bad right in the middle of the sync process and it did not even get to my SavedClips

marcone, 2024-01-19:
they're archived in alphabetical order, so SavedClips will be archived before SentryClips

muenchris, 2024-01-19:
> they're archived in alphabetical order, so SavedClips will be archived before SentryClips
> 
> 

Then the card might have been bad before. We just found the footage still in the TeslaX buffer and hope we can store on a new teslausb PI

marcone, 2024-01-19:
What do you mean by "the TeslaX buffer"?

closed 2024-01-25 by marcone