marcone/teslausb #816: Question reguarding NFS

mabushey, 2024-02-03:
### Describe the problem

Yes, you asked to use Discord, but they blocked my account after I joined and asked this question. I don't think they like my email server.

CIFS/SMB will NOT work for me. Is there any reason I can't easily modify this to use NFS? As there is a SSH server is there any reason why I can't use rsync over SSH to sync the files?

### Device

Raspberry Pi Zero 2 W

### OS Image

Prebuilt TeslaUSB image

### Car Model

Model Y

### USB connection

Glove box

marcone, 2024-02-03:
> Yes, you asked to use Discord, but they blocked my account after I joined and asked this question. I don't think they like my email server.

Discord shows you as an active Discord member since April 2022, so it doesn't look like you're blocked.
 
> CIFS/SMB will NOT work for me. Is there any reason I can't easily modify this to use NFS? As there is a SSH server is there any reason why I can't use rsync over SSH to sync the files?

rsync has been supported since the beginning of the project:
- [relevant section in the config file](https://github.com/marcone/teslausb/blob/main-dev/pi-gen-sources/00-teslausb-tweaks/files/teslausb_setup_variables.conf.sample#L25)
- [instructions](https://github.com/marcone/teslausb/blob/main-dev/doc/SetupRSync.md)

mabushey, 2024-02-04:
> Discord shows you as an active Discord member since April 2022
I created this account today

> it doesn't look like you're blocked.
Might want to eat some more carrots.

> rsync has been supported since the beginning of the project
Perfect, that completely resolves my issue. Thank you for this project and for responding to my question.

closed 2024-02-04 by mabushey

marcone, 2024-02-04:
> > Discord shows you as an active Discord member since April 2022
> 
> I created this account today

You joined the TeslaUSB Discord today, but you're a Discord member since April 29, 2022:

[image]

mabushey, 2024-02-04:
[image]

Discord probably outsourced the account creation to Microsoft.