marcone/teslausb #818: TeslaUSB Refresh_Token error - Endpoint is only available on fleetapi"

miles267, 2024-02-08:
### Describe the problem

It appears the TESLA_REFRESH_TOKEN function behavior has changed.  Refreshing the access token doesn't appear to resolve the issue.  Also saw a similar thread here: 

Has anyone managed to resolve this issue?  I'm trying with one of our cars before attempting the other.

### Device

Raspberry Pi 4

### OS Image

Prebuilt TeslaUSB image

### Car Model

Model Y

### USB connection

Glove box

### Logs

root@teslausb-ModelY:~# /root/bin/tesla_api.py --debug list_vehicles
Thu 08 Feb 09:09:51 CDT 2024: Loading mutable data from disk...
Thu 08 Feb 09:09:51 CDT 2024: Sending GET Request: https://owner-api.teslamotors.com/api/1/vehicles; Data: {}
Thu 08 Feb 09:09:51 CDT 2024: {
  "response": null,
  "error": "Endpoint is only available on fleetapi. Visit https://developer.tesla.com/docs for more info",
  "error_description": ""
}
Thu 08 Feb 09:09:51 CDT 2024: {
  "response": null,
  "error": "Endpoint is only available on fleetapi. Visit https://developer.tesla.com/docs for more info",
  "error_description": ""
}

miles267, 2024-02-08:
Am able to create new Tesla Refresh Tokens without issue. Also update config file and delete the json file from /mutable folder to clear any old, cached tokens.

marcone, 2024-02-09:
As I understand it, Tesla has a new API to interact with the car and the old API has been turned off.
My layman's understanding of the new API is that it requires signing up with Tesla as a developer and obtaining a secret key to be used when communicating with the API server. That is not going to work for TeslaUSB I think.

miles267, 2024-02-09:
Are there any other automated methods to keep the car awake long enough to complete the sync before allowing it to sleep? Only the Teslafi API?

Arthur-Philip-Dent, 2024-02-09:
1) Try to switch sentry mode on -> keeps the MCU/middle console powered, but causes “Vampire Drain”.
2) Powerbank/UPS for the SBC. You have to search a bit to find the appropriate one, also being able charging while powering the SBC. Also some UPS cause to reboot the SBC when power line comes back on.

KrzysztofHajdamowicz, 2024-02-09:
Teslamate had same issue. Fortunately, legacy API still works, but you have change /vehicle to /products.

Migrating to new API is the best option long-term.

miles267, 2024-02-09:
Thanks @KrzysztofHajdamowicz. I’ve since opted to move to TeslaFi’s API which I’m now testing. It seems to be working and not reporting errors in Teslausb archive log. I’ll set thid up for our other car tonight which has a significant backlog of videos that need to sync. That will be true test if it stays awake. Before paying for 2 TeslaFi subscriptions.

closed 2024-02-09 by miles267

marcone, 2024-02-14:
> Teslamate had same issue. Fortunately, legacy API still works, but you have change /vehicle to /products.

I could only get the basic "list_vehicles" to work using the /products URL, but actually getting vehicle state or controlling the vehicle doesn't seem to work.

marcone, 2024-02-14:
> > Teslamate had same issue. Fortunately, legacy API still works, but you have change /vehicle to /products.
> 
> I could only get the basic "list_vehicles" to work using the /products URL, but actually getting vehicle state or controlling the vehicle doesn't seem to work.

OK that's not entirely true.The "list" API works on the /products URL and can be used to get the vehicle ID. That ID can then be used with the /vehicles URL to get vehicle info. A lot of the other legacy API commands no longer work this way though.