marcone/teslausb #826: Unable to disable AP when enabled + Onedrive token issue

mews-se, 2024-03-20:
### Describe the problem

After enabling the internal wifi-AP - it's not longer possible to disable it just by editing the config.

I tried enabling it on my Zero 2 W but it took to much resources. Unfortunatly the only solution was to flash and start over.

I changed to a Pi4 which handled it much better.

Regarding Rclone. No matter how many tries, the token is unable to be accepted (prob because its 4 miles long), and the only workaround seem to be to create the rclone.conf on another machine and then copying it over. A bit wonky.

### Device

Raspberry Pi Zero 2 W

### OS Image

Prebuilt TeslaUSB image

### Car Model

Model 3

### USB connection

Glove box

marcone, 2024-03-20:
> I tried enabling it on my Zero 2 W but it took to much resources.

What do you mean by that?

mews-se, 2024-03-20:
> > I tried enabling it on my Zero 2 W but it took to much resources.
> 
> What do you mean by that?

The Zero W2 didnt like having the ap running and a webserver. It took to much resources - so much that it affected the normal dashcam performance. Just opening and browsing current dash events (in the car) could take up to 2 minutes

marcone, 2024-03-21:
That's... unexpected. I'm using a Raspberry Pi Zero and have no problems with the AP. Having it enabled has no noticeable effect on CPU load or memory use of the Pi.

mews-se, 2024-03-21:
> That's... unexpected. I'm using a Raspberry Pi Zero and have no problems with the AP. Having it enabled has no noticeable effect on CPU load or memory use of the Pi.

Interesting... Oh well - the plan was a 4 from the beginning and its running like a charm with better speeds overall!

Arthur-Philip-Dent, 2024-03-21:
I have a RPi0W2 since 3 years working great. 
A significant factor I observed were good cables to connect it to the car and as well a good performing and error free Micro-SD-card. 
I guess, you ruled that out?

mews-se, 2024-03-21:
> I have a RPi0W2 since 3 years working great. A significant factor I observed were good cables to connect it to the car and as well a good performing and error free Micro-SD-card. I guess, you ruled that out?

Yes sir. Also, before activating the AP and webserver all was well (as good as a zero w can do via 4g :) )

AmarnB94, 2024-07-05:
Can confirm there seems to be an issue with the keep alive/awake feature.

Tried providing the refresh token using the supplied method w/ "Auth for Tesla" app, both with quotes and without, but keep getting error "Failed to contact car using Tesla API, retrying in 5s..." in the ArchiveLoop logs.

Have tried running this command mentioned in a pretty old post: "/root/bin/tesla_api.py --debug wake_up_vehicle" and receive the following output, seems to be complaining about some errors in the python script for wakeup:

root@teslausb:~# /root/bin/tesla_api.py --debug wake_up_vehicle
Fri 05 Jul 22:24:45 AEDT 2024: Loading mutable data from disk...
Fri 05 Jul 22:24:45 AEDT 2024: Calling wake_up_vehicle()...
Fri 05 Jul 22:24:45 AEDT 2024: Sending wakeup API command...
Fri 05 Jul 22:24:45 AEDT 2024: Attempting to wake up Vehicle (ID:3744611613337051)
Fri 05 Jul 22:24:45 AEDT 2024: Sending POST Request: https://owner-api.teslamotors.com/api/1/vehicles/3744611613337051/wake_up; Data: {}
Fri 05 Jul 22:24:46 AEDT 2024: {
  "error": "token expired (401)"
}
Traceback (most recent call last):
  File "/root/bin/tesla_api.py", line 597, in <module>
    main()
  File "/root/bin/tesla_api.py", line 579, in main
    result = function(**kwargs)
  File "/root/bin/tesla_api.py", line 397, in wake_up_vehicle
    return _execute_request()
  File "/root/bin/tesla_api.py", line 65, in _execute_request
    if result['response'] is None:
KeyError: 'response'

AmarnB94, 2024-07-06:
Can confirm there seems to be an issue with the keep alive/awake feature.

Tried providing the refresh token using the supplied method w/ "Auth for Tesla" app, both with quotes and without, but keep getting error "Failed to contact car using Tesla API, retrying in 5s..." in the ArchiveLoop logs.

Have tried running this command mentioned in a pretty old post: "/root/bin/tesla_api.py --debug wake_up_vehicle" and receive the following output, seems to be complaining about some errors in the python script for wakeup + have refreshed my token for good measure, but no change:

root@teslausb:~# /root/bin/tesla_api.py --debug wake_up_vehicle
Fri 05 Jul 22:24:45 AEDT 2024: Loading mutable data from disk...
Fri 05 Jul 22:24:45 AEDT 2024: Calling wake_up_vehicle()...
Fri 05 Jul 22:24:45 AEDT 2024: Sending wakeup API command...
Fri 05 Jul 22:24:45 AEDT 2024: Attempting to wake up Vehicle (ID:3XXXXXXXXXXXXXXX)
Fri 05 Jul 22:24:45 AEDT 2024: Sending POST Request: https://owner-api.teslamotors.com/api/1/vehicles/3XXXXXXXXXXXXXXX/wake_up; Data: {}
Fri 05 Jul 22:24:46 AEDT 2024: {
"error": "token expired (401)"
}
Traceback (most recent call last):
File "/root/bin/tesla_api.py", line 597, in
main()
File "/root/bin/tesla_api.py", line 579, in main
result = function(**kwargs)
File "/root/bin/tesla_api.py", line 397, in wake_up_vehicle
return _execute_request()
File "/root/bin/tesla_api.py", line 65, in _execute_request
if result['response'] is None:
KeyError: 'response'

marcone, 2026-01-20:
On systems using Network Manager (e.g. Bookworm, Trixie) the AP can disabled by removing it from the config file.