marcone/teslausb #827: Rsync issue "waiting for archive to be reachable..."

arnfiix, 2024-03-23:
### Describe the problem

I've got TeslaUSB up and running, but I have not managed to get it to rsync data across from the Pi to my Synology NAS, but I can't quite work out why. I have attached the diagnostics and archiveloop log file, and each time it appears to get stuck "waitng for archive to be reachable..."

I can SSH to the Pi, to the Synology NAS, and the NAS appears to have the correct ssh key. Browsing via the web interface I can see more recent TeslaCam files than are available on the NAS share.

I'm really at a bit of a loss, sorry, hoping you can point me in the right direction. Pi and Linux aren't my strong suits (by which I mean I know nothing about them) :(

### Device

Raspberry Pi 4

### OS Image

Prebuilt TeslaUSB image

### Car Model

Model 3

### USB connection

Center console

### Logs

[diagnostics.txt](https://github.com/marcone/teslausb/files/14731885/diagnostics.txt)
[archiveloop.log](https://github.com/marcone/teslausb/files/14731891/archiveloop.log)
[teslausb-headless-setup.log](https://github.com/marcone/teslausb/files/14731903/teslausb-headless-setup.log)

marcone, 2024-04-04:
Can you try:
```
sudo -i
source bin/envsetup.sh
bin/archive-is-reachable.sh $ARCHIVE_SERVER
echo $?
```
and see what that says?

arnfiix, 2024-04-04:
root@teslausb:~# source bin/envsetup.sh
bin/archive-is-reachable.sh $192.168.50.30
echo $?
255

marcone, 2024-04-04:
Does:
```
sudo -i
source bin/envsetup.sh
ssh  "$RSYNC_USER"@"$ARCHIVE_SERVER" exit
```
give a more specific error?

arnfiix, 2024-04-04:
Unfortunately not. I noticed that SSH is disabled for all but admin accounts, so I added the TeslaCam account as an admin (something I was hoping to avoid) but even that doesn't seem to resolve the issue.

Now when I do the following, it just goes back to the pi@teslausb command line.

root@teslausb:~# ssh TeslaCam@192.168.50.30 exit
TeslaCam@192.168.50.30's password:

If I remove the exit and enter the correct password this now gets me onto the NAS (TeslaCam@SynNAS:~$), just as if I was accessing it from my PC. So that works, but I'm not seeing any data moving if I trigger a manual sync.

marcone, 2024-04-04:
Does:
```
sudo -i
source bin/envsetup.sh
bin/archive-is-reachable.sh $ARCHIVE_SERVER
echo $?
```
print "0" now?

arnfiix, 2024-04-04:
Nope, still 255.

marcone, 2024-04-04:
> If I remove the exit and enter the correct password

You shouldn't have to enter a password. When using rsync, ssh should be configured to use a public/private key pair, not a password, as described [here](https://github.com/marcone/teslausb/blob/main-dev/doc/SetupRSync.md) which is also referenced from the config file.

arnfiix, 2024-04-04:
Ok, I will go through that process again. I did try and set that up, but I
am guessing I must have done it wrong. Will advise if I am still having
difficulties.

Should I assume that when I can SSH from the pi to the nas without entering
a password, then I have got it correct?

marcone, 2024-04-04:
> Should I assume that when I can SSH from the pi to the nas without entering a password, then I have got it correct?

I think so yes

arnfiix, 2024-04-04:
Ok, so have been doing some more exploring. I've confirmed that a key is being generated, and it is being copied to the server. I can see a key on the server with the unique comment I entered for testing purposes. However I still cannot login via ssh. With verbose logging turned on, I get the following:

debug1: Next authentication method: publickey
debug1: Offering public key: /root/.ssh/id_rsa RSA SHA256:XXXXXXX Hashed key goes here
debug1: Authentications that can continue: publickey,password
debug1: Trying private key: /root/.ssh/id_dsa
debug1: Trying private key: /root/.ssh/id_ecdsa
debug1: Trying private key: /root/.ssh/id_ecdsa_sk
debug1: Trying private key: /root/.ssh/id_ed25519
debug1: Trying private key: /root/.ssh/id_ed25519_sk
debug1: Trying private key: /root/.ssh/id_xmss
debug1: Next authentication method: password

So while it appears to be offering the id_rsa key, it isn't working for some reason.

marcone, 2024-04-04:
I have no experience with Synology, but maybe you just need to change some permissions: https://blog.aaronlenoir.com/2018/05/06/ssh-into-synology-nas-with-ssh-key/

marcone, 2024-04-08:
This seems to be a (possibly Synology specific) ssh configuration issue, unrelated to TeslaUSB.

closed 2024-04-08 by marcone