marcone/teslausb #828: boombox doesn't mount, webui shows a random folder

sanxfxteam, 2024-03-25:
### Describe the problem

I've setup the config to create a boombox drive, but it doesn't mount it.
I'm not sure how to troubleshoot more.

There was an error message during the upgrade.
```boombox_disk.bin: failed to set up loop device: Resource temporarily unavailable```

files
```
root@teslausb:~# ls -l /backingfiles/
total 10604616
-rw-r--r-- 1 root root   104857600 Mar 24 20:27 boombox_disk.bin
-rw-r--r-- 1 root root          45 Mar 24 20:33 boombox_disk.bin.opts
-rw-r--r-- 1 root root 32212254720 Mar 24 20:54 cam_disk.bin
-rw-r--r-- 1 root root 10737418240 Mar 24 21:00 music_disk.bin
-rw-r--r-- 1 root root          93 Mar 24 20:30 music_disk.bin.opts
drwxr-xr-x 3 root root          25 Mar 24 20:53 snapshots
```

fstabs
```
/backingfiles/cam_disk.bin /mnt/cam teslausb noauto 0 0
/backingfiles/music_disk.bin /mnt/music teslausb noauto 0 0
/backingfiles/boombox_disk.bin /mnt/boombox teslausb noauto 0 0
```

WebUI
[image]

### Device

Raspberry Pi Zero W

### OS Image

Prebuilt TeslaUSB image

### Car Model

Model S

### USB connection

Center console

marcone, 2024-03-25:
I fixed the issue where it would show a different folder if it couldn't access the selected drive.

The failure to mount the drive looks like it might be the issue described [here](https://lore.kernel.org/lkml/20210615084259.yj5pmyjonfqcg7lg@ws.net.home/) where the kernel reports the operation should be tried again, but the userspace code just fails without retrying.
Unfortunately Raspberry Pi OS buster does not have the fix.
How reproducible is this problem for you?

sanxfxteam, 2024-03-25:
It's 100% reproducible. cam and music are mounted but not boombox.

I've tried to run the command:
```mount /mnt/boombox```
but it says mount not found. I've never seen that command fail like that.

I've added the diagnostics, just in case.
[diagnostics.txt](https://github.com/marcone/teslausb/files/14750492/diagnostics.txt)

marcone, 2024-03-25:
Can you capture the diagnostics again after trying `mount /mnt/boombox`, without rebooting?

sanxfxteam, 2024-03-25:
```root@teslausb:~# mount /mnt/boombox
mount: /mnt/boombox: mount point does not exist.
```
No new entry in diagnostics.

Here is the setup log
[teslausb-headless-setup.log](https://github.com/marcone/teslausb/files/14750574/teslausb-headless-setup.log)

marcone, 2024-03-25:
> mount: /mnt/boombox: mount point does not exist.

Is that true? Does /mnt/boombox not exist?

sanxfxteam, 2024-03-26:
it exists in fstab
```
root@teslausb:~# mount /mnt/boombox
mount: /mnt/boombox: mount point does not exist.
root@teslausb:~# cat /etc/fstab
proc            /proc           proc    defaults          0       0
PARTUUID=8b3aede6-01  /boot           vfat    rw,ro          0       2
PARTUUID=8b3aede6-02  /               ext4    defaults,noatime,ro  0       1
# a swapfile is not a swap partition, no line here
#   use  dphys-swapfile swap[on|off]  for that
LABEL=backingfiles /backingfiles xfs auto,rw,noatime 0 2
LABEL=mutable /mutable ext4 auto,rw 0 2
tmpfs /var/tmp tmpfs nodev,nosuid 0 0
tmpfs /tmp    tmpfs nodev,nosuid 0 0
tmpfs /var/spool tmpfs nodev,nosuid 0 0
tmpfs /var/lib/ntp tmpfs nodev,nosuid 0 0
/backingfiles/cam_disk.bin /mnt/cam teslausb noauto 0 0
/backingfiles/music_disk.bin /mnt/music teslausb noauto 0 0
/backingfiles/boombox_disk.bin /mnt/boombox teslausb noauto 0 0
tmpfs /var/log/nginx tmpfs nodev,nosuid 0 0
tmpfs /var/lib/nginx tmpfs nodev,nosuid 0 0
mount.ctts#/mutable/TeslaCam /var/www/html/TeslaCam fuse defaults,nofail,x-systemd.requires=/mutable 0 0
```

marcone, 2024-03-26:
> it exists in fstab
> 
> ```
> root@teslausb:~# ls /mnt/
> cam  music
> ```

But clearly it does not exist on disk. So step 1 would be to `mkdir /mnt/boombox`.

sanxfxteam, 2024-03-26:
```
root@teslausb:~# bin/remountfs_rw
Remounting / as read/write
Remounting /boot as read/write
Done. Make any changes needed, and reboot the Pi when ready.
Filesystems / and /boot will be mounted as read only after reboot.
root@teslausb:~# mkdir /mnt/boombox
root@teslausb:~# mount /mnt/boombox
mount: /mnt/boombox: unknown filesystem type ''.
```

marcone, 2024-03-26:
Looks like setup failed before it was able to initialize the boombox image. It's probably easiest at this point to delete all of the disk images and run `setup-teslausb upgrade` to recreate them.

sanxfxteam, 2024-03-26:
Ok, I did it, it works now. I did it before but it didn't work for some reason. Thanks marcone.

marcone, 2024-03-26:
> I did it before but it didn't work for some reason.

You probably didn't delete all of the images first.

sanxfxteam, 2024-03-26:
I did delete the cam and music .bin files

Laurent,

marcone, 2024-04-05:
I put in a workaround for the loop device bug.

closed 2024-04-05 by marcone