marcone/teslausb #829: Everything appears to be working, but no recordings are copied

mondagoar, 2024-03-29:
### Describe the problem

I setup my Pi Zero a couple of days ago and followed the simpled guide. Many things seems to be working without a problem:
- I can use the web interface of teslausb
- I can connect via USB to my Windows PC and see the contents of the TeslaCam directory. Recordings are being added by my Car (Model Y)
- The setup logs look good, no obvious errors
- The ArchiveLoop.log looks good, no obvious errors. 
- Mounting of the archive appears to be working

I try a couple of environments for testing:
1) Connect my Pi to the glovebox in my car, and connect over wifi via web or ssh to the unit and read logs, etc
2) Connect my Pi to my USB on my PC, and connect over over wifi, etc

Despite new recordings being created, the ArchiveLoop.log always shows things like this:
- Fri 29 Mar 16:44:55 GMT 2024: taking snapshot of cam disk in /backingfiles/snapshots/snap-000001
- Fri 29 Mar 16:44:57 GMT 2024: took snapshot
- Fri 29 Mar 16:44:57 GMT 2024: comparing new snapshot with /backingfiles/snapshots/snap-000000/snap.bin
- Fri 29 Mar 16:44:57 GMT 2024: new snapshot is identical to previous one, discarding

Even if, when connected to my PC, I copy files around in the TeslaCam directory (over USB), the logs still report "new snapshot is identical". In an effort to provoke it to detect a change, I went into the /backingfiles/snapshots directory and deleted all of the contents. I then forced a resync. The unit detected a "change" and said:
- Fri 29 Mar 16:48:05 GMT 2024: taking snapshot of cam disk in /backingfiles/snapshots/snap-000000
- Fri 29 Mar 16:48:06 GMT 2024: took snapshot
- Fri 29 Mar 16:48:06 GMT 2024: comparing new snapshot with /backingfiles/snapshots/snap--00001/snap.bin
- Fri 29 Mar 16:48:06 GMT 2024: making links for /tmp/snapshots/snap-000000, retargeted to /backingfiles/snapshots/snap-000000/mnt
- Fri 29 Mar 16:48:06 GMT 2024: made all links for /tmp/snapshots/snap-000000

But from what I can tell, it didn't see any files to copy, and certainly didn't copy any.

I don't know what else to try. It's just like it's not seeing my files on the USB drive. 

### Device

Raspberry Pi Zero W

### OS Image

Prebuilt TeslaUSB image

### Car Model

Model S

### USB connection

Center console

### Additional information

These are the lines from my teslausb_setup_variables.conf file that are uncommented. The "xxx" values have been changed, obviously:

export ARCHIVE_SYSTEM=cifs
export ARCHIVE_SERVER=192.168.1.81
export SHARE_NAME='teslacam'
export SHARE_USER=teslausb
export SHARE_PASSWORD='xxx'
export CIFS_VERSION="3.0"

export ARCHIVE_RECENTCLIPS=true

export CAM_SIZE=30G
export SSID='xxx'
export WIFIPASS='xxx'

mondagoar, 2024-03-29:
Also to note, on the web interface, on the "Recordings" page, I see two folders: SavedClips and SentryClips. Both directories are empty. And the Viewer page also says "no recordings".  
Meanwhile, the USB shows the recordings and they are playable. And the recordings can be played by the car also.

marcone, 2024-03-29:
Can you download the diagnostics from the web interface and attach them here?

mondagoar, 2024-03-29:
[diagnostics.txt](https://github.com/marcone/teslausb/files/14808161/diagnostics.txt)

Thanks. Is this what you mean?

mondagoar, 2024-03-29:
I've tried a lot of things. Many of them failed. So I guess you'll see some errors from those experiments.

marcone, 2024-03-29:
> Meanwhile, the USB shows the recordings and they are playable

What do you mean by "the USB" in this context?
What you described above (nothing showing showing up on the Pi web interface, but recordings being visible in the car) kind of sounds like you have a regular USB drive connected in addition to the Pi.

marcone, 2024-03-29:
Though also note that recordings don't immediately show up in the web interface. The web interface shows what's in the snapshots, and TeslaUSB only takes a snapshot every hour. You can force one by rebooting the Pi or triggering a sync in the Tools tab of the web interface.

mondagoar, 2024-03-29:
Thanks for your quick reply. When I said "the USB shows the recordings and they are playable", I meant that I plugged the Pi into my computer's USB port. I can see the recordings on it, that the car has made. I do not have any other USB drive connected.

mondagoar, 2024-03-29:
I've never seen any recordings in the web interface. I've even tried copying files onto the "USB" drive from my PC, but it always seems to say the snapshot is the same as the current, ie no recordings are detected. Even rebooting and triggering a sync does nothing of interest.

marcone, 2024-03-29:
Can you paste the output of `ls -lLR /mutable/TeslaCam` ?

mondagoar, 2024-03-29:
Thanks. Here is the output:

/mutable/TeslaCam:
total 2
drwxr-xr-x 2 root root 1024 Mar 28 18:50 SavedClips
drwxr-xr-x 2 root root 1024 Mar 28 18:50 SentryClips

/mutable/TeslaCam/SavedClips:
total 0

/mutable/TeslaCam/SentryClips:
total 0

marcone, 2024-03-29:
hmm, but you say that when you plug the Pi in to a PC, you see files in the "CAM" drive?

mondagoar, 2024-03-29:
Well. That was the case. But then I thought "I'll wipe the contents of the drive" coz I saw this ticket: https://github.com/marcone/teslausb/issues/825

So when I've run the about ls command, the drive was empty. I think I perhaps made a mistake with wiping it though because my car is saying that the USB drive is unavailable. 

I'll get the drive operational again and then try the ls command again.

mondagoar, 2024-03-29:
Ok. The drive is restored again. And this is what it all looks like:

[image]

[image]

[image]

mondagoar, 2024-03-29:
FWIW, on the Pi, I created a subdirectory under /mutable/TeslaCam/SavedClips and put an mp4 file in there. I could see the directory and file through the TelsaUsb web interface. 

So when it's working, how are the files supposed to get into this directory? Are they supposed to be copied from the USB storage (.bin file)?

Do you think I should just rebuild everything? I've ordered a new SD card to test with. So I can build a new system tomorrow without damaging my current one.

marcone, 2024-03-29:
/mutable/TeslaCam normally doesn't contain files, it contains symbolic links to snapshots.

mondagoar, 2024-03-30:
Please don’t spend any more time on this. I’ve reinstalled and everything is working ok on the new installation.

closed 2024-03-30 by mondagoar