marcone/teslausb #846: Feature: consult recordings from the cifs/smb share

SashaBusinaro, 2024-05-19:
### Describe the problem

Add the ability to view recordings from the web interface directly from the cifs/smb share used as the destination to eliminate any latency and to be able to view them even when the car (and Raspberry) is off.

### Device

Raspberry Pi 4

### OS Image

Prebuilt TeslaUSB image

### Car Model

Model 3

### USB connection

Glove box

marcone, 2024-05-19:
The web interface is hosted by the Pi itself, so it won't work when the Pi is off.

What you're really asking is to add a viewer to the CIFS server.
Alternatively, consider something like https://sentrycam.video/ and point it at your share.

SashaBusinaro, 2024-05-19:
Thank you, that sounds like a good option!

closed 2024-05-19 by SashaBusinaro