marcone/teslausb #851: Should it be possible to run TeslaSolarCharger BLE API on same RPi

idominiki, 2024-06-24:
### Describe the problem

Hi guys. 

As Tesla changed their API a few weeks ago, TeslaSolarCharger has a few problems. There is now a BLE API which requires eg a RPi. 

Should it be possible to run this docker container on teslausb?

thanks!
Dominik

### Device

Raspberry Pi Zero 2 W

### OS Image

Prebuilt TeslaUSB image

### Car Model

Model 3

### USB connection

Glove box

KrzysztofHajdamowicz, 2024-06-24:
Since / is mounted as read-only, docker fails to start due to few directories being non-writable.

You would need to create separate partition for writable data, mount it on top of read-only system, mount --bind subdirectories into appropriate locations and running docker image should be possible.

marcone, 2024-06-24:
> Since / is mounted as read-only, docker fails to start due to few directories being non-writable.
> 
> You would need to create separate partition for writable data, mount it on top of read-only system, mount --bind subdirectories into appropriate locations and running docker image should be possible.

If you know which directories need to be writeable, you can either mount a tmpfs on them (for data that doesn't need to be preserved across reboots), or symlink them to folders in /mutable (for data that does need to be preserved)