marcone/teslausb #856: After a few days teslausb does not connect to wifi any more but still nointed and recording.

thesourcerer, 2024-07-16:
### Describe the problem

Running on Pi Zero W2,
Working fine for a few days, then stops connecting to wifi (router says device is not connected).
Device is still mounted to tesla and recording during this time.
Power cycling resolves the issue until it comes back a few days later.

What can i check? 

### Device

Raspberry Pi Zero 2 W

### OS Image

Prebuilt TeslaUSB image

### Car Model

Model Y

### USB connection

Glove box

### Logs

Not connected to wifi

marcone, 2024-07-19:
The kernel log might have information as to why it's not connecting to wifi, but the only way to get that is by logging in to the Pi, and you'd need a wifi connection for that.

thesourcerer, 2024-07-21:
Yep... 
For now i have added a loop in the archiveloop that monitors the connection every 15 min and logs the issue if found and also tries to recover with stopping\starting the interface.

Sportwyse, 2024-07-28:
I'm having the same issue. We will stay connected for 3-4 days then stop connecting to wifi.

marcone, 2024-09-04:
> I'm having the same issue. It will stay connected for 3-4 days then stop connecting to wifi. It looks like when the drive gets full. The wifi won't connect.

Which drive gets full?

marcone, 2024-10-29:
Nothing actionable here.

closed 2024-10-29 by marcone

thesourcerer, 2024-10-30:
I managed to work around it by creating a cron task to reboot the pie twice a day. Fixed all issues.