marcone/teslausb #861: STOP: The MicroSD card is too small - using  teslausb-20231012-bullseye prebuilt image

blacktiger333, 2024-07-24:
### Describe the problem

tried using the latest Pre-release image with my Raspberry Pi Zero 2 W that ended up with the error message STOP:The MicroSD card is to small during the installation process. Using successfully the "teslausb-20211106.2" prebuilt image with same teslausb-setup_variables.conf settings before (with CAM_SIZE=60% and MUSIC_SIZE=40%). I'm using a 128 GB MicroSD card. 

### Device

Raspberry Pi Zero W

### OS Image

Prebuilt TeslaUSB image

### Car Model

Model Y

### USB connection

Glove box

marcone, 2024-07-24:
Without the actual full message from the setup log there isn't much I can do to diagnose this

sedace, 2024-08-23:
Running into similar issue, will do some more diagnostics and searching, ran the one step setup, can ssh / connect, get the following in the setup.log

Fri 23 Aug 18:03:01 EDT 2024 : Downloaded /tmp/verify-configuration.sh ...
Fri 23 Aug 18:03:01 EDT 2024 : Checking XFS support
Fri 23 Aug 18:03:03 EDT 2024 : XFS supported
Fri 23 Aug 18:03:03 EDT 2024 : DATA_DRIVE is not set. SD card will be used.
Fri 23 Aug 18:03:03 EDT 2024 : Verifying that there is sufficient space available on the MicroSD card...
Fri 23 Aug 18:03:03 EDT 2024 : STOP: The MicroSD card is too small: 108003328 bytes available.
Fri 23 Aug 18:03:03 EDT 2024 : Model: SD SN256 (sd/mmc)
Disk /dev/mmcblk0: 256GB
Sector size (logical/physical): 512B/512B
Partition Table: msdos
Disk Flags:

Number  Start   End     Size    Type     File system  Flags
 1      4194kB  273MB   268MB   primary  fat32        lba
 2      273MB   2324MB  2051MB  primary  ext4
 3      2324MB  256GB   253GB   primary

marcone, 2024-08-25:
That looks like setup was interrupted while the partitions were being created, and then restarted.
Can you attach the complete setup log?

sedace, 2024-08-25:
I had the same hunch so I restarted the install from scratch. ( I ended up having different issues and eventually got it going. ) It's working now.   

I honestly don't recall what fixed it but: I know at one point I had to go into diskmgmt and clean off the card partitions, then formatted the card, re-imaged it and re-did the setup.conf, I still ran into a different snag after that... so I re-did all that once more, but this time I connected it to a USB power supply directly instead if using a usb port on my PC during the setup.  After that it went smoothly, got the setup "all done" and all appears to be good. 

 Thanks for your time and efforts on this project!