marcone/teslausb #865: Not Syncing (Music)

alexstrbonn, 2024-08-07:
### Describe the problem

Hello!

Apologies for asking a question here, but on discord I only get "Whoops... Unable to accept invite"

I have a SMB share in the NAS with the MUSIC_SHARE_NAME=fritz.nas/UDISK_2_0/teslamusic". The music share does not seem to sync to TeslaUSB, it remains empty. 
I have attached the diagnostic files donwloaded from the TeslaUSB Web interface. At the moment TeslaUSB is not connected to the car to allow better troubleshooting. 

Any help would be much appreciated, unfortunately i am not a shell/Linus expert.
Many thanks
[archiveloop.log](https://github.com/user-attachments/files/16529178/archiveloop.log)
[diagnostics.txt](https://github.com/user-attachments/files/16529179/diagnostics.txt)
[teslausb-headless-setup.log](https://github.com/user-attachments/files/16529181/teslausb-headless-setup.log)

### Device

Raspberry Pi Zero W

### OS Image

Prebuilt TeslaUSB image

### Car Model

Model S

### USB connection

Center console

### Logs

Please see attached files

marcone, 2024-08-07:
Your issue report says you're using a Raspberry Pi Zero W, however your diagnostics show a Raspberry Pi 4.
In any case, the log shows it successfully mounted the music share, but then:
```
Copied 0 music file(s), deleted 0, skipped 0 previously-copied files, and encountered 3 errors.
```
so I'm guessing you have three files/folders in your share that can't be accessed due to a file permissions issue.

You can try the following after logging in to the Pi using ssh:
```
sudo -i
mount /mnt/musicarchive
ls -lR /mnt/musicarchive
```
to maybe get some more information.

alexstrbonn, 2024-08-07:
Many thanks for the reply. 
You are correct in all points:
-It is a Raspbery 4 - I have reedited the original post for future reference.
You are also correct: There are three testfiles in the share.

I tried using your SSH commands with the following results:

**_**pi@teslausb:~ $ sudo -i
Run 'bin/setup-teslausb upgrade' to update to the latest version of teslausb,
or run 'bin/remountfs_rw' to allow writing to the root partition.

root@teslausb:~# mount /mnt/musicarchive
root@teslausb:~# ls -lR /mnt/musicarchive
/mnt/musicarchive:
total 10496
-rwxrwxrwx 1 root root 4147755 May 27  2019 'Kraftwerk - Das Model.m4a'
-rwxrwxrwx 1 root root 3097921 May 27  2019 'Modern Talking - You_re My Heart, You_re My Soul (Official Music Video).m4a'
-rwxrwxrwx 1 root root 3470409 May 27  2019 'Vangelis - Chariots Of Fire.m4a'
root@teslausb:~#**_** 

I am sorry I have insufficient knowledge to interpret the permissions?

marcone, 2024-08-07:
Those permissions look OK, the files should be readable by anyone.
Try logging in to the Pi again and running `cat /tmp/rsyncmusiclog.txt`

alexstrbonn, 2024-08-07:
Thanks. 
<pre>root@teslausb:~# cat /tmp/rsyncmusiclog.txt
rsync: [sender] send_files failed to open &quot;/mnt/musicarchive/Kraftwerk - Das Model.m4a&quot;: Stale file handle (116)
rsync: [sender] send_files failed to open &quot;/mnt/musicarchive/Modern Talking - You_re My Heart, You_re My Soul (Official Music Video).m4a&quot;: Stale file handle (116)
rsync: [sender] send_files failed to open &quot;/mnt/musicarchive/Vangelis - Chariots Of Fire.m4a&quot;: Stale file handle (116)

Number of files: 4 (reg: 3, dir: 1)
Number of created files: 3 (reg: 3)
Number of deleted files: 0
Number of regular files transferred: 3
Total file size: 10716085 bytes
Total transferred file size: 10716085 bytes
Literal data: 0 bytes
Matched data: 0 bytes
File list size: 0
File list generation time: 0.048 seconds
File list transfer time: 0.000 seconds
Total bytes sent: 241
Total bytes received: 76

sent 241 bytes  received 76 bytes  634.00 bytes/sec
total size is 10716085  speedup is 33804.68
rsync error: some files/attrs were not transferred (see previous errors) (code 23) at main.c(1333) [sender=3.2.3]
root@teslausb:~# 

</pre>

alexstrbonn, 2024-08-08:
@marcone Many thanks. 
It appears the error is "Stale file handle (116)" and  "rsync error: some files/attrs were not transferred (see previous errors) (code 23) at main.c(1333) [sender=3.2.3]" see my last post.  I am using a FritzBox as a SMB Server, so not the best device..

ANy Idea for a workaround?

marcone, 2024-08-08:
A quick Google found a few potential workarounds. One is to [add `noserverino` to the mount options in /etc/fstab](https://askubuntu.com/questions/1265164/stale-file-handler-when-mounting-cifs-smb-network-drive-from-fritz-router), the other is to explicitly [specify protocol version 1.0](https://bbs.archlinux.org/viewtopic.php?id=249503) which you can do by adding `export CIFS_VERSION=1.0` to the config file and rerunning setup.

alexstrbonn, 2024-08-09:
@marcone 
Thank you, and surprising. The FritzBox Device I have (7560) supports  SMB Versions 1, 2 and 3 https://at.avm.de/service/wissensdatenbank/dok/FRITZ-Box-7560/3327_Von-der-FRITZ-Box-unterstutzte-SMB-Versionen/
The company exlicitly states that version 1 is routinely deactivted for security reasons.
Therefore I would have expected that specifying protocol version 1 does not work.

Nevertheless I will proceed as you suggested.

alexstrbonn, 2024-08-13:
@marcone 
Many thanks. 
The issue can be marked as "solved". As per your suggestion I explicitly specify protocol version 1.0 (https://bbs.archlinux.org/viewtopic.php?id=249503) by adding export CIFS_VERSION=1.0 to the config file and rerunning setup.

Also, just for other FritzBox users, before rerunning the Setup, in the FritzBox settings SMBv1 needs to be explicitly activated.

Now everything works well. Excellent project @marcone !

closed 2024-08-13 by marcone