marcone/teslausb #871: Re-reading the partition table failed

nmiller0113, 2024-08-29:
### Describe the problem

On a fresh install after running the following`curl https://raw.githubusercontent.com/marcone/teslausb/main-dev/setup/generic/install.sh | bash` I receive a bunch of fail/error messages like `Checking that no-one is using this disk right now ... FAILED` and `This disk is currently in use - repartitioning is probably a bad idea.
Umount all file systems, and swapoff all swap partitions on this disk.
Use the --no-reread flag to suppress this check.`. Then the last message is `Re-reading the partition table failed.: Device or resource busy
The kernel still uses the old table. The new table will be used at the next reboot or after you run partprobe(8) or partx(8).`. 

After it automatically reboots I can run the curl again and everything runs fine, but this is not stated in the Rock Pi 4C Installation instructions so I thought I'd call it out. I'll add the complete output from the command in the logs section below. I should also state that I'm using the OS image referenced in the install instructions and not doing anything outside the scope of a normal install. Also, I'm using a 1TB microSD. Thanks!

### Device

Rock Pi 4C plus

### OS Image

other

### Car Model

Model 3

### USB connection

Glove box

### Logs

root@teslausb:~# curl https://raw.githubusercontent.com/marcone/teslausb/main-dev/setup/generic/install.sh | bash
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100  7767  100  7767    0     0  12576      0 --:--:-- --:--:-- --:--:-- 12567
shrinking root partition to match root fs, 1979711488 sectors
Checking that no-one is using this disk right now ... FAILED

This disk is currently in use - repartitioning is probably a bad idea.
Umount all file systems, and swapoff all swap partitions on this disk.
Use the --no-reread flag to suppress this check.

Disk /dev/mmcblk0: 953.55 GiB, 1023871549440 bytes, 1999749120 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes
Disklabel type: dos
Disk identifier: 0x020468b7

Old situation:

Device         Boot Start        End    Sectors  Size Id Type
/dev/mmcblk0p1      32768 1979744256 1979711489  944G 83 Linux

/dev/mmcblk0p1:
New situation:
Disklabel type: dos
Disk identifier: 0x020468b7

Device         Boot Start        End    Sectors  Size Id Type
/dev/mmcblk0p1      32768 1979744255 1979711488  944G 83 Linux

The partition table has been altered.
Calling ioctl() to re-read partition table.
Re-reading the partition table failed.: Device or resource busy
The kernel still uses the old table. The new table will be used at the next reboot or after you run partprobe(8) or partx(8).
Syncing disks.
update-initramfs: Generating /boot/initrd.img-6.1.63-current-rockchip64
update-initramfs: Armbian: Converting to u-boot format: /boot/uInitrd-6.1.63-current-rockchip64
Image Name:   uInitrd
Created:      Thu Aug 29 14:53:33 2024
Image Type:   AArch64 Linux RAMDisk Image (gzip compressed)
Data Size:    21618855 Bytes = 21112.16 KiB = 20.62 MiB
Load Address: 00000000
Entry Point:  00000000
update-initramfs: Armbian: Symlinking /boot/uInitrd-6.1.63-current-rockchip64 to /boot/uInitrd
'/boot/uInitrd' -> 'uInitrd-6.1.63-current-rockchip64'
update-initramfs: Armbian: done.

marcone, 2024-08-29:
Did you actually see a failure to install due to this?
You mentioned "After it automatically reboots I can run the curl again", however after the automatic reboot setup should continue automatically, with no further intervention required until you see the flashing leds that (depending on the blink pattern) indicate there was an error, or that the config file is ready to be edited.

nmiller0113, 2024-08-29:
@marcone, that's a great question. Perhaps I didn't provide it enough time after it rebooted. After it rebooted I ssh'd back into it right after it started to respond to pings and went to edit the config file and it didn't exist. So I ran the curl again, then it ran without any errors, rebooted, then rebooted itself again and never came back so I had to pull power and then it came back and the green and blue lights started flashing. Maybe I should have given it more time, but I will say that the second time I ran the curl command it had no errors and produced a TON more lines of output before it rebooted. So I'm not sure if it was me or indeed a bug. I'd have to test it again. If you look in the log section above you can see the entire output from the first time I ran the curl and errors.

nmiller0113, 2024-08-29:
@marcone, so I tested that theory, even used a different and smaller (64 gig) sd card as I thought the 1tb may be the issue, and it gave the same errors and then rebooted. Following the reboot I let it sit for 20 mins and the install never did anything further and no config file was available. Still sitting with a solid green light and flashing blue light...nothing more.

nmiller0113, 2024-08-30:
Used a brand new 1TB microsd card and fresh install and got the exact same behavior as stated above. Ran the curl again and it did its reboots and got the blue green flashing lights with a config file. So there is definitely something causing it to have issues the first time we run it. I'm happy to provide more info or a video to capture what I'm doing.

nmiller0113, 2024-08-30:
I also wanted to update something I said earlier:

> and never came back so I had to pull power

This was actually not the case and I've not experienced that. That was me recalling something incorrectly.

nmiller0113, 2024-09-03:
Hello again, I just wanted to follow up to see if you had any other ideas? Happy to help troubleshoot this further as well. Thank You!

marcone, 2024-09-03:
This should be fixed now. It was due to a typo in a change I made back in February.

closed 2024-09-04 by marcone

nmiller0113, 2024-09-14:
Thank you @marcone, finally back in town and had a chance to test it. Works great!