marcone/teslausb #873: Stopped uploading on Jun 29th.  Still creates folders but no files.

herrjon, 2024-09-02:
### Describe the problem

Stopped uploading on Jun 29th.  Still creates folders but no files.
I have a samba share on a Synology device that my TeslaUSB uses to store saved clips, sentry, etc.
The actual copying of video files stopped on 6/29/2024 but it still creates folders.  
Video files are being saved as I can view them in the car but it won't back anything up anymore.
What information do you need to help me figure this out?
Oh, and the log file states that it's waiting for the archive to become available although it has already created a folder for today.
Thank You

### Device

Raspberry Pi Zero W

### OS Image

Prebuilt TeslaUSB image

### Car Model

Model Y

### USB connection

Glove box

marcone, 2024-09-02:
>  What information do you need to help me figure this out?

Like it said when you created this issue:

If the web interface is working, use it to download diagnostics and copy/paste or attach them here.
If the web interface is not working but you can ssh in to the Pi, run sudo /root/bin/setup-teslausb diagnose on the Pi to obtain diagnostics and attach them here.

herrjon, 2024-09-03:
Diagnostics attached.    Thank you!

[diagnostics.txt](https://github.com/user-attachments/files/16858804/diagnostics.txt)

marcone, 2024-09-04:
Your archive server appears to be full:
```
rsync: [receiver] close failed on "/mnt/archive/.teslausbtmp/2024-08-24_17-53-22-back.mp4.kUvPZE": No space left on device (28)
```

closed 2024-09-04 by marcone

herrjon, 2024-09-04:
Is that referring to the network share or a local folder?
The network share has terabytes of free space...

marcone, 2024-09-04:
It says there's no space left on /mnt/archive, which is the mountpoint where it mounts the archive server while it's archiving. Maybe your server has a quota for that specific user or something?
You could log in to the Pi and do:
```
sudo -i
mount /mnt/archive
df -h
```
to see how much space the Pi thinks there is on the archive.

herrjon, 2024-09-04:
Interesting...  it thinks there's only 500G out there.
I don't remember imposing a quota on the share.  Will check.

Thank You!

herrjon, 2024-09-04:
I don't have a quota set...  and there are 9TB free on the drive itself.  
Can I force the rsync operation so I can try to debug it further?
Thanks

marcone, 2024-09-04:
There's a button in the web interface to trigger a sync, or you can run `/root/bin/force_sync.sh` on the Pi.
Might be faster to just `mount /mnt/archive` though and then tweak things until `df -h /mnt/archive` shows the expected amount of available space on the server.

herrjon, 2024-09-04:
All fixed...  Synology offers various forms of quotas and just as many submenus to find the right one.
I apparently configured a quota when I set up the network share.  The location of the setting was far from intuitive.
Thanks for your help with this problem!
-Jon