marcone/teslausb #875: web viewer: dropdown menus misaligned on the Viewer tab

ylexus, 2024-09-05:
### Describe the problem

macOS Safari 17.6 (19618.3.11.11.5), macOS 14.6.1. the problem is shown in the attached video - each sub-menu is glued to the left of the screen and is not aligned to the menu title.

https://github.com/user-attachments/assets/25cf4b3a-ff57-4fa0-a686-86d1c4aa0bfb

Works in Chrome.

### Device

Raspberry Pi Zero W

### OS Image

Prebuilt TeslaUSB image

### Car Model

Model Y

### USB connection

Glove box

marcone, 2024-09-05:
I don't have a Mac, so if you want this fixed you're probably going to have to come up with a patch yourself.

ylexus, 2024-09-05:
I might as well, can you point me to a relevant starting point in the code please? Is it some library or all coded by hand?

marcone, 2024-09-05:
It's coded by hand. The javascript that handles the positioning of the dropdown starts here: https://github.com/marcone/teslausb/blob/main-dev/teslausb-www/html/index.html#L2632

It first calculates the number of columns for the dropdown, and then if it finds there will be only one column, it sets the left edge of the dropdown so that it's positioned below the parent. I'm guessing that Safari either doesn't get the correct client rect for the parent div, or that setting the position of the dropdown div doesn't work on Safari.

marcone, 2024-09-06:
https://github.com/marcone/teslausb/commit/575448d8a2bfb54957ef908b1ba13d51da1180d0 fixes the issue. There might be a better way so feel free to submit a better fix.

marcone, 2024-09-06:
(I should add the fix was only tested on iOS)

closed 2024-09-06 by marcone

ylexus, 2024-09-07:
Works on macOS Safari too now, thanks!